ct-Open-Source/tuya-convert #102: Successfully flashed Merkury 11w Color+white

ForSquirel, 2019-02-20:
Managed to get this flashed but it could not get it flashed as directed. Don't know if its on my end or just whats required but had to do the reset on the light after the ```start_flash script```.

Using the MagicHome module does allow for it to be used in the Tasmota web interface and gives brightness control.

Have not figured out how to do colors though.

kueblc, 2019-02-20:
We [just added support](https://github.com/arendst/Sonoff-Tasmota/pull/4991) for this LED driver. You can compile the latest development branch and set your module type to `SYF05` and GPIO13 to `SM16716 PWR` for color support.

maikthulhu, 2019-02-21:
I've uploaded a template for these Merkury bulbs at [User Created Templates](https://github.com/arendst/Sonoff-Tasmota/wiki/User-created-templates)

ForSquirel, 2019-02-28:
building and flashing latest dev branch allows for bulb to work as expected.

closed 2019-02-28 by ForSquirel