ct-Open-Source/tuya-convert #103: Sucessfully Flashed MartinJerry Smart Switch

jluzonpr, 2019-02-21:
https://www.amazon.com/gp/product/B07C235D9G

```
GPIO 5 Led1
GPIO 12 Relay1
GPIO 13 Button1
```

Gonna post all of this since it might be useful.

```
ChipID: ca883b
SdkVersion: 1.5.3(aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1440c8
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 0
system_upgrade_userbin_check: user2 0x81000
```

erwill2, 2019-02-27:
I noticed a red light and another button under the main. Can these also be setup?

jluzonpr, 2019-02-27:
@erwill2 I could definitely do some digging to find the red light but not sure what this other button would be. Could you explain more?

kueblc, 2019-03-23:
Thanks for the report @jluzonpr please feel free to add your findings to the wiki for visibility

closed 2019-03-23 by kueblc