ct-Open-Source/tuya-convert #106: Slitinto configuration after flasing

davepal, 2019-02-23:
Successfully flashed Slitinto US WiFi Dual Smart Plug with Power Monitoring NX-SP201

AS per  compatibility list following are the gpio settings : 
GPIO0: Led1i, GPIO4: Button1, GPIO5: HLW8012 CF, GPIO12: Relay1, GPIO13: Button2, GPIO14: HLWBL CF1, GPIO15: Relay2, GPIO16 HLWBL SEL 

I do not see any of the HLW settings in the drop down menu. Do they correspond to the PWM? options ?

Thanks.

DavinKD, 2019-02-24:
There's a template I uploaded on the Tasmota git page.

davepal, 2019-02-24:
I see a user created template for  SP201 based dual plugs.
 {"NAME":"SP-201","GPIO":[31,0,0,131,17,134,0,0,21,18,132,22,0],"FLAG":0,"BASE":45}

I do not see an option for creating or importing a template  as specified here : 
https://github.com/arendst/Sonoff-Tasmota/wiki/Templates

Information shows my version as : 
Program Version | 6.4.1(basic)

Tried upgrade by web server : http://thehackbox.org/tasmota/release/sonoff-basic.bin and http://sonoff.maddox.co.uk/tasmota/sonoff.bin 

Doesn't seem to be upgrading anything

DavinKD, 2019-02-24:
You have to use a dev version. Templates aren't in general release yet. I do have a custom bin you can use on my github page under SmartThings folder.

DavinKD, 2019-02-24:
https://github.com/DavinKD/SmartThings?files=1

davepal, 2019-02-24:
Do I need to manually add device to Smartthings ? Pressing the button for extended period of time does not seem to get it in pairing ( led flashing ) mode.

DavinKD, 2019-02-24:
Assuming you've flashed it, got the bin working, you would need to add it to SmartThings manually. If you use my device handler which is in the same github branch, you just need to put in the IP address of the device.

One thing to note is that SmartThings does not have a native mqtt server. I don't use one. My device handler is one way and polls every 5 minutes for the latest status. You won't get real time power usage.

There are other ways to do it with home assistant and mqtt, but I don't use them...yet

davepal, 2019-02-24:
I do have a few sonoff basic switches, use Brett's device handler.  When I go to add, it does not find device, if I manually add, it does not work. I think I also have erocm123's DH, haven't tried adding device manually with that handler.

I saw this device handler, going to give it a try -  devicetypes/davindameron/sp201-dual-energy-monitoring-plug.src/sp201-dual-energy-monitoring-plug.groovy

kueblc, 2019-03-05:
Looks like you've got everything covered, if you still have questions please open a new issue on the relevant project.

closed 2019-03-05 by kueblc