ct-Open-Source/tuya-convert #107: Amysen Smart Plug - Successfull flashing but no turn on or off rendering it useless

manjulapra, 2019-02-24:
Successfully Flashed to Amysen Brand tuya smart plug. Thanks for this great stuff. After flashing, I can access the Sonoff page and toggle the switch and the plug status light responds correctly. Following is from the console:
`00:00:00 Project sonoff Sonoff Version 6.4.1(basic)-2_4_2
00:00:00 WIF: Connecting to AP1 PraHome in mode 11N as sonoff-2283...
00:00:06 WIF: Connected
00:00:06 HTP: Web server active on sonoff-2283 with IP address 192.168.1.34
20:28:50 MQT: Attempting connection...
20:28:50 MQT: Connected
20:28:50 MQT: tele/sonoff/LWT = Online (retained)
20:28:50 MQT: cmnd/sonoff/POWER = 
20:28:50 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic","Version":"6.4.1(basic)","FallbackTopic":"cmnd/DVES_98A8EB_fb/","GroupTopic":"sonoffs"}
20:28:50 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-2283","IPAddress":"192.168.1.34"}
20:28:50 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
20:28:50 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
20:28:50 MQT: stat/sonoff/POWER = OFF
20:28:58 MQT: tele/sonoff/STATE = {"Time":"2019-02-24T20:28:58","Uptime":"0T00:00:16","Vcc":3.514,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"PraHome","BSSId":"B0:39:56:8E:63:E1","Channel":2,"RSSI":100}}
20:29:16 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
20:29:16 MQT: stat/sonoff/POWER = ON
20:29:42 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
20:29:42 MQT: stat/sonoff/POWER = OFF
20:33:58 MQT: tele/sonoff/STATE = {"Time":"2019-02-24T20:33:58","Uptime":"0T00:05:16","Vcc":3.500,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"PraHome","BSSId":"B0:39:56:8E:63:E1","Channel":2,"RSSI":100}}`

Yet the actual plug does not turn on or off. It does not even respond to the physical button. Any help is appreciated.

tkotz8105, 2019-02-24:
You probably have to set the module to generic and then trial and error to see which GPIOs work.  Here's a very good video that goes through the process.  I hope it helps.
https://youtu.be/m_O24tTzv8g

manjulapra, 2019-02-25:
Thank you. It worked. Please close this issue.

closed 2019-02-25 by manjulapra