ct-Open-Source/tuya-convert #108: Can't open sonoff page after network reset

manjulapra, 2019-02-25:
Hi, I successfully flashed a tuya smart plug. It does not work on the basic module so I changed it to Generic, changed the GPIO settings and hit save (This was the 4th plug, so I did not test with GPIO settings. Just set the correct ones) While the save was ongoing, my laptop suddenly dropped the WiFi connection and got disconnected. Since then, I can not access the Sonoff page by the IP address. The plug connects to WiFi (It connects to my Guest Network, which I gave as 2nd SSID. I still can't access the page even when the laptop is connected to it) I can't configure anything. What can I do?
Thanks in advance!

onebyside, 2019-02-25:
Does the button hold for 40 seconds reset?

kueblc, 2019-02-25:
See https://github.com/arendst/Sonoff-Tasmota for firmware support, this project only handles flashing firmware OTA which you stated has completed without issue. Please close this issue and direct future inquires to the relevant project.

FWIW your guest network probably isolates clients from talking to each other. Log on to your router config page and temporarily disable client isolation.

manjulapra, 2019-02-25:
Thanks. Changing the router settings worked.

closed 2019-02-25 by manjulapra