ct-Open-Source/tuya-convert #109: README and Project Management Improvements

kueblc, 2019-02-25:
Hi @ctandi and @merlinschumacher 

I've been keeping up with issues on the repo here and I'm noticing a trend with users seeking Tasmota support here where they may be better helped by existing support channels. With that may I recommend some additions to the README.

- Clarify that this project's function is only to flash firmware of your choice over stock Tuya firmware
- Direct users to seek assistance with their custom firmware to their respective projects
- Ask that flashing successes and failures be added directly to the wiki instead of opening an issue

For increased visibility these items could also be made part of the issue template.

I'd also ask you to consider the following, as a significant amount of energy and time has been dedicated by several others in the parallel research and development of these techniques, independently of your paper. After this project was published, much of the support for maintaining and supporting the project has come from the community.

- Add mentions to parallel projects (#1) and credit contributors in the README

For ongoing support of the project, you may wish to consider asking one or more community members to be a member of this repository so they may better assist with answering and closing issues, pull requests, and maintenance. I would be happy to take on a more active role to keep the project moving and address support issues faster.

Thanks for your consideration
// CK

Jason2866, 2019-02-25:
Everyone flashing Tasmota to a unknow devices is doing that at his own risk
To clarify. There is no support for devices from Tasmota that are not listed on Github side.
Since there are so many different devices....
Everyone flashing Tasmota to a unknow devices is doing that at his own risk!

kueblc, 2019-02-25:
Hi @Jason2866 

I'm not sure what you're getting at, would you please clarify? Sounds like you maybe misunderstood my post, as it does not concern awareness of the risk of flashing custom firmware. tuya-convert is not specific to Tasmota, it can flash any custom firmware on stock Tuya devices.

Many issues opened here are reporting successful flashing (which is the goal of this project) but then going on to ask for support setting up Tasmota (which this project is not responsible for) and they would be better served by directing them to the existing support channels for the firmware they're trying to set up, whatever that may be.

dshanske, 2019-02-25:
Maybe change from support to help, and have a few different people look at the description? For example, if someone wanted help setting up Tasmota (popular as it is) on a new device, does the Tasmota wiki have a page of advice for that?

kueblc, 2019-02-25:
Hi @dshanske 

> Maybe change from support to help, and have a few different people look at the description?

I'm not sure what you're asking.

Whatever the firmware that a user chooses to install, once installed, the job of tuya-convert is done. Any issues after that should be directed to the relevant project. If a user has an issue with Tasmota, there are no actionable items as tuya-convert does not own or maintain Tasmota.

> For example, if someone wanted help setting up Tasmota (popular as it is) on a new device, does the Tasmota wiki have a page of advice for that?

Indeed, the Sonoff-Tasmota wiki has several pages dedicated to [initial configuration](https://github.com/arendst/Sonoff-Tasmota/wiki/Initial-Configuration) of devices.

dshanske, 2019-02-25:
I was brainstorming how such an entry in the readme might read.

Jason2866, 2019-02-25:
What i want to say. Just flashing Tasmota to devices means not the will work with.
We (the Tasmota Support Team) cant help with devices we dont know 
So it makes no sense to open a issue device xy doesnt work with Tasmota! This issue will be closed.
How to setup devices initial in general is explained in wiki. For general!! questions is Tasmota Discord

kueblc, 2019-02-25:
> I was brainstorming how such an entry in the readme might read.

Ahah, thank you for your input.

> Just flashing Tasmota to devices means not the will work with

There is no claim that it will

> We (the Tasmota Support Team) cant help with devices we dont know

Neither can tuya-convert. If there is any place to ask for support for any given software, it is from that software's community.

> So it makes no sense to open a issue device xy doesnt work with Tasmota! This issue will be closed.

That is your prerogative

> How to setup devices initial in general is explained in wiki. For general!! questions is Tasmota Discord

Then this is where users should go for support, not tuya-convert

> We have to many users who bricked there devices with a Tuya flash tool and coming to Tasmota and ask for help. We cant give support for that....

I'm not saying that devices bricked through the flashing process itself should be issues of Tasmota. If the issue is with flashing then indeed, tuya-convert is the place to ask for support. If the issue is with setup of Tasmota, tuya-convert is *not* the place to ask.

Jason2866, 2019-02-25:
> If the issue is with setup of Tasmota, tuya-convert is _not_ the place to ask.
And that is the point, we cant help in most cases too. We dont know which GPIO is used for what
or if there is a additional MCU build in for something... 
So we are _not_ the place to ask too.  This are 90% of the questions.

kueblc, 2019-02-25:
Sure, if you wish you limit the scope of your project (a good idea, and exactly what I'm doing here) then you may wish to indicate which devices you will support. I'm not arguing with you here. Again, that is entirely your prerogative and not relevant to this particular issue.

The point made here is only that the scope of tuya-convert does not include setting up the firmware you choose to flash. Issues here should only pertain to the flashing process.

kueblc, 2019-02-25:
I think where your confusion lies @Jason2866 is that I'm *not* suggesting the Sonoff-Tasmota community do anything. That is entirely up to you/them. I am simply declaring what the tuya-convert community will *not* do.

Jason2866, 2019-02-25:
I agree with you 100%   All users should know that the get support here for Software Tuya not for Tasmota Software.  And Tasmota cant provide support for unknown specific device questions.

kueblc, 2019-02-25:
Great @Jason2866, I'm glad we have found common ground and agree with the approach. Thanks for your input!

GeorgeIoak, 2019-02-25:
The [Sonoff User Forum](https://groups.google.com/forum/#!forum/sonoffusers) is a perfect place to discuss new devices that have had Tasmota flashed to them.

kueblc, 2019-09-28:
Resolved

closed 2019-09-28 by kueblc