ct-Open-Source/tuya-convert #111: Gosund SP1 flashed - but only one device

hdiessner, 2019-02-25:
Successfully flashed (only) one Gosund SP1 (out of a box of 4 SP1s).
The other 3 from that box, flashing doesn't start (only the dots are counting up forever).

Device used for flashing:
- Raspberry Pi 2 with WLAN USB Stick, connected via LAN.
- latest Raspbian

After the successful flash of the first, and after the 2nd didn't start to flash, I tried:
- rebooting the Raspi
- re-run the install_prereq.sh script
- deleted the log Files from first flash
- moved the backup firmware from first flash to a new directory

Still not flashing any other of the three remaining SP1.

Any idea?

kueblc, 2019-02-25:
You should not need to reinstall the prereqs (considering it worked at least once, we know they are installed) or delete or move logs or backup files.

Let's cover the basics first. Have you ensured that the device is in pairing mode? Are you starting by running `start_flash.sh` and ending by running `stop_flash.sh`?

hdiessner, 2019-02-25:
Yes, device is in pairing mode (blue led blinking fast).
Yes, start_flash.sh executed, followed on-screen instructions until dots are piling up.
Yes, stop_flash.sh executed after first (and only so far) successful flash.
No, stop_flash.sh not executed after following, unsuccessful attempt to flash (quitted start_flash.sh with CTRL-C).

kueblc, 2019-02-25:
Does the device exit pairing mode when running the script?

Do you have anything in your `smarthack-smartconfig.log`?

If you are having issues with pairing, you may wish to try the Python version of smart-config by doing `git clone -b smart-link-py https://github.com/kueblc/tuya-convert.git`

hdiessner, 2019-02-25:
smarthack-smartconfig.log has 6-times this:
Put Device in Learn Mode!, sending Smartlink Packets now
Smartlink in progress
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
Sending Packets and wait 60 seconds.

Just executed the git clone -b smart-link-py https://github.com/kueblc/tuya-convert.git
Same behaviour: device in pairing mode, on Raspberry dots are accumulating.

kueblc, 2019-02-25:
Just to be clear, is the device still in pairing mode after this? I'm trying to isolate which part of the process is failing for you.

hdiessner, 2019-02-25:
The LED of the device is quickly blinking (in blue), no change so far. So it seems to be in pairing mode still.

kueblc, 2019-02-25:
Have you tried connecting another device (like a smartphone) to the fake AP during the smart-config process?

hdiessner, 2019-02-25:
Yes I did: yesterday, I tried with an iPhone - no success at all.
Today with an Android device - as soon as this was connected, the flashing of the first SP1 succeeded.
Now, with the Android device connected (IP Address: 10.42.42.18, the SP1 doesn't connect.

kueblc, 2019-02-25:
Thanks for the information. Interesting to note that the iPhone did not help but Android did. As of now I have no more suggestions but I'll write back if I think of anything. If anyone else has suggestions, feel free to chime in.

hdiessner, 2019-02-25:
Thank you so much for your quick response! Does the DHCP server perhaps „reserve“ IP addresses for specific devices which prevents the SP1 from receiving a specific IP address? (Not sure how the fake-AP supposed to work)

kueblc, 2019-02-25:
You're welcome!

DHCP IP assignments are defined by (this line)[https://github.com/ct-Open-Source/tuya-convert/blob/master/scripts/setup_ap.sh#L35]. It gives devices an available IP address range of `10.42.42.10` to `10.42.42.40` but does not specify any reservations.

kueblc, 2019-02-25:
It does not seem likely that would be the source of your problem. It seems much more likely that the smart-config process is the issue, as your device is still in pairing mode.

Emulating the smart-config process is a little tricky as it relies on UDP packets being delivered in a precise order with precise timing, something which UDP isn't necessarily designed to do.

mobidarabbit, 2019-02-26:
Is it possible that your other Gosund SP1 devices had received the newest tuya firmware which maybe closed the "security breach". Could you check the firmware version of the stock firmware. Should be possible in smartlife app. everything other than 1.0.0.0 could maybe be the issue.

hdiessner, 2019-02-26:
Versions according to the SmartLife App:
MCU-module: 1.0.5
Wifi-module: 1.0.5
This is exactly the same version as in the SP1 I was able to flash successfully.

mobidarabbit, 2019-02-26:
Ok so that is not the issue. Just wanted to exclude this. I have no other idea. Maybe its possible to sniff the packets between flashing device and the device you want to flash and see whats going on at this layer.

hdiessner, 2019-02-26:
Thank you for your ideas.
In all my desparation, I set up another Raspberry Pi 2B, with WLAN adapter - but still no luck.
Any other ideas or would I have to give up?

mobidarabbit, 2019-02-26:
Giving up is not an option.. :) you still have the soldering flash method left.  Is the wifi stick working good on another system? Maybe there are problems with the stick. I am not thinking its the tuya-convert software which is failing. I think it must be a problem with something else. Perhaps its helpful to sniff with tcpdump what pakets leave the wifi stick and if some pakets come back from the gosund device. Or try another stick if you have one.

hubgitbb, 2019-02-28:
hi
i have the same problem and would like to know what exactly to do with tcpdump.
if i type tcpdump in the terminal it show somethin like this in very fast 
11:25:51.244217 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 371
11:25:51.263028 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 231
11:25:51.281827 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 137
11:25:51.300611 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 372
11:25:51.319409 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 301
11:25:51.338110 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 358
11:25:51.356956 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 364
11:25:51.375865 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 133
11:25:51.394646 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 138
11:25:51.413425 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 353
11:25:51.432311 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 371
11:25:51.451157 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 360
11:25:51.469822 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 256
11:25:51.689241 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 18
11:25:51.710974 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 43
11:25:51.732816 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 48
11:25:51.754534 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 67
11:25:51.776279 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 194
11:25:51.798105 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 128
11:25:51.819876 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 271
11:25:51.841634 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 358
11:25:51.863368 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 364
11:25:51.885762 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 353
11:25:51.907647 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 228
11:25:51.929578 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 129
11:25:51.951328 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 371
11:25:51.973240 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 360
11:25:51.994979 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 365
11:25:52.016797 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 357
11:25:52.038761 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 212
11:25:52.060637 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 130
11:25:52.082690 IP notebook-pc.63145 > 255.255.255.255.30011: UDP, length 361

mobidarabbit, 2019-02-28:
Your device sends the Broadcast for the Smartconfig (looks like tuya-convert is working fine), but it looks like there is no answer from the gosund device.  So if the device is in pairing mode, it seems like the manufactures closed this possibility to hack the device with a FW-Update.

kueblc, 2019-02-28:
Hi @hubgitbb

We cannot be sure that this is the same issue. Please open a new issue with your device info and logs.

@hdiessner Would you share your logs as well?

hubgitbb, 2019-03-01:
i tried with another pc and now it work
raspi zero w with the latest raspian didnt work
laptop with ubuntu 18.04 also didnt work
old pc with ubuntu 18.04 works

red-hood, 2019-03-01:
I tried flashing a shipment of Gosund SP1 from today using as Raspberry 3 B+, everything worked fine so far using the bundled 3rd party tasmota firmware. I also tried using a Intel Wifi Module on a laptop, but hostapd would fail with FW errors from the wifi.

kueblc, 2019-03-05:
Is this issue resolved?

hdiessner, 2019-03-05:
Sorry, no - still open. But this week, I am not able to do any more tests. However, new Raspi (3B) ordered and on the way. Hopefully, I can find out more next week.

chrhaeusler, 2019-03-05:
Same problem here and it's driving me nuts...
I managed to flash one gosund SP111 Smart Socket (by accident?) and fail to flash a second smart socket ( and quickly tried to flash a Luminea Smart Bulb, and a Joywell Power Strip)

I tried:
- using Kali Linux installed on a netbook with internal and couple of external wifi adapters (tried the python fork, as well)
- using raspbian running on a raspberry pi 3 B+ 
- a VBox Image of Kali Linux on a desktop computer with internal and external wifi adapters.

...but no luck. To be honest, I have absolutely no clue what I might have done wrong.

Anyways, the last screen output says:
Waiting for the upgraded device to appear.
If this does not work have a look at the '*.log'-files in the 'scripts' subfolder!
......................

The last modified log-file in the mentioned folder is the smarthack-smartconfig.log saying:
Smartlink in progress
Sending SSID vtrust-flash
Sending wifiPassword flashmeifyoucan
Sending Packets and wait 60 seconds.

...maybe this information is useful for the pros.

WaaZaa666, 2019-03-06:
I am running into exactly the same on the same devices. First device was working fine. Second- fourth devices are without luck. I am running from an Ubuntu live USB 18.10. Also tried restarting the ubuntu and reinstall (who know what cache). 

It seems I now have 2 bricked. Since the LED will not turn on any more. Working on screwing these open. 

Attached my logs, perhaps some clues in there?
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/2938022/smarthack-mqtt.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/2938024/smarthack-smartconfig.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/2938025/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/2938026/smarthack-wifi.log)

WaaZaa666, 2019-03-07:
For an update. To clarify what I meant with bricked: button did not do anything anymore. LED would only flash once and stay off. 

During flashing with tuyaconvert, the LED would turn red first, then it would turn off and go into the state I said above. 

In any case, I decided to just open them up and do the serial soldering flash, this worked just fine. 

For all others, I presume that tuya has implemented some security update to prevent this from happening after the first device.

kueblc, 2019-03-07:
Hi @chrhaeusler and @WaaZaa666 

Please open a separate issue as we can't be sure these are the same problem. I'll take a look at the logs you posted @WaaZaa666 and could you please post them too @chrhaeusler?

kueblc, 2019-03-07:
> For all others, I presume that tuya has implemented some security update to prevent this from happening after the first device.

Let's not panic just yet. I'm fairly doubtful they would have "patched" this "bug", and so quickly at that.

kueblc, 2019-03-07:
@WaaZaa666 it looks like your device flashed the intermediate firmware successfully, but you modified the access point and password so it can't connect. Launch the vtrust access point and you should be able to use the `/undo` or `/flash` commands.

mobidarabbit, 2019-03-07:
> For all others, I presume that tuya has implemented some security update to prevent this from happening after the first device.

I  don't think that it's possible for the manufacturer to block this flash method after the first device has been flashed. It's just possible that there is a firmware update that closes the "weak spot" and 3 of 4 devices has been running this firmware. But thats just a possibility i thrown in the ring. My devices where running FW 1.0.0 and worked fine, so FW 1.0.5 combined with the message from tuya they closed the "bug" and are rolling out the firmware to customer brands were a possible reason that the flash is not working. But it seems like if the logs/sniffs look good there are still other reasons why it wont work.Just like the issue @hubgitbb has with his hardware/software combo.

kueblc, 2019-03-07:
> combined with the message from tuya they closed the "bug" and are rolling out the firmware to customer brands

I haven't heard about this, could you share your reference?

mobidarabbit, 2019-03-07:
There was a heise.de message. It is a german article, but maybe you find something in english somewhere else. 

https://www.heise.de/newsticker/meldung/Smart-Home-Hack-Tuya-veroeffentlicht-Sicherheitsupdate-4292028.html

kueblc, 2019-03-07:
Good to be aware of, thanks for sharing. Looks like they'll be pushing firmware updates via HTTPS now, and also signing firmware updates at some point, so soon we may have to find another way around when those changes take effect.

Frustrating that manufacturers ignore the problem (the cloud) and treat the symptoms (those who break the cloud).

In the meantime better stock up on flashable devices and keep them sandboxed from the cloud so they don't silently push an update.

chrhaeusler, 2019-03-07:
just a quick addition: 
I bought the devices one week after the [original article](https://www.heise.de/ct/artikel/Tuya-Convert-IoT-Geraete-ohne-Loeten-vom-Cloud-Zwang-befreien-4283623.html) about the tuya hack was published at heise.de, and the devices never had internet access (they were never connected to my flat's wlan)

mobidarabbit, 2019-03-08:
@chrhaeusler If the plugs were offline the hole time, the firmware issue should not apply. For me, a Pi Zero W with raspbian strecht lite worked well. I just had trouble with installing node (because no ARMv6 binaries in package) but i was able to manual install them via tarball method. And you have to connect you directly (HDMI & USB keyboard) to the Zero W, because you are not able to ssh, if you want to start the ap. Also "killall wpa_supplicant" processes before starting the ap. So if you are able to get an Pi Zero W from somewhere without buying it, test it. The node problem is now solved with the python branch, so give it a try on Zero W. Wish you luck.

hdiessner, 2019-03-10:
Hello,
my new Raspi 3B arrived and I just was able to flash the three other devices successfully. Worked like a charm. So to wrap up this issue: Raspi 2B with Edimax USB Wifi worked only once - with Raspi 3B it worked repeatedly. Thank you all for your help. Closing this issue.

closed 2019-03-10 by hdiessner

kueblc, 2019-03-10:
Glad you got it resolved. Thanks for updating us and closing the issue!

kueblc, 2019-03-14:
If anyone has **bricked** their device and they have a backup of the firmware before or after the brick, please let me know so we can track down the bug.

kb9nate, 2019-10-12:
I ran into this same problem today using a Raspberry Pi 3 Model B V1.2. Yesterday I successfully flashed a Vivitar LB-95 and was trying to flash another one today. The Pi had been powered down since yesterday. So today I go to flash another one of the same bulb and it would not work. Long story short what I did to get it working was to do the ./stop_flash.sh command even after I had done CTRL + C to exit from trying to flash. Then started it again and it worked right away. 

There most be some process that keeps running and even runs again on power up if ./stop_flash.sh has not been sent.

madjack84, 2020-04-23:
@kueblc : I have a gosund SP1 and want to flash it with a working pipeline (RPI3, tuya convert) already flashed multiple plugs (aisierer, zoozee)

The SP1 does blink and stops blinking when starting  ./start_flash
But the it does not continue.

smarthack-wifi.log says: pairwise key handshake completed (RSN)
smarthack-web.log
most recent call list:
./fake-registration-server.py, line 132 in main()

Any help on this base? As the plug is from a friend, I do not know whether he upgraded the firmware :( but is it maybe the updated one?
(FYI: I once had problems flashing cause of a underpowered RPI.... so use a potent adapter:) )

egrotius, 2021-11-13:
Mine got bricked after flashing. I guess no way to resolve? 

My logs
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/7531647/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/7531648/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/7531649/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/7531650/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/7531651/smarthack-wifi.log)

kueblc, 2021-11-13:
@egrotius fret not, it is not bricked! It is running the intermediate tuya-convert firmware.

You can restart the `start_flash` script, but don't put the device in pairing mode. Instead, unplug the device and when it asks you to press enter, just plug the device back in. In a few seconds, it should make a new backup and then ask which firmware to install.

egrotius, 2021-11-13:
Thanks, no luck though. I also soldered the wires and flashed (i did that
first before i used your solution). Flashing is possible, and sometimes i
do get the AP mode (and then connect to the wifi). But after short time it
stops working.

kueblc, 2021-11-13:
Hmm, so do you mean that you flashed your alternative firmware and were able to connect, for a short time, to the configuration AP?

If that is the case, it sounds like it is an issue with the alternative firmware. tuya-convert only provides the means to install the firmware, but no guarantee that the firmware will work well on the device. You might have luck reaching out the alternative firmware's community for support.

egrotius, 2021-11-13:
Correct, i flashed 1 other gosund p1 without any problems. Only this one has errors. Now nothing happens. I still can flash (and got succesfull statement but in the end the unit does not start up. Thank you for all your answers