ct-Open-Source/tuya-convert #112: Trying to flash Plugs and they're not being picked up?

Springs87, 2019-02-26:
Hi all, i'm trying to flash a AISIRER plug.

I've flashed another plug some time ago and since then i can't get the tuya-convert to pick up any devices.
All the plugs i've tried have already been configured in the tuya app. I've reflashed the SD card and gone through the pre-req's again.

I've pressed the button and the led has flashed rapidly and i've got an android tabled connected to the SSID. I've ran the ./start-flash.sh as both the standard user as well as under sudo.

I've looked at the logs but they don't seem to show anything particular that i can understand.

Any ideas on what i can look at? I've attached the logs as well

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/2907847/smarthack-mqtt.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/2907848/smarthack-smartconfig.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/2907849/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/2907850/smarthack-wifi.log)

erwill2, 2019-02-27:
They fastest way I got them to connect was to set everything up and remove them from the tuya app. Mine immediately went in pairing mode/ fast flashing blue light. Then I ran start flash.

bigdavediode, 2019-02-27:
Also you can try resetting the device to defaults in the app which forces it into pairing mode. Then flash.

bigdavediode, 2019-02-28:
erwill is right, above -- that worked for me with Merkury bulbs. Also leave your network cable plugged in even though you're going through wireless. For me it wouldn't work when the network cable was unplugged.

kueblc, 2019-02-28:
Based on the logs, it does not look like a pairing issue.

Can you confirm that 
1. the device was in pairing mode (quickly blinking)
2. the device exited pairing mode during the script

From the web log, I can see the device verifying the token, and later taking the upgrade response. I can also see several Ctrl-C. You may have exited the script while it was still working.

Springs87, 2019-03-05:
I managed to get one to connect and the convert ran and read the original firmware but then it decided to stop without touching the software and now that plug is bricked.

kueblc, 2019-03-05:
What do you mean when you say the plug is "bricked"? Did the firmware backup complete successfully? Does the plug light or do anything when plugged in or button pressed? Can you see any new APs or devices on your network?

Springs87, 2019-03-05:
The firmware backed up fine, went to flash the new one and it said it couldn't find the device on the network.
No lights showing on the plug either and no response to trying to turn it on or off manually

kueblc, 2019-03-23:
It's probably not bricked, but just waiting to be configured. A fresh install of Sonoff-Tasmota might not react to any button presses. Look for a `sonoff-*` access point.

kueblc, 2019-04-04:
Closing due to inactivity. If you're still experiencing issues, try again with the latest version and open a new issue if you encounter problems.

closed 2019-04-04 by kueblc