ct-Open-Source/tuya-convert #113: How to reflash the backup created by tuya-convert

alfb113, 2019-02-27:
Thanks for the tuya-convert project, it works well.
What is the recommended method to re-install the back file created by tuya-convert.
In my case the file is 1Mb and the tasmota basic was only < 512Kb.
One is warned that the file to flash must fit!

Regards, Alf

kueblc, 2019-02-27:
You can return your device to stock firmware by [opening it and connecting a serial programmer](https://github.com/arendst/Sonoff-Tasmota/wiki/Hardware-Preparation) and then [using the esptool](https://github.com/arendst/Sonoff-Tasmota/wiki/Upload#esptool) to flash the backup file back onto the device.

As the backup file is a one-to-one dump of the flash contents, including unused areas, you will likely not be able to restore it via OTA methods as is. In theory I can imagine that one might be able to strip the backup file down but no guarantees there. I have not heard of any attempts to do this so far.

alfb113, 2019-02-27:
Colin.

Thanks for your quick response.

tuya-convert worked well, and I was able to decipher the switch, led and 
relay settings for this previously unmentioned device.. . Ardwolf WS-1 
from Amazon. Its a round plug with no apparent method of entry (barring 
extreme force).

It responds to MQTT, the tasmota web-app and the physical switch. Sounds 
like the relay clicks, the led responds, and the app and MQTT respond to 
button presses.  I was also able to upgrade Tasmota to 6.4.1.19 using 
the OTA method.

However, the power to the load remains on all the time.  This failure to 
switch the load either chose now to fail, or Tasmota does play nicely 
with this hardware.

Were it relatively easy, as a test  I would have backdated the firmware 
to the old tuya app where it is known to have worked.

Thanks again for your interest.

Alf

kueblc, 2019-02-27:
You're welcome. Feel free to share the configuration you're using if you'd like me to double check it. If you feel your issue has been resolved you can close this ticket.

Thanks // CK

closed 2019-02-28 by alfb113

nospam2k, 2020-12-17:
Appologies if I missed this, but shouldn't this be documented that you cannot go back to stock without dissassembly? I wish I would have know this before I flashed them. In the mean time, does anyone have a procedure to remove the LED board without damaging the housing?