ct-Open-Source/tuya-convert #114: Flashed Merkury LED RGB strip, unsure of configuration

ForSquirel, 2019-02-27:
I've managed to flash a Merkury RGB led strip 

I've found that working with the sonoff basic that
- gpio4->pwm1->red
- gpio5->pwm1->cool white
- gpio12->pwm1->green
- gpio13->pwm1->warm white
- gpio14->pwm1->blue
- gpio0->button0

but cannot get anything working past that. 

After this I'm not sure where to go or how to help from here. It looks like the only thing going on is that the ESP is running the LEDs through PWM channels. There are 3 LEDS on the strip itself. 1 cool white, 1 warm white, and 1 RGB

DavinKD, 2019-02-27:
In Tasmota go to configure module.  Pick Generic and the fill out the pins.

gpio4 - pwm1
gpio5 - pwm5
gpio12 - pwm2
gpio13 - pwm4
gpio14 - pwm3
gpio0 - button1

You may need the latest dev release to use gpio0 in generic.  Also, I could have pwm4/5 backwards.  I never remember the order of the whites.

DavinKD, 2019-02-27:
PS, you can test it using the console.

Color FF00000000 - Red
Color 00FF000000 - Green
Color 0000FF0000 - Blue
Color 000000FF00 - One of the whites
Color 00000000FF - The other white

Use the ct 153..500 command to test the whites.  If they are backwards, swap them.  153 would be warmest, 500 would be coldest.

ForSquirel, 2019-02-27:
Thanks so much.  I wasn't quite sure how to put that in to the module config. Everything works like it should now.

closed 2019-02-27 by ForSquirel

Crepetsky, 2019-05-03:
Perhaps a stretch to ask here, but I'll give it a shot:

I was able to flash the same Merkury LED RGB Strip, and got it working with the above config options. Has anyone successfully got the color changing features working with Home Assistant? I can see the strip over MQTT just fine, but all I can do is turn it on and off.

In contrast, sending color commands to it directly works just fine with the Sonoff page.