ct-Open-Source/tuya-convert #115: BSD29 UK plug works

AS137430, 2019-03-01:
The UK version is BSD29.

GPIO12 =  Relay 1
GPIO13 = Button 1
GPIO14 = Led1i

ChipID: fb88de
MAC: XX:XX:XX:XX:XX:XX
Sdk Version: 1.5.3 (aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M DOUT @ 40Mhz
FlashChipId: 144051
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 3
system_upgrade_userbin_check: user2 0x81000

_Originally posted by @AS137430 in https://github.com/ct-Open-Source/tuya-convert/issues/54#issuecomment-468272439_

AS137430, 2019-03-03:
Energy monitoring GPIOs added:
```
GPIO4 = BL0937 CF
GPIO5 = HLWBL CF1
GPIO15 = HLWBL SELi
```
 Updated in wiki and Tasmota template:
{"NAME":"BSD29","GPIO":[0,0,0,0,134,132,0,0,21,17,56,131,0],"FLAG":0,"BASE":45}

closed 2019-03-03 by AS137430

tyjtyj, 2019-04-01:
I am not able to calibrate the power monitoring correctly using tasmota.... Apparent Power always same value

kueblc, 2019-04-01:
@tyjtyj Please refer to the support channels for Tasmota

Woodpeckercz, 2020-03-11:
> I am not able to calibrate the power monitoring correctly using tasmota.... Apparent Power always same value

The template is wrong, this is the correct one:
{"NAME":"BSD29","GPIO":[0,0,0,131,134,132,0,0,21,17,56,0,0],"FLAG":0,"BASE":52}