ct-Open-Source/tuya-convert #116: Maxcio Curtain Switch GPIOs required

GACHAPO, 2019-03-02:
I have successfully flashed a Maxcio Curtain Switch but it doesn't work manually with button presses and via tasmota, as I clearly have the incorrect GPIOs set.  Can anybody please assist me in setting the correct ones if possible please?

AS137430, 2019-03-02:
#49 

> If the WP3 configuration doesn't work, @digiblur has a procedure in his latest YouTube livestream to determine the proper GPIO configuration. Here's how I would summarize that:

```
Set every GPIO (excluding, at first, "dedicated" like Tx/Rx) to relays. 
Save configuration. 
Use web GUI toggles to find which GPIO control relays and LEDs. 
If unable to find relays and LEDs, set the "dedicated" GPIO and retry. Set the "active" GPIO to associate them with the RelayX or LEDX. Operation may dictate regular or inverted (RelayXi/LEDXi).
Set every remaining GPIO (excluding, at first, "dedicated" like Tx/Rx) to Switches. 
Save configuration. Display Status 8 (sensors) in the Console to see the current state of the GPIO. 
Then, while pressing the button(s), display Status 8 again. Whichever GPIO changes state is the GPIO tied to the button. If none change state, then configure the "dedicated" GPIO and retry. 
Change the GPIO to a Button (may need to determine whether the internal pull-up is used or not - ButtonXn for no pull-up). If you want to use short-press/long-press functionality (described in a couple other of Travis' videos), configure the button as a Switch instead.
Set remaining GPIO to None
```

GACHAPO, 2019-03-04:
Thank you this was very helpful and I have (to a fashion) best matched the Relays, Leds and Buttons as I will indicate below.

However, it must be noted that I was unable to match the Tuya switch configuration fully as the Switch is made up of three button; down, pause and up.  Tasmota only has the option of a single button or relay being in two states; on & off and that functionality is unable to be shared across two buttons or relays.  Therefore, you are unable to have relay1 & button1 just acting as on, relay2 & button2 acting as pause and relay3 and button3 acting as on.  So, the upshot is that the pause button is redundant and the up and down buttons are pressed twice to pause.  I also failed somewhat with the LEDs.  The best configuration means that the up button works okay, blue when inactive and orange when active, but the down button is darker blue when inactive and remaining so when active, but changing the up buton to orange when active.
[image]

[image]

Thanks again

AS137430, 2019-03-04:
You can try to use Tasmota rules to aggregate relay1 and button1 etc.

kueblc, 2019-03-04:
This is a question for the Tasmota project, sounds like you have gotten tuya-convert to work successfully. Please see Sonoff-Tasmota if additional support is needed, you can close this issue.

closed 2019-03-05 by GACHAPO