ct-Open-Source/tuya-convert #117: IoT device won't connect

tmaex, 2019-03-02:
Hi there! I'm about to flash sonoff devices (4CH Pro, TH16) and set up everything on my ubuntu laptop.

I disabled systemd-resolved successfully for that.
Now everything seems to be fine, but non of the devices wants to connect.

1. started the script
2. `yes` enter
3. my android phone connects successfully to the wifi
4. set the device to searching mode (flashes constant fast, also tried "short short long" mode)
5. press enter again
6. nothing is happening besides more and more dots
7. devices leaves paring mode on its own (flashes constant slow)

`tuya-convert/scripts$ cat *.log
1551529179: mosquitto version 1.4.15 (build date Wed, 13 Feb 2019 00:27:01 +0000) starting
1551529179: Using default config.
1551529179: Opening ipv4 listen socket on port 1883.
1551529179: Opening ipv6 listen socket on port 1883.
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.
Listening on port 80
Backing up NetworkManager.cfg...
Restarting NetworkManager...
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
Starting AP on wlp2s0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Using interface wlp2s0 with hwaddr 14:ab:c5:xxx and ssid "vtrust-flash"
wlp2s0: interface state UNINITIALIZED->ENABLED
wlp2s0: AP-ENABLED 
wlp2s0: STA d4:38:9c:xxx IEEE 802.11: authenticated
wlp2s0: STA d4:38:9c:xxx IEEE 802.11: associated (aid 1)
wlp2s0: AP-STA-CONNECTED d4:38:9c:xxx
wlp2s0: STA d4:38:9c:xxx RADIUS: starting accounting session D192BCAxxx
wlp2s0: STA d4:38:9c:xxx WPA: pairwise key handshake completed (RSN)
`

I also tried the python branch.

Sadly I can''t help myself without errors. Any advice here?

erwill2, 2019-03-02:
Tuya-covert will not work on itead/sonoff devices or devices that use the ewlink app. This project is for Tuya / SmartLife / devices.

tmaex, 2019-03-02:
Thanks @erwill2. My bad, sorry.

closed 2019-03-02 by tmaex