ct-Open-Source/tuya-convert #118: DIGOO DG-SP01

fragolinux, 2019-03-02:
today i flashed 3 devices, so i know the procedure, but can't succeed in flashing the one in title...
attached logs... device seems connecting to rpi AP as it stops blinking led but after about 10 minutes is still there, waiting to do don't know what...
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/2922352/smarthack-smartconfig.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/2922353/smarthack-mqtt.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/2922354/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/2922355/smarthack-wifi.log)

kueblc, 2019-03-02:
Out of curiosity, which devices flashed successfully?

It looks like the pairing process completed successfully and the upgrade message was sent but the device did not fetch the firmware file. This indicates to me that the format of the response was not as expected.

I would like to track this down but I cannot reproduce this issue so any information you can provide will be very helpful.

fragolinux, 2019-03-02:
tell me if we can chat on discord tomorrow and i'll do any test you ask...
these are the devices i flashed:

https://github.com/arendst/Sonoff-Tasmota/wiki/BlitzWolf-SHP6
https://blakadder.github.io/templates/houzetek-swa1.html
https://blakadder.github.io/templates/alfawise-pe1004.html

latest 2 are their templates i published after configuring them

an other device i couldn't flash is this, but then i opened it and saw it has a mediatek chip inside, not esp8266... https://www.amazon.it/Interruttore-spegnimento-elettronico-telecomando-elettrodomestici/dp/B01N11AW60

fragolinux, 2019-03-03:
succeeded to flash today, don't know what was different... i think something with dhcp leases, if you do many devices at once... i started this evening and it just worked... i flashed an Emylo switch, too, no problems... thanks!
[image]

kueblc, 2019-03-03:
OK perfect, glad it worked out! You can close this issue if you are satisfied.

closed 2019-03-03 by fragolinux

Selters, 2019-05-28:
> succeeded to flash today, don't know what was different.

Hi @fragolinux ,

Could you please let me know how to open the case of DIGOO DG-SP01?
Are there any hidden screws?

Thanks in advance!

fragolinux, 2019-05-29:
> > succeeded to flash today, don't know what was different.
> 
> Hi @fragolinux ,
> 
> Could you please let me know how to open the case of DIGOO DG-SP01?
> Are there any hidden screws?
> 
> Thanks in advance!

ehm, as we are discussing on how to flash them without opening, obviously i never opened it, flashed via tuya-convert...

Selters, 2019-05-29:
> how to flash them without opening

Thanks!
This was the missing part ;)

MrMennens, 2019-06-09:
Did you flash a specific firmware? Or just the normal tasmota?
Wondering if you can choose to turn on/off the socket, light & the 2 usb ports

fragolinux, 2019-06-10:
flash 6.5.0 based on sdk 2.3.0, then go on blakadder site and search for the template to use, done

MrMennens, 2019-06-10:
thnx for the reply,

can't seem to get it flashed with tuya-convert..
putty only shows ..........................
led of the digoo stopped blinking after a while..

MrMennens, 2019-06-25:
Tried a couple different pi's,
Tried a couple of new installs
still nothing, could it be the firmware of the Digoo DG-SP01 is just not working with the tuya convert anymore?
If it doesn't work i'm checking out to see if i can make it work with a IFTTT workaround...

zehrer, 2021-02-01:
The Template Settings Sems not working well. E.g. the status led is on when the light is on. Somebody have the same issue?