ct-Open-Source/tuya-convert #119: Meross mss425e

fragolinux, 2019-03-02:
same for the other device i just posted an issue, no way to flash this one, too... these are the logs...
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/2922373/smarthack-web.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/2922374/smarthack-smartconfig.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/2922375/smarthack-mqtt.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/2922376/smarthack-wifi.log)

kueblc, 2019-03-02:
Can you control this device with the Tuya SmartLife app? It was my understanding that this is not a Tuya device.

fragolinux, 2019-03-02:
I'll try... Don't know 😬

fragolinux, 2019-03-03:
nope, it's not recognized by tuya nor smartlife... but it has a tuya chip inside, from a similar item already on tasmota wiki (different socket): https://github.com/arendst/Sonoff-Tasmota/wiki/meross-MSS425

kueblc, 2019-03-03:
Do you know that it is a Tuya chip from opening it up? Can you share pictures?

Otherwise, given that it's not recognized by SmartLife, I don't believe we can flash it using Tuya OTA unfortunately.

fragolinux, 2019-03-03:
i didn't open it up, but it's pretty much the same as the one on tasmota wiki i linked, the final E is for European sockets...

kueblc, 2019-03-03:
Puzzling. According to [the FCC documents](https://fccid.io/2AMUU-MSS425/Internal-Photos/Internal-Photos-3707507) it looks like it's a Mediatek chip inside. Not sure how the other fellow managed to flash his, or why it might be a different chip.

fragolinux, 2019-03-03:
i'll open it up, if i find the suitable screw driver and let you know :)

fragolinux, 2019-03-04:
removed back cover, without unscrewing all the pcb i can see the chipset, indeed is a mediatek mt7687f unfortunately... don't know what did that guy on tasmota wiki... closing issue, thanks anyway :)

closed 2019-03-04 by fragolinux

kueblc, 2019-03-04:
Thanks for following up, good to know for the future. Sorry we could flash this one!