ct-Open-Source/tuya-convert #123: Proxmox - Tuya-Convert LXC enviroment - Automated Install 1 liner by Whiskerz007

SirRedZ, 2019-03-06:
For those who are running ProxMox - Whiskerz007 has created a stellar script to auto-install everything based on 1 line of script to be run on PVE > Shell/Console. Once executed, all pre-req.s will be installed, your wifi adapter(s) will be identified in a selectable list, and you'll be flashing in minutes. Done flashing for now? - Once you flash a module the script will auto-stop and logout. Need to flash another? Just login.

https://github.com/whiskerz007/proxmox_tuya-convert_container/tree/whiskerz007-patch-1

He did a great job!

kueblc, 2019-09-28:
Created a [wiki page](https://github.com/ct-Open-Source/tuya-convert/wiki/Additional-Resources) to host these sorts of things, feel free to edit and add

closed 2019-09-28 by kueblc

nunokaneco, 2020-07-29:
Hello 

I'm trying to install but I'm receiving this error, can you please help me ? 

'''
root@pve:~# bash -c "$(wget -qLO - https://github.com/whiskerz007/proxmox_tuya-convert_container/raw/master/create_container.sh)"
E: Failed to fetch https://enterprise.proxmox.com/debian/pve/dists/buster/InRelease  401  Unauthorized [IP: 212.224.123.70 443]
E: The repository 'https://enterprise.proxmox.com/debian/pve buster InRelease' is not signed.
[ERROR] 100@64 Unknown failure occured.
[ERROR] 100@67 Unknown failure occured.
root@pve:~#

Tollbringer, 2020-07-29:
You need to also disable the Enterprise REPO and setup " A community version ".
Copy and paste this entire section as 1 script and run it once. It will persist after Proxmox updates.

```# Disable Commercial Repo
sed -i "s/^deb/\#deb/" /etc/apt/sources.list.d/pve-enterprise.list
apt-get update

# Add PVE Community Repo
echo "deb http://download.proxmox.com/debian/pve $(grep "VERSION=" /etc/os-release | sed -n 's/.*(\(.*\)).*/\1/p') pve-no-subscription" > /etc/apt/sources.list.d/pve-no-enterprise.list
apt-get update

# Remove nag
echo "DPkg::Post-Invoke { \"dpkg -V proxmox-widget-toolkit | grep -q '/proxmoxlib\.js$'; if [ \$? -eq 1 ]; then { echo 'Removing subscription nag from UI...'; sed -i '/data.status/{s/\!//;s/Active/NoMoreNagging/}' /usr/share/javascript/proxmox-widget-toolkit/proxmoxlib.js; }; fi\"; };" > /etc/apt/apt.conf.d/no-nag-script
apt --reinstall install proxmox-widget-toolkit```