ct-Open-Source/tuya-convert #124: Teckin SP21

Lyr3x, 2019-03-06:
I tried to flash the Teckin SP21 yesterday but the socket wont connect to the AP. Smartphone Connection worked fine. Did anyone had success with the SP21?

kueblc, 2019-03-06:
[Check the wiki](https://github.com/ct-Open-Source/tuya-convert/wiki/Compatible-devices) to see if anyone else has had success for any given device, in this case it does look like people have successfully flashed the Teckin SP21.

Please share the rest of your logs so we can help you debug your issue.

Lyr3x, 2019-03-06:
I found the SP21 in the compatible devices table, thats why i bought them. I'll post the rest later 👍

HennieLeermans, 2019-03-06:
I have the same problem. I am using an HP probook with Kali Linux kali-linux-mate-2019.1a-amd64
I will try another distro later today..

connecting to the AP is no problem.
have you checked smarthack-smartconfig.log? Mine is complaing about the debug module missing.

I do not have a pi.

kueblc, 2019-03-23:
Awaiting feedback

HennieLeermans, 2019-03-23:
You can close on my end.
I installed NPM into my kali live cd and then it was no problem.

kueblc, 2019-03-23:
Glad to hear it @HennieLeermans 

@Lyr3x if you still need help, please try the latest version and then post your logs

closed 2019-03-23 by kueblc

smoothunit, 2019-07-27:
Yeah, does this for me in all my deb vm's. Different wifi cards make no diff.