ct-Open-Source/tuya-convert #125: Red flashing LED

sobo85, 2019-03-06:
Hi

I flashed my Gosund SP1 with Tasmota. After flashing i've connected to the "sonof xxxx" wifi and filled my ssid and password. After reboot automatic of the SP1 it can't connect to wifi because of wrong password (fritzbox told me). Can i reset the wifi setting? The buttom of the SP1 doesn't work.

AS137430, 2019-03-07:
[https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-entered-the-wrong-wi-fi-password-and-i-have-no-button-to-reset-config](Tasmota Wiki)

sobo85, 2019-03-09:
1 device still flashing red :/

kueblc, 2019-03-09:
@sobo85 This is the issue tracker for tuya-convert. Please see Sonoff-Tasmota for support with their firmware. Thanks and good luck!

closed 2019-03-09 by kueblc