ct-Open-Source/tuya-convert #127: Tried flashing (flash3) but didn't work, tried undoing, now device unresponsive - Tonbux switch

melikyuksel, 2019-03-10:
I'm kinda freaking out. I successfully flashed the middleware firmware, got this info:

```
READ FLASH: http://10.42.42.42/get?read=HEX-ADDRESS
ChipID: 258c75
MAC: 3C:71:BF:25:8C:75
SdkVersion: 1.5.3(aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1440c8
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 0
system_upgrade_userbin_check: user2 0x81000
```

Then I did `curl http://10.42.42.42/flash3`. It said `Device should flash http://10.42.42.1/files/thirdparty.bin and restart`. But the device didn't restart, it just stayed connected to the `vtrust-flash` and the "sonoff-xx" AP never came up.

I waited an hour, and then did `curl http://10.42.42.42/undo` to try again. It said it would undo and restart, and the device disconnected from `vtrust-flash` but never turned back on. Now I can't get it to turn on at all.

onebyside, 2019-03-10:
system_upgrade_userbin_check: user2 0x81000

That tells you to flash "flash2" .....this put the flash loader in the correct poistion.

Then flash "flash3".
When it restarts you wont see any lights. You will see sonoffxxx appear in your wifi list to connect to.

Then you will configure tasmota.

melikyuksel, 2019-03-10:
Hmm, that wasn’t clear from the documentation. Is there any way I can get the switch to get un-bricked? Is it even bricked?

AlphaDE, 2019-03-10:
I ran into the same issue with my Heygo H02, I simply restarted the process from the beginning and the script recognized the plug instantly and I could issue flash2 and then flash3. Similiar worked when the firmware backup failed, restarted the script and could continue.

melikyuksel, 2019-03-10:
@AlphaDE How did you get it to go into pairing mode again? I tried to start the process again, but my switch has no lights or clicks or any reaction at all anymore when I hold down the button like before, and it doesn't appear to be sending any probe requests or any sort of packets (judging from Wireshark). No signs of life :(

AlphaDE, 2019-03-10:
@melikyuksel 
I did not had to get it manually into pairing mode, the device reacted when I just restarted the script and on another (where I assumed that I bricked it, because it did not react in any way anymore, no button, no led) it also connect and resumed after plugging it in.

The stub installed by the script obviously stays up/comes up without manual intervention.

melikyuksel, 2019-03-10:
@AlphaDE I followed those steps, but the device isn't responding. Gave up after over an hour. So it looks the undo operation screwed it up in a big way where it doesn't even boot anymore. I don't know what to do to salvage it

AlphaDE, 2019-03-10:
The undo removes the stub, so if not told otherwise there is nothing to boot anymore. If you can open it, you could try to flash a firmware via the serial port of that chip. But that's more effort (and needs a USB-serial-convert and more to do it).

melikyuksel, 2019-03-10:
I wasn't able to resuscitate the first switch OTA, so I'll try the soldering for it. I have no idea how to do it, but will look at guides. Perhaps I can use a Pi Zero in gadget mode as the USB serial device.

However, I tried the OTA process again on another one of this switch, and I got it to successfully flash! This time, this switch kept timing out while backing up the factory firmware, but the start_flash.sh would just keep going.

```
Traceback (most recent call last):
  File "./backup.py", line 42, in <module>
    main()
  File "./backup.py", line 27, in main
    conn.request("GET", "/get?read=%X" % address )
  File "/usr/lib/python3.5/http/client.py", line 1107, in request
    self._send_request(method, url, body, headers)
  File "/usr/lib/python3.5/http/client.py", line 1152, in _send_request
    self.endheaders(body)
  File "/usr/lib/python3.5/http/client.py", line 1103, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python3.5/http/client.py", line 934, in _send_output
    self.send(msg)
  File "/usr/lib/python3.5/http/client.py", line 877, in send
    self.connect()
  File "/usr/lib/python3.5/http/client.py", line 849, in connect
    (self.host,self.port), self.timeout, self.source_address)
  File "/usr/lib/python3.5/socket.py", line 712, in create_connection
    raise err
  File "/usr/lib/python3.5/socket.py", line 703, in create_connection
    sock.connect(sa)
OSError: [Errno 113] No route to host
~/tuya-convert
```

I tried it over and over like 3 times, and on the 3rd try, it worked to back up the firmware without errors. When I did `curl http://10.42.42.42/flash2` as recommended, it said `Device is allready booting from userspace 2 (0x81000)`. So that wasn't the problem with the first one...

I did `curl http://10.42.42.42/flash3`, and got the `Device should flash http://10.42.42.1/files/thirdparty.bin and restart` message. I crossed my fingers. And my switch started blinking green! An improvement. Then the `sonoff-xxxx` access point came up, and I was able to configure the switch successfully!

Going to solder the first one. The second, successfully flashed switch is now blinking weird colors (blinking green/nothing when off, blinking red/yellow when on). How does one go about "fixing" this?

onebyside, 2019-03-11:
Gonna pass you this vid to watch so it makes clear how to go about configing Tasmota....figuring out the pins.

https://www.youtube.com/watch?v=m_O24tTzv8g

melikyuksel, 2019-03-11:
@onebyside @AlphaDE -- Got the first one flashed as well using the serial ports with the Raspberry Pi Zero. Nice! Both have Tasmota now. I realized the first failure never managed to backup the factory firmware. Perhaps some sort of Wi-Fi connection issue with the vtrust-flash AP on the Pi Zero the first time around.

closed 2019-03-11 by melikyuksel

kueblc, 2019-03-14:
@melikyuksel Do you have any firmware backups of problematic devices? I am tracking down a bug and a copy of firmware where the flash did not go as expected would be very helpful for narrowing this down.

AlphaDE, 2019-03-14:
Where did the script store the backup? I run into these problems (timout while saving the original firmware, but resuming was possible) with my HEYGO H02 US plugs and on my Raspberry Pi3B that backup should exist.

kueblc, 2019-03-14:
The backups are stored in the `tuya-convert` directory by default, with a timestamp and `_readout.bin`

AlphaDE, 2019-03-14:
I have nothing such there...

kueblc, 2019-03-14:
Too bad, thank you for checking anyway!

melikyuksel, 2019-03-30:
@kueblc I do have firmware from the exact same devices as the one that initially failed, but I’ve flashed 4 of them so far, and only the first one failed. I feel like there was some sort of corruption of the sent firmware files.

The reason I think this is is because I used the Raspberry Pi Zero W’s built-in WiFi to try to flash the first device. There were many disconnects. I then switched to an external USB WiFi adapter I use for penetration testing (monitor mode capable), and the process worked well for the rest of the devices. However, on one of the later devices even with this new adapter, the device was trying to connect to “vtrt-t-flash” instead of the usual “vtrust-flash” (checked with a packet sniffer), so I had to reflash it. So obviously some sort of glitch.

So there appears to be a noticeable potential for data corruption between the script and the ESP device during the OTA flashing.

kueblc, 2019-03-31:
There was a bug in the intermediate firmware that caused undo to fail if you were already booting from userspace 1. I believed this to be related to the problems you reported, but the bug is now fixed in the current master. I have found identical devices from the same box that boot from different userspaces so it's entirely possible that you just happened to undo on one of the devices that booted from 1.

I have not gotten any other reports of corruption but if you have any data I would be happy to look into it.