ct-Open-Source/tuya-convert #130: Device connects but script does not continue

nickdos, 2019-03-11:


kueblc, 2019-03-11:
Could you post the rest of your logs?

nickdos, 2019-03-11:
Thanks @kueblc - here they are...

smarthack-web.log:
```
Listening on port 80
```
smarthack-smartconfig.log:
```
Put Device in Learn Mode!, sending Smartlink Packets now
Smartlink in progress
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
Sending Packets and wait 60 seconds.
```

smarthack-mqtt.log (593 lines long):
```
1552273999: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1552273999: Using default config.
1552273999: Opening ipv4 listen socket on port 1883.
1552273999: Opening ipv6 listen socket on port 1883.
1552273999: New connection from 192.168.1.65 on port 1883.
1552273999: New client connected from 192.168.1.65 as DVES_0753CC (c1, k10, u'family').
1552273999: Sending CONNACK to DVES_0753CC (0, 0)
1552273999: Received PUBLISH from DVES_0753CC (d0, q0, r1, m0, 'tele/sonoff-livingroom/LWT', ... (6 bytes))
1552273999: Received PUBLISH from DVES_0753CC (d0, q0, r0, m0, 'cmnd/sonoff-livingroom/POWER', ... (0 bytes))
1552273999: Received SUBSCRIBE from DVES_0753CC
1552273999:     cmnd/sonoff-livingroom/# (QoS 0)
1552273999: DVES_0753CC 0 cmnd/sonoff-livingroom/#
1552273999: Sending SUBACK to DVES_0753CC
1552273999: Received SUBSCRIBE from DVES_0753CC
1552273999:     cmnd/sonoffs/# (QoS 0)
1552273999: DVES_0753CC 0 cmnd/sonoffs/#
1552273999: Sending SUBACK to DVES_0753CC
1552273999: Received SUBSCRIBE from DVES_0753CC
1552273999:     cmnd/DVES_0753CC_fb/# (QoS 0)
1552273999: DVES_0753CC 0 cmnd/DVES_0753CC_fb/#
1552273999: Sending SUBACK to DVES_0753CC
1552273999: New connection from 192.168.1.8 on port 1883.
1552273999: New client connected from 192.168.1.8 as DVES_074D6E (c1, k10, u'family').
1552273999: Sending CONNACK to DVES_074D6E (0, 0)
1552273999: Received PUBLISH from DVES_074D6E (d0, q0, r1, m0, 'tele/sonoff-laundry/LWT', ... (6 bytes))
1552273999: Received PUBLISH from DVES_074D6E (d0, q0, r0, m0, 'cmnd/sonoff-laundry/POWER', ... (0 bytes))
1552273999: Received SUBSCRIBE from DVES_074D6E
1552273999:     cmnd/sonoff-laundry/# (QoS 0)
1552273999: DVES_074D6E 0 cmnd/sonoff-laundry/#
1552273999: Sending SUBACK to DVES_074D6E
1552273999: Received SUBSCRIBE from DVES_074D6E
1552273999:     cmnd/sonoffs/# (QoS 0)
1552273999: DVES_074D6E 0 cmnd/sonoffs/#
1552273999: Sending SUBACK to DVES_074D6E
1552273999: Received SUBSCRIBE from DVES_074D6E
1552273999:     cmnd/DVES_074D6E_fb/# (QoS 0)
1552273999: DVES_074D6E 0 cmnd/DVES_074D6E_fb/#
1552273999: Sending SUBACK to DVES_074D6E
1552274009: Received PINGREQ from DVES_074D6E
1552274009: Sending PINGRESP to DVES_074D6E
1552274009: Received PINGREQ from DVES_0753CC
1552274009: Sending PINGRESP to DVES_0753CC
1552274019: Received PINGREQ from DVES_074D6E
1552274019: Sending PINGRESP to DVES_074D6E
1552274019: Received PINGREQ from DVES_0753CC
1552274019: Sending PINGRESP to DVES_0753CC
1552274029: Received PINGREQ from DVES_074D6E
1552274029: Sending PINGRESP to DVES_074D6E
1552274029: Received PINGREQ from DVES_0753CC
1552274029: Sending PINGRESP to DVES_0753CC
1552274039: Received PINGREQ from DVES_074D6E
1552274039: Sending PINGRESP to DVES_074D6E
1552274039: Received PINGREQ from DVES_0753CC
[... 516 lines cut ...]
1552275272: Sending PINGRESP to DVES_0753CC
1552275281: Received PINGREQ from DVES_074D6E
1552275281: Sending PINGRESP to DVES_074D6E
1552275282: Received PINGREQ from DVES_0753CC
1552275282: Sending PINGRESP to DVES_0753CC
1552275287: Received PUBLISH from DVES_074D6E (d0, q0, r0, m0, 'tele/sonoff-laundry/STATE', ... (214 bytes))
1552275287: Received PUBLISH from DVES_074D6E (d0, q0, r0, m0, 'tele/sonoff-laundry/SENSOR', ... (129 bytes))
1552275291: Received PINGREQ from DVES_074D6E
1552275291: Sending PINGRESP to DVES_074D6E
1552275292: Received PINGREQ from DVES_0753CC
1552275292: Sending PINGRESP to DVES_0753CC
1552275301: Received PINGREQ from DVES_074D6E
1552275301: Sending PINGRESP to DVES_074D6E
1552275302: Received PINGREQ from DVES_0753CC
1552275302: Sending PINGRESP to DVES_0753CC
1552275311: Received PINGREQ from DVES_074D6E
1552275311: Sending PINGRESP to DVES_074D6E
1552275312: Received PINGREQ from DVES_0753CC
1552275312: Sending PINGRESP to DVES_0753CC
1552275313: Received PUBLISH from DVES_0753CC (d0, q0, r0, m0, 'tele/sonoff-livingroom/STATE', ... (218 bytes))
1552275321: Received PINGREQ from DVES_074D6E
1552275321: Sending PINGRESP to DVES_074D6E
1552275322: Received PINGREQ from DVES_0753CC
1552275322: Sending PINGRESP to DVES_0753CC
```

kueblc, 2019-03-11:
Nothing more in the `smarthack-web.log`? What kind of wireless adapter are you using?

nickdos, 2019-03-11:
No, that's it in `smarthack-web.log`. I'm using built-in wifi radio... does the 3B+ not support the built-in wifi?

onebyside, 2019-03-11:
3b+ worked fine for me

kueblc, 2019-03-12:
The device exiting pairing mode would suggest it received the token assigned by the smart config process, but we should see that in the `smarthack-web.log` and we don't. Interestingly though, we see the device connect to the AP in `smarthack-wifi.log` so we know it got the credentials. This may suggest the requests to `tuya*.com` are not being intercepted.

Can you check if `dnsmasq` is running properly? We should be able to connect to the fake AP and perform a `dig` to see if `tuya*.com` resolves to your rPi's IP address.

nickdos, 2019-03-12:
Thanks, I'll try it when I get home.

Could it be relayed to this error line in wifi log:
`Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver`?

nickdos, 2019-03-13:
Just tried it again now. When the script is running, I can connect my phone to the AP and then use Chrome to go to `http://tuya.com` and I get a page back with "Hello world" text. So that suggests `dnsmasq` is working, I think.

See also:
```
pi@raspberrypi:~ $ ps -ef | grep dnsmasq
dnsmasq    636     1  0 04:20 ?        00:00:00 /usr/sbin/dnsmasq -x /run/dnsmasq/dnsmasq.pid -u dnsmasq -r /run/dnsmasq/resolv.conf -7 /etc/dnsmasq.d,.dpkg-dist,.dpkg-old,.dpkg-new --local-service --trust-anchor=.,19036,8,2,49aac11d7b6f6446702e54a1607371607a1a41855200fd2ce1cdde32f24e8fb5 --trust-anchor=.,20326,8,2,e06d44b80b8f1d39a95c0b0d7c65d08458e880409bbc683457104237c7f8ec8d
```

ianxtreem, 2019-03-14:
I have the same error in my log.

`Failed to create interface mon.wlan0: -95 (Operation not supported) wlan0: Could not connect to kernel driver?`

I dont see the AP vtrust-flash.   Just a data point.

I think it has something to do with the driver nl80211 as when I do a modprobe i dont see it.  I am not a linux expert so I may be totally off base here.
--

kueblc, 2019-03-14:
This error is benign. There may be other issues present but we do not need monitor mode for this application. @ianxtreem if you could, open a new issue with your logs and info on your devices and wifi hardware and I'd be happy to take a closer look.

kueblc, 2019-03-14:
@nickdos Sounds similar to https://github.com/codetheweb/tuyapi/issues/175, could it be related?

kueblc, 2019-09-28:
Closing due to inactivity

See #279 for HTTPS support

closed 2019-09-28 by kueblc