ct-Open-Source/tuya-convert #131: Succesfully flashed Panamalar NX-SM200

eelcohn, 2019-03-11:
**Successfully flashed Panamalar NX-SM200**
Amazon link: [https://www.amazon.de/gp/product/B07JBRRW1M/](https://www.amazon.de/gp/product/B07JBRRW1M/)

As described in [this issue](https://github.com/ct-Open-Source/tuya-convert/issues/67) the procedure for flashing the Panamalar devices is a bit different. After starting the `start_flash.sh` do NOT connect any laptops/smartphones etc to the `vtrust-flash` Wifi network. Instead just skip step 1 and start with step 2 to connect the Panamalar device to the Wifi network.

**Template**
I slightly modified the known user template because the Panamalar device does not have any user defined GPIO's available. The Panamalar device also has 2 LED's, a blue one and a red one. Blue signals Wifi and 'switched off'. Red signals 'switched on'.
```
{"NAME":"NX-SM200","GPIO":[0,0,0,0,52,134,0,0,131,17,132,21,0],"FLAG":0,"BASE":45}
```
**GPIO's**
```
GPIO0 : None (0)
GPIO1 : None (0)
GPIO2 : None (0)
GPIO3 : None (0)
GPIO4 : Led1 (52)
GPIO5 : BL0937 CF (134)
GPIO9 : None (0)
GPIO10: None (0)
GPIO12: HLWBL SELi (131)
GPIO13: Button1 (17)
GPIO14: NLWBL CF1 (132)
GPIO15: Relay1 (21)
GPIO16: None (0)
```
**Device info**
```
ChipID: xxxxxx
MAX: 84:0D:8E:xx:xx:xx
SdkVersion: 1.5.3(aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144068
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 3
system_upgrade_userbin_check: user2 0x81000
```

lug-gh, 2019-03-19:
For me it worked exactly like documented in tuya-convert, no need to skip steps...

kueblc, 2019-03-23:
Thanks for the contribution, did you add it to the wiki?

closed 2019-03-23 by kueblc

eelcohn, 2019-03-25:
@kueblc : I just added the device to the wiki

kueblc, 2019-03-25:
Thank you very much!

eelcohn, 2019-04-17:
It looks like Panamalar re-designed this smart plug circuitry and used different GPIOs. I just received a new batch of the Panamalar SM-SX200 smart plugs and the Template above doesn't work with the new plugs anymore.

My preliminary result for a new template is:
```
{"NAME":"NX-SM200","GPIO":[17,0,0,0,52,132,0,0,0,130,21,134,0],"FLAG":0,"BASE":45}
```
Device info on the new plugs:
```
ChipID: xxxxxx
MAC: B4:E6:2D:xx:xx:xx
SdkVersion: 1.5.3(aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 146085
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 0
system_upgrade_userbin_check: user2 0x81000
```
Notice the different MAC, FlashChipId and FlashChipMode.

LukeAQ, 2019-04-22:
I have two Panamalar from the new batch. Your template doesn't work.
- the button is ok
- the red led is ok but the blue one continue to blink
- the reported voltage is ok but the power of any load is reported as always 0
Moreover your template sets GPIO13=(130) but your GPIO list sets GPIO13=(131)

wecl, 2019-05-14:
Same issue here. Current (ampere) is always 0.
The blue LED is at GPIO13

MAC-Adresse | B4:E6:2D:61:1D:A2
ESP Chip ID | 6364578

kueblc, 2019-05-14:
You are more likely to get help with Sonoff-Tasmota at their project page

eelcohn, 2019-05-15:
Thanks for all the feedback. Unfortunately I haven't had much time lately to dive into this issue. I hope to do this before the end of may.
The cause of this problem is due to an incorrect assignment of BL0937CF, HLWBL CF1, HLWBL SEL and/or HLWBL SELi. These are the pins connected to the BL0937 chip, which manages the voltage, current and power measurements.
If you're feeling adventurous you can help by trying different combinations yourself as described here: https://github.com/arendst/Sonoff-Tasmota/wiki/Configuration-Procedure-for-New-Devices

wecl, 2019-05-17:
I've found a working combination for my device:

{"NAME":"NX-SM200","GPIO":[17,0,0,0,134,132,0,0,0,52,21,130,0],"FLAG":0,"BASE":45}

tacruc, 2019-11-24:
What worked for me:
{"NAME":"NX-SM200","GPIO":[158,0,0,131,0,134,0,0,0,17,132,21,0],"FLAG":0,"BASE":18}