ct-Open-Source/tuya-convert #132: Device connects requests IP

micktech, 2019-03-12:
Everything works up until the device connects to the wifi network. Checking /var/log/syslog can see the device asks for an IP and gets 10.42.42.25. Scanned the IP and only port 6668 is open, no port 80. Any ideas on what I'm doing wrong ?

kueblc, 2019-03-12:
Please post your logs so we can better understand the problem

micktech, 2019-03-12:
smarthack-smartconfig.log

>Put Device in Learn Mode!, sending Smartlink Packets now
>Smartlink in progress
>Sending SSID                  vtrust-flash
>Sending wifiPassword          flashmeifyoucan
>Sending Packets and wait 60 seconds.

webserver log

> Listening on port 80

wifi log

mac address: e4:xx (iPhone) 60:01 (Brilliant Smart WiFi Plug) 

> Backing up NetworkManager.cfg...
> Restarting NetworkManager...
> Backing up /etc/dnsmasq.conf...
> Writing dnsmasq config file...
> Creating new /etc/dnsmasq.conf...
> Writing hostapd config file...
> Configuring AP interface...
> Applying iptables rules...
> Starting DNSMASQ server...
> Starting AP on wlan0 in screen terminal...
> Configuration file: /etc/hostapd/hostapd.conf
> Using interface wlan0 with hwaddr 1c:xx:xx:xx:xx:37 and ssid "vtrust-flash"
> wlan0: interface state UNINITIALIZED->ENABLED
> wlan0: AP-ENABLED
> wlan0: STA 60:01:xx:xx:a5:67 IEEE 802.11: authenticated
> wlan0: STA 60:01:xx:xx:a5:67 IEEE 802.11: associated (aid 1)
> wlan0: AP-STA-POSSIBLE-PSK-MISMATCH 60:01:xx:xx:a5:67
> wlan0: AP-STA-POSSIBLE-PSK-MISMATCH 60:01:xx:xx:a5:67
> wlan0: AP-STA-POSSIBLE-PSK-MISMATCH 60:01:xx:xx:a5:67
> wlan0: STA 60:01:xx:xx:a5:67 IEEE 802.11: disassociated
> wlan0: STA 60:01:xx:xx:a5:67 IEEE 802.11: disassociated
> wlan0: STA 60:01:xx:xx:a5:67 IEEE 802.11: disassociated
> wlan0: STA 60:01:xx:xx:a5:67 IEEE 802.11: disassociated
> wlan0: STA 60:01:xx:xx:a5:67 IEEE 802.11: disassociated
> wlan0: STA 60:01:xx:xx:a5:67 IEEE 802.11: disassociated
> wlan0: STA 60:01:xx:xx:a5:67 IEEE 802.11: disassociated
> wlan0: STA 60:01:xx:xx:a5:67 IEEE 802.11: disassociated
> wlan0: STA 60:01:xx:xx:a5:67 IEEE 802.11: disassociated
> wlan0: STA 60:01:xx:xx:a5:67 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
> wlan0: STA e4:xx:xx:xx:xx:11 IEEE 802.11: authenticated
> wlan0: STA e4:xx:xx:xx:xx:11 IEEE 802.11: associated (aid 1)
> wlan0: AP-STA-CONNECTED e4:xx:xx:xx:xx:11
> wlan0: STA e4:xx:xx:xx:xx:11 RADIUS: starting accounting session 85xxxxxxxxxxxx6F
> wlan0: STA e4:xx:xx:xx:xx:11 WPA: pairwise key handshake completed (RSN)
> wlan0: STA 60:01:xx:xx:a5:67 IEEE 802.11: authenticated
> wlan0: STA 60:01:xx:xx:a5:67 IEEE 802.11: associated (aid 2)
> wlan0: AP-STA-CONNECTED 60:01:xx:xx:a5:67
> wlan0: STA 60:01:xx:xx:a5:67 RADIUS: starting accounting session BCxxxxxxxxxxxxE2
> wlan0: STA 60:01:xx:xx:a5:67 WPA: pairwise key handshake completed (RSN)
> wlan0: AP-STA-DISCONNECTED e4:xx:xx:xx:xx:11
> wlan0: STA e4:xx:xx:xx:xx:11 IEEE 802.11: authenticated
> wlan0: STA e4:xx:xx:xx:xx:11 IEEE 802.11: associated (aid 1)
> wlan0: AP-STA-CONNECTED e4:xx:xx:xx:xx:11
> wlan0: STA e4:xx:xx:xx:xx:11 RADIUS: starting accounting session 85xxxxxxxxxxxx6F
> wlan0: STA e4:xx:xx:xx:xx:11 WPA: pairwise key handshake completed (RSN)
> wlan0: AP-STA-DISCONNECTED e4:xx:xx:xx:xx:11
> wlan0: STA e4:xx:xx:xx:xx:11 IEEE 802.11: disassociated due to inactivity
> wlan0: STA e4:xx:xx:xx:xx:11 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

After this the device is pingable at 10.42.42.25 ,
port 6668 is open but nothing else.

kueblc, 2019-03-12:
Could you try capturing the activity with `tcpdump` or Wireshark?

micktech, 2019-03-12:
Thanks for your time @kueblc 
Is this dump ethernet is the name of my kali box

> 12:20:54.948887 ARP, Request who-has 10.42.42.42 tell ethernet, length 28
> 12:20:55.707490 IP 10.42.42.25.49153 > ethernet.domain: 50810+ A? a3.tuyaeu.com. (31)
> 12:20:55.707720 IP ethernet.domain > 10.42.42.25.49153: 50810* 1/0/0 A 10.42.42.1 (47)
> 12:20:55.713790 IP 10.42.42.25.39183 > ethernet.https: Flags [S], seq 8134, win 4380, options [mss 1460], length 0
> 12:20:55.713837 IP ethernet.https > 10.42.42.25.39183: Flags [R.], seq 0, ack 8135, win 0, length 0
> 12:20:56.532411 ARP, Request who-has 10.42.42.42 tell ethernet, length 28
> 12:20:56.592981 ARP, Request who-has 10.42.42.25 tell 10.42.42.25, length 28
> 12:20:56.593004 ARP, Request who-has 10.42.42.25 tell 10.42.42.25, length 28
> 12:20:57.540951 ARP, Request who-has 10.42.42.42 tell ethernet, length 28
> 12:20:58.564934 ARP, Request who-has 10.42.42.42 tell ethernet, length 28
> 12:20:58.832209 IP 10.42.42.25.49153 > ethernet.domain: 21504+ A? a3.tuyaeu.com. (31)
> 12:20:58.832418 IP ethernet.domain > 10.42.42.25.49153: 21504* 1/0/0 A 10.42.42.1 (47)
> 12:20:58.837786 IP 10.42.42.25.15639 > ethernet.https: Flags [S], seq 8192, win 4380, options [mss 1460], length 0
> 12:20:58.837841 IP ethernet.https > 10.42.42.25.15639: Flags [R.], seq 0, ack 8193, win 0, length 0
> 12:21:00.152668 ARP, Request who-has 10.42.42.42 tell ethernet, length 28
> 12:21:01.156886 ARP, Request who-has 10.42.42.42 tell ethernet, length 28
> 12:21:01.963471 IP 10.42.42.25.49153 > ethernet.domain: 31385+ A? a3.tuyaeu.com. (31)
> 12:21:01.963684 IP ethernet.domain > 10.42.42.25.49153: 31385* 1/0/0 A 10.42.42.1 (47)
> 12:21:01.970793 IP 10.42.42.25.2852 > ethernet.https: Flags [S], seq 8250, win 4380, options [mss 1460], length 0
> 12:21:01.970850 IP ethernet.https > 10.42.42.25.2852: Flags [R.], seq 0, ack 8251, win 0, length 0
> 12:21:02.180883 ARP, Request who-has 10.42.42.42 tell ethernet, length 28

kueblc, 2019-03-12:
Interesting, it looks like the device is demanding HTTPS. This complicates things a bit, you may have been affected by the patch Tuya released to close our firmware freedom.

Can you share details about the device and firmware version? I'd like to start building a workaround for this latest patch if we can.

micktech, 2019-03-13:
The device is a "Brilliant" branded Smart WiFi Plug. Because it was being used with their app originally I'd say it has the latest firmware. MCU: 1.0.3 and Wifi: 1.0.3.
Thanks for your help, I might buy another one and test it before connecting through any app.

nickdos, 2019-03-13:
This looks similar to what I'm seeing with the same device and firmware version - https://github.com/ct-Open-Source/tuya-convert/issues/130. The Tuya app forced me to update the firmware as the device stopped working and it dropped off my HA setup (using Tuya component). I now wish I'd tried this project then, as opposed to waiting until now :-(

kueblc, 2019-03-14:
@nickdos Can you capture your AP activity with `tcpdump` or Wireshark? Any captures or firmware dumps will be helpful in tracking down what they might've changed.

> The Tuya app forced me to update the firmware as the device stopped working and it dropped off my HA setup (using Tuya component). 

Did the update break your HA setup or was that unrelated? That would be a huge bummer if they take an active stance against local control as well.

nickdos, 2019-03-14:
> Did the update break your HA setup or was that unrelated? That would be a huge bummer if they take an active stance against local control as well.

No, the device just went offline and didn't appear in HA anymore. When I went into the Tuya app, it prompted me to update the firmware and I tried to skip but it kept popping up the upgrade dialog so I clicked upgrade, just to get it back on the network. Its working fine again now in HA.

kueblc, 2019-03-23:
@micktech @nickdos Any news? Are you able to do a network capture?

micktech, 2019-03-24:
Not as of yet, I've tried attaching to the RX TX pins but think one of the copper pads got pulled off in the process. I'll give network capture another go when I get a bit ore time.

Joshi425, 2019-05-11:
@micktech any news on this? I'm having the same Issue with a new Gosund SP1 Plug

micktech, 2019-05-12:
I managed to get the pins connected and dumped the firmware before flashing it to Tasmota. Before that, the device was trying to connect to the server via https and I couldn't get much more information than that. If you could do a network capture on yours maybe you could see something that I couldn't pick up

aschwersenz, 2019-06-08:
I have the same problem. Is there a solution by now?

magichew, 2019-06-26:
I'm experiencing a similar issue with a smart RGBW bulb. I made mistake of connecting via Smart Life app first, but it didn't overtly do a firmware update that I could tell (maybe firmware was patched already from factory or maybe it was done at set up).

If it will help I can send dumps of whatever, will just need a bit of instruction on how to do it.

magichew, 2019-06-27:
I have tried this on both Raspberry Pi and custom Linux build both with same result.

joshuafallon, 2019-08-05:
I have this ESP8266 based [garden RGB LED ](https://sunshoweronline.com.au/garden-lighting/led-garden-lighting/wifi-controlled-led-garden-lights/brilliant-smart-wi-fi-garden-light-controller) set from Brilliant, it seems that they are affected by the TLS issue.
[aldi-lights3.pcap.zip](https://github.com/ct-Open-Source/tuya-convert/files/3466180/aldi-lights3.pcap.zip)

Upon powering up the lights they attempt to lookup a3.tuyaus.com then connect to it port 443.
My basic attempts to impersonate a3.tuyus.com have been to create a local Nginx server on the Pi, create self-signed certificates... but that doesn't get me far.

The RGB controller sends a TLS 1.2 client hello packet with 2 ciphers offered:
1. TLS_PSK_WITH_AES_128_CBC_SHA256
2. TLS_EMPTY_RENEGOTIATION_INFO_SCSV

I believe this is typical of the 2.0 TLS patch ... happy to provide additional logs & packet captures as needed.

kueblc, 2019-09-28:
See #279 for HTTPS support

closed 2019-09-28 by kueblc