ct-Open-Source/tuya-convert #133: Merkury wifi bulbs

ragenhe, 2019-03-12:
Using a dedicated (to tuya-convert) SD Card and a RPI3b I have successfully flashed eight Gosund plugs.
All moving parts worked in other words.

When attempting to flash the Merkury 9W white wifi bulb, the process just hangs at dots going across the screen while waiting for the device to connect / appear.

I have confirmed that the Tuya Smart app can connect and control the bulbs.

I will continue trying some of the notes / misc ways people are reporting success after work today.

ragenhe, 2019-03-12:
kueblc -  will the newer process (smart-link-py) maybe have different results?

kueblc, 2019-03-12:
It may, `smart-link-py` uses two strategies (broadcast and multicast) to pair devices as opposed to just broadcast. It may also resolve some timing issues, certainly worth a try.

ragenhe, 2019-03-12:
Same steps as original project? @kueblc

kueblc, 2019-03-12:
Yes

ragenhe, 2019-03-13:
@kueblc - I do have the Vtrust recovery AP and I can get to the device info page.  I understand this is a good finding. However, I couldn't get the curl command to complete the flash, nor would (re)running the ./start_flash make any difference.  In both cases, I attempted flash3 without certainty it is correct.  From the info page, I read that available storage was 512K, not 1MB. Does this impact which method? Should my next step be flash method other than 3, or possible the smart-link-py branch?

ragenhe, 2019-03-14:
No progress with the smart-link-py on the previous attempted bulb, or a fresh one.  The fresh one has this info available...... Wifi firmware version 1.1.0.  MCU module firmware version 1.1.0
Observation.... It seems these two bulbs are picking up an IP from my DHCP server before they pickup on the vtrust-flash AP.  I disconnected the DHCP while attempting, no difference, but see next note.
Also, the fresh bulb stops blinking as soon as it pairs with the tuyasmart app.  Previously I thought, on my first attempt, when it stopped blinking it was out of pairing mode all together.  On the first attempt, even though it stopped blinking (as if it was paired) it didn't register as paired with the flashing process.  I'm going to take the fresh bulb offline and attempt to really scrub the process.  Stay tuned.

kueblc, 2019-03-14:
Yes, if it pairs it will stop blinking. Feel free to share your logs for additional feedback.

Check if your wifi log shows the device's mac address connecting. This would indicate the pairing was successful.

Check also to see if the web log shows any activity. If it does, please share. If it doesn't, see if you can capture what's happening on the network with `tcpdump` or Wireshark on the device running the access point.

ragenhe, 2019-03-14:
Info page from device:

VTRUST-FLASH 1.0
(c) VTRUST GMBH https://www.vtrust.de/35c3/

READ FLASH: http://10.42.42.42/get?read=HEX-ADDRESS
ChipID: 7eb841
MAC: 5C:CF:7F:7E:B8:41
SdkVersion: 1.5.3(aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1440c8
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 0
system_upgrade_userbin_check: user2 0x81000

kueblc, 2019-03-14:
This is great, you've successfully loaded the flash stub. You can now backup your firmware and replace it with your choice of alternatives. Please let us know what trouble you experienced and how you resolved it for future reference.

ragenhe, 2019-03-14:
wlan0: STA 5c:cf:7f:7e:b8:41 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 5c:cf:7f:7e:b8:41
wlan0: STA 5c:cf:7f:7e:b8:41 RADIUS: starting accounting session 0DB96D51CEF1F8A7
wlan0: STA 5c:cf:7f:7e:b8:41 WPA: pairwise key handshake completed (RSN)
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED 5c:cf:7f:7e:b8:41
wlan0: AP-DISABLED
nl80211: deinit ifname=wlan0 disabled_11b_rates=0

ragenhe, 2019-03-14:
Traceback (most recent call last):
  File "./fake-registration-server.py", line 132, in <module>
    main()
  File "./fake-registration-server.py", line 128, in main
    tornado.ioloop.IOLoop.current().start()
  File "/usr/local/lib/python3.5/dist-packages/tornado/platform/asyncio.py", line 132, in start
    self.asyncio_loop.run_forever()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 421, in run_forever
    self._run_once()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 1388, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.5/selectors.py", line 445, in select
    fd_event_list = self._epoll.poll(timeout, max_ev)
KeyboardInterrupt

ragenhe, 2019-03-14:
I haven't been able to do anything even though I have the recovery AP.  The logs above are from me attaching my computer to the recovery ap, and attempting the curl command to flash.  It says it will flash and restart, but when it restarts, it is still in recovery mode, and not flashed.

kueblc, 2019-03-14:
> wlan0: STA 5c:cf:7f:7e:b8:41 IEEE 802.11: associated

Is `5c:cf:7f:7e:b8:41` the mac address of the device you're trying to flash?

> Traceback (most recent call last):
> File "./fake-registration-server.py", line 132, in
> main()
> File "./fake-registration-server.py", line 128, in main
> tornado.ioloop.IOLoop.current().start()
> File "/usr/local/lib/python3.5/dist-packages/tornado/platform/asyncio.py", line 132, in start
> self.asyncio_loop.run_forever()
> File "/usr/lib/python3.5/asyncio/base_events.py", line 421, in run_forever
> self._run_once()
> File "/usr/lib/python3.5/asyncio/base_events.py", line 1388, in _run_once
> event_list = self._selector.select(timeout)
> File "/usr/lib/python3.5/selectors.py", line 445, in select
> fd_event_list = self._epoll.poll(timeout, max_ev)
> KeyboardInterrupt

Is this everything that was in the web log?

> I haven't been able to do anything even though I have the recovery AP. The logs above are from me attaching my computer to the recovery ap, and attempting the curl command to flash. It says it will flash and restart, but when it restarts, it is still in recovery mode, and not flashed.

Are you still running `fake-registration-server.py` when you `curl`? Can you see any requests on your end?

ragenhe, 2019-03-14:
Yes.
There is more, but it is the same (duplicated).
Ummmmm.... no.  I have been running ./start_flash again, should I be running ./fake-registration-server instead?

kueblc, 2019-03-14:
`start_flash` should launch `fake-registration-server` as one of its subprocesses, you should be fine. Check the web log to see the output. The `KeyboardInterrupt` events are from hitting `Ctrl-C` to terminate the process. We're interested to see if the device makes a request for `thirdparty.bin`.

ragenhe, 2019-03-15:
I cleared all logs so I knew what was from today's attempts.  I attempted to run ./fake-registration-server to no avail.  OK.  No new logs to delete.  I ran ./start-flash.  I did not connect another device first.  ./start-flash recognized the bulb online and appeared to be attempting to do something.  Previous attempts informed me it wouldn't be any different.  As I walked out of the room I accidentally hit the light switch.  $@#@^^%#@@......  Turned it back on and BAM..... I see the backup lines flying by.  About 15-20 seconds later the message to flash and have fun.  I flashed it. It rebooted and is now online with tasmota running.

ragenhe, 2019-03-15:
@kueblc - Colin. Before I attempt the second bulb from the pack, what can I do to help?  If it is real complex I need detailed instructions.  Or have I provided enough for you to pin point what's going on? If need be, I can hold off on the second bulb so I can test any update you may make?  Just let me know.

kueblc, 2019-03-15:
You'd need to be running the AP `vtrust-flash` before running `fake-registration-server`. The `start_flash` script takes care of this as well as creating logs from the output of the subprocesses, which is why you didn't get a log when you ran `fake-registration-server` by itself.

So just try running `start_flash` again for the second bulb and proceed as normal. Sounds like you've got the hang of it if you had flashed 8 Gosund plugs successfully. I can't say why this one device was more problematic, could be just a fluke.

ragenhe, 2019-03-15:
Yes.  I do have a grasp. However, others have reported very similar struggles.  I will proceed with second bulb and see how it goes.

SirRedZ, 2019-03-15:
> 
> 
> Yes. I do have a grasp. However, others have reported very similar struggles. I will proceed with second bulb and see how it goes.

The process is rather amazing. Some devices start faster than others. I have stopped the flashing process about 8 times during the process of reading, or paused, and I have easily recovered from each and every time.  Even when I rebooted the device (mid process) and had no lights and responses- it worked perfectly (and instantly as the device was already waiting for firmware).

ragenhe, 2019-03-16:
Second bulb flashed successfully.....
Steps:
1) Place device in pairing mode.
2) run ./start_flash (no other devices connected)
Pairing was FAST.... < 10 seconds.

closed 2019-03-16 by ragenhe

pilot1981, 2020-11-03:
so, now using this version of TUYA-CONVERT https://github.com/kueblc/tuya-convertit's possibile to flash via OTA also when installed TUYA firmware is v.1.0.5?

Tollbringer, 2020-11-03:
I have since learned each device's FIRMWARE revision is different. The only way to know is to try.