ct-Open-Source/tuya-convert #134: Tuya based Smart Power Monitoring Plug

vaidyasr, 2019-03-12:
I was able to flash successfully flash the Tuya based Smart Power Monitoring Plug with Tasmota and with various settings on the Config, I was able to control the Plug. But I can't find the correct power settings, so that the info on the main page gives correct voltage. Am from India and the Voltage is max 230v, whereas the page shows 263v. My Other Sonoff-POW shows correctly as 226v. Could someone help which one to select on GPIO4, GPIO5 and GPIO12?

SirRedZ, 2019-03-15:
I am guessing the exact model number, and a link to the place purchased would be helpful.

kueblc, 2019-03-15:
As this is a Sonoff-Tasmota configuration question, and not an issue with tuya-convert, you would be best served by asking there. Feel free to continue this conversation but I will close it as there is nothing for tuya-convert to do here.

closed 2019-03-15 by kueblc