ct-Open-Source/tuya-convert #136: TypeError: unsupported operand type(s) for %: 'bytes' and 'tuple'

DerHolgi, 2019-03-12:
Hi, I´m trying to flash the Jisusu SP111 powerplug but getting the above error.
Every thing looks good at the beginning, but when the connection is established, I can see the following output on **screen -r smarthack-web** and also at **tail -f screenlog.0**:

**Answer s.gw.upgrade**
^[[31m[E 190312 23:14:27 web:1670]^[[m^O Uncaught exception POST /gw.json?a=s.gw.upgrade&gwId=17014830b4e62d5e3efe&t=8&sign=2781130ec8e0356b58ced66cf1ce1b28 (10.42.42.28)
    HTTPServerRequest(protocol='http', host='10.42.42.1', method='POST', uri='/gw.json?a=s.gw.upgrade&gwId=17014830b4e62d5e3efe&t=8&sign=2781130ec8e0356b58ced66cf1ce1b28', version='HTTP/1.1', remote_ip='10.42.42.28')
    Traceback (most recent call last):
      File "/root/.local/lib/python3.4/site-packages/tornado/web.py", line 1590, in _execute
        result = method(*self.path_args, **self.path_kwargs)
      **_File "./fake-registration-server.py", line 86, in post_**
        answer = b'{"result":{"auto":3,"fileSize":"%d","etag":"0000000000","version":"9.0.0","url":"http://10.42.42.1/files/1.bin","md5":"%s"},"t":100,"e":false,"success":true}' % (file_len,file_md5.encode('utf-8'))
    **_TypeError: unsupported operand type(s) for %: 'bytes' and 'tuple'_**
^[[31m[E 190312 23:14:27 web:2162]^[[m^O 500 POST /gw.json?a=s.gw.upgrade&gwId=17014830b4e62d5e3efe&t=8&sign=2781130ec8e0356b58ced66cf1ce1b28 (10.42.42.28) 42.62ms
1552428867: New connection from 127.0.0.1 on port 1883.
1552428867: New client connected from 127.0.0.1 as P1 (c1, k60).
1552428867: Sending CONNACK to P1 (0)
1552428867: Received PUBLISH from P1 (d0, q0, r0, m0, 'smart/device/in/17014830b4e62d5e3efe', ... (147 bytes))
1552428867: Sending PUBLISH to 17014830b4e62d5e3efe (d0, q0, r0, m0, 'smart/device/in/17014830b4e62d5e3efe', ... (147 bytes))
1552428867: Socket error on client P1, disconnecting.

[screenlog.0.txt](https://github.com/ct-Open-Source/tuya-convert/files/2959024/screenlog.0.txt)

Any idea how to fix this?

kueblc, 2019-03-12:
This may be a Python version issue.

Try changing line 86 in `fake-registration-server.py` from

`answer = b'{"result":{"auto":3,"fileSize":"%d","etag":"0000000000","version":"9.0.0","url":"http://10.42.42.1/files/1.bin","md5":"%s"},"t":100,"e":false,"success":true}' % (file_len,file_md5.encode('utf-8'))`

to 

`answer = (u'{"result":{"auto":3,"fileSize":"%d","etag":"0000000000","version":"9.0.0","url":"http://10.42.42.1/files/1.bin","md5":"%s"},"t":100,"e":false,"success":true}' % (file_len,file_md5.encode('utf-8'))).encode('ascii')`

DerHolgi, 2019-03-13:
Yessss. This did the trick. 👍 
I now was able to flash the plug. 
Many thanks and good night.

closed 2019-03-13 by DerHolgi