ct-Open-Source/tuya-convert #138: Smartplug not detected: Waiting for the upgraded device to appear

hista, 2019-03-15:
Hi,
I've bought a DIGOO-NX-SP202. 
Just out of the box, no smartphone app installed, and in pairing mode (LED blinking fast)
I've followed the tuya-convert installation on my debian, but I'm stuck at the detecting device step ("Waiting for the upgraded device to appear"):
https://i.imgur.com/kiSjte2.png

Here are the logs:
- smarthack-wifi.log : https://framabin.org/p/?4915ddacd5b39d67#Hfu+1LrvEyuzilVOfoEr3v5lIc6jQ6FkdIEJjtpPadU=

- smarthack-web.log : https://framabin.org/p/?bdd94958bdecc0be#QkK5xhY1fHNMPxE0sjOY/ntMYXIBf/CxSHY0j28THqg=

- smarthack-mqtt.log : https://framabin.org/p/?ad3af8e1ab003fa3#Jj4tMXLvr4bWi2BAnUsYQZ/AouM4ijkTn/LHd1SsoCw=

- smarthack-smartconfig.log : https://framabin.org/p/?b9db7cba60acd6f4#pLX8N8HNF/jwswKgd8d0kOhN1KGfv8nHtdeKFLrRsfk=
 
Could you please help me?

onebyside, 2019-03-15:
Do we know that it has the correct chip?

https://i.imgur.com/kiSjte2.png

hista, 2019-03-15:
I've read feebacks of people who could flash it.
Plus, https://github.com/xoseperez/espurna/issues/1502

kueblc, 2019-03-15:
Log indicates you are missing the `@tuyapi/link` package from `npm` so smartconfig failed

hista, 2019-03-15:
Thanks, yes I've noticed that, but what surprised this package was not in the prereq.
Anyway, I've run npm i @tuyapi/link, logs are better now, but I'm stuck somewhere else:
https://i.imgur.com/PH6TD5g.png
I guess it's related to this error in smarthack-smartconfig.log:
"URI:/gw.json?a=tuya.device.dynamic.config.get&gwId=05560740b4e62d5a7dce&t=7&v=1.0&sign=898ede7c966f73f4333b6d459270e3c2
WARN: unknown request: tuya.device.dynamic.config.get (/gw.json?a=tuya.device.dynamic.config.get&gwId=05560740b4e62d5a7dce&t=7&v=1.0&sign=898ede7c966f73f4333b6d459270e3c2)
[32m[I 190315 14:48:30 web:2246][m 200 POST /gw.json?a=tuya.device.dynamic.config.get&gwId=05560740b4e62d5a7dce&t=7&v=1.0&sign=898ede7c966f73f4333b6d459270e3c2 (10.42.42.30) 0.35ms
Traceback (most recent call last):
  File "mq_pub_15.py", line 8, in <module>
    import sys, getopt, time, pyaes, base64
ModuleNotFoundError: No module named 'pyaes'"

So now:
- should I run sudo python3 -m pip install pyaes to install the missing module?
- may I stop the start_flash.sh script without bricking the device, to run it again after the module installation?

kueblc, 2019-03-15:
`install_prereq.sh` should have installed all of the dependencies. Did you run this script first?

kueblc, 2019-03-15:
You will not brick your device stopping the script unless you do so in the middle of a firmware upload

hista, 2019-03-15:
> `install_prereq.sh` should have installed all of the dependencies. Did you run this script first?

Yes I did, but there was indeed a problem in the prereq step, and I finally solved it!

The problem:

`curl is already the newest version (7.64.0-2).
dnsmasq is already the newest version (2.80-1).
haveged is already the newest version (1.9.1-6).
mosquitto is already the newest version (1.5.7-1).
nodejs is already the newest version (10.15.2~dfsg-1).
python3-pip is already the newest version (18.1-4).
python3-setuptools is already the newest version (40.8.0-1).
screen is already the newest version (4.6.2-3).
python3-wheel is already the newest version (0.32.3-2).
hostapd is already the newest version (2:2.7+git20190128+0c1e29f-2).
The following packages were automatically installed and are no longer required:
  debian-policy developers-reference-fr doc-debian-fr doc-linux-fr-text libclamav7 libhunspell-1.6-0 maint-guide-fr node-fstream
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
Traceback (most recent call last):
  File "/usr/local/bin/pip3", line 7, in <module>
    from pip import main
ImportError: cannot import name 'main' from 'pip' (/usr/lib/python3/dist-packages/pip/__init__.py)
`
I already fixed the syntax to install the python modules, but I didn't realize this error prevents the install_prereq.sh to continue to install the npm packages.

The solution:
In install_prereq.sh script, I replaced `sudo pip3 install paho-mqtt pyaes tornado` 
by 
`sudo python3 -m pip install paho-mqtt pyaes tornado`, and ran again the script.

It works now, thanks for your kind patience for the newbie I'm :-)

kueblc, 2019-03-15:
Excellent, glad you worked it out. If your issue is resolved you may close it now.

closed 2019-03-15 by hista