ct-Open-Source/tuya-convert #139: Error hostapd_free_hapd_data: Interface wlan0 wasn't started

rickitaly, 2019-03-15:
Hello, i am trying to use the script. I successfully ran the install_prerequisites and then i am now trying to start it but the AP does not appear and i get this wifi log.
I am running Kali 2019.1 on vmware and i attached to the vm a wifi usb dongle TP-Link TL-WN823N and also the host pc integrated wifi card.
In both cases this is the message.
Where am i wrong?
Thank you

`Backing up NetworkManager.cfg...
Restarting NetworkManager...
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
Error: Device for nexthop is not up.
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Could not set interface wlan0 flags (UP): Operation not possible due to RF-kill
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlan0 wasn't started`

AntSanVar, 2019-03-26:
Hello,
I fix it unblocking the wifi adapter:
`sudo rfkill unblock wifi`
You can check the blocking status with:
`rfkill list`

Best regards

rickitaly, 2019-03-27:
Thank you for your help, but still not solved here is the new message:

```
Backing up NetworkManager.cfg...
Restarting NetworkManager...
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlan0 wasn't started
```

kueblc, 2019-04-04:
Hi @rickitaly,

You can try the `create-ap` branch to see if it resolves your issue. Download it with `git clone -b create-ap https://github.com/kueblc/tuya-convert.git` and let us know if it works for you. FYI you'll need to run `install_prereq` again with the new version.

rickitaly, 2019-04-07:
Hi, thank you for your answer.
With this branch in the log i get the message
"ERROR: Your adapter does not support AP (master) mode"
i tried with both dongle TP-Link TL-WN823N and a RTL8187_Wireless attached to a kodi vmware machine.

kueblc, 2019-04-07:
Thanks for trying it out and getting back to us! Have you tried it outside of a VM?

mainframecn, 2019-10-01:
Same issue here, using kali on an old lenovo T500. 
"hostapd_free_hapd_data: Interface wlan0 wasn't started" and with the other branch "ERROR: Your adapter does not support AP (master) mode"

kueblc, 2019-10-01:
@mainframecn did you try the `create-ap` branch like suggested above?

kueblc, 2019-10-01:
If you have another machine on your network handling DNS queries you could redirect `*.tuya(us|eu|cn).com` requests to the machine running `fake-registration-server`.

Otherwise, without support for AP (master) mode, there is no way to create an isolated WiFi network for the Tuya device to connect to, and so nothing can be done here.

closed 2019-10-01 by kueblc