ct-Open-Source/tuya-convert #140: Tuya-Convert process disabled - Firmware v.1.0.5 MCU v1.0.5

SirRedZ, 2019-03-15:
I have confirmed with multiple tries on identical items (from the same batch).

My Teckin SB50 RGBW light flashes fine using Tuya-Convert out-of-the-box.

After a firmware update, v1.0.5 is downloaded, and the tuya-convert script is blocked. The light will logon to the server (flashing stops and light goes solid) - but it does not proceed further.

FYI- I hope this helps.

DavinKD, 2019-03-15:
Pretty disappointing.  It's strange that they derive no revenue from their app, yet they want to prevent this.  I would think the ability to flash them with Tasmota has helped sales of the products.

SirRedZ, 2019-03-15:
Well; I would assume the manufacturers license the Tuya platform understanding the value for data collection as part of their renumeration and income. When we flash the devices and "ET" never phones home - the platform does not offer the marketing data, error feedback, lifespan information the manufacturer's are likely counting on. It is our right to treat our products as we wish - I can only assume these facts are true.

I think it will likely be a matter of time before the same group addresses the situation - I hope.

For now: Do not connect brand new devices to the app; if you desire to flash them.

OR: If the device can be opened and soldered to: get a backup first.

So far, not helpful to me using a Teckin SB50 RGBW bulb.

kueblc, 2019-03-15:
Thank you for sharing @SirRedZ, very good to know. Do you have any firmware backups or network captures you could share?

SirRedZ, 2019-03-15:
No I do not. I took apart a factory unit but ended up destroying the bulb. The circuity inside is fine, but the "case" for the bulb (metal threaded part) was crushed and destroyed while I tried to remove it carefully.

After I got to the ESP- The solder points are not accessible without de-soldering the ESP chip from the power supply.

Having a backup would serve little purpose as it can not be easily applied for recovery purposes.

SirRedZ

kueblc, 2019-03-15:
Firmware backup would assist us in reproducing the issue and reverse engineering the changes. Thank you for your efforts, they are really appreciated.

If anyone happens to get a copy of the firmware before or after the upgrade, or any network captures of registering or upgrading a device affected by this, please share :+1:

ragenhe, 2019-03-15:
I just completed after much frustration one of two Merkury bulbs and it I assume was running the same as it's pair in the same box... both firmwares were 1.1.0

It wasn't a smooth finish.  I will be updating my open issue now.

ragenhe, 2019-03-15:
https://github.com/ct-Open-Source/tuya-convert/issues/133#issuecomment-473449002

kueblc, 2019-03-15:
@ragenhe if you believe your issue is related to your firmware, please try to capture network activity using `tcpdump` or Wireshark so we can see what's going on under the hood

ragenhe, 2019-03-15:
I don't think it is firmware at all.  I was sharing my progress, so we can rule out the firmware version as the culprit, or possible confirm it is firmware and a work around has been stumbled upon.

Just trying to help with this great project.

ragenhe, 2019-03-16:
Try this .... It worked for me on bulbs I thought weren't gonna work and seems very similar to your issue.
Steps:
1) Place device in pairing mode.
2) run ./start_flash (no other devices connected)
Pairing was FAST.... < 10 seconds.

sermayoral, 2019-03-25:
Hello everyone :-)

I would like to flash a BW-SHP2 plug using this method, but I would not like to break it. It has Firmware v.1.0.5 MCU v1.0.5.

According to this thread, tuya-convert does not work for these firmware versions. It is right?

In that case, is there anything I can do to help find a solution to this? I can try to get a backup of the current firmware.

Regards

kueblc, 2019-03-25:
Firmware versioning is per device, so while the Teckin SB50 v1.0.5 might not be working for some, it is not necessarily the same v1.0.5 as the BW-SHP2. Regardless, this process should not break the device if the firmware is too new, it just simply won't be able to proceed flashing.

If it does turn out that you cannot flash this device on that firmware, a backup of the current firmware would be extraordinarily helpful in reverse engineering the new changes so we might be able to work around them.

SirRedZ, 2019-03-25:
Hello!

if the plug can be opened, and backed up - that is the 1st place to start - before adding it to the Tuya App, and before converting it. I would suggest a few serial backups- then attempt to proceed with Tuya-convert process.

If it is going to work, it will connect and begin a readout/backup. Note this backup should be renamed or labelled as "tuya-backup' - indicating it is a 'modified' backup by Tuya. The serial backup is the only back valiud for full restoration after-the-fact.

If the Tuya process is going to work you'll notice it will give you dots... then connect and backup. At this point, you can decide to 'undo' and the plug will be restored. If you decide to flash any 3rd party bin, the device will "forever" be converted to Sonoff/other. The serial backup WILL allow you to restore the plug back to factory, for use in Tuya or Later convert. I have done this several times on 1 device, although I have a small sample size to compare with. As mentioned just now by **kueblc**; your data is invaluable for helping understand what has changed -.

-=SrZ=-

sermayoral, 2019-03-26:
Thanks for your answers guys!

@SirRedZ, @SirRedZ Unfortunately, as you had assumed, the plug was already connected to Tuya's application. It was there where I saw the firmware version I had. Is there any way to undo this?

If I understand correctly, making a backup "modified" by Tuya will not allow me to make a full restoration. But if the process fails, and I restore the backup "modified" by Tuya, what happens?

I'm waiting to receive a new SD, because the one I had in the RPI died. Can I do the process in an Ubuntu system or better wait to receive the SD and do it in the RPI?

SirRedZ, 2019-03-26:
I might suggest using the Proxmox-Tuya-convert listed here. Whiskerz007 has done a lot of work on this project, and it works in conjunction with the original tuya-convert scripting (which has recently been updated). Whiskerz007 will be posting an update shortly to correct for the changes. The Tuya-Convert team has done a great job with this project - and deserve full-credits.

Option #1 - proxmox (KVM) for running Virtual Machines and using Tuya-Convert LXC container (lightweight/fast) -- here -- How To Video ( https://www.youtube.com/watch?v=J8I6GPlDjbo )
1-liner script- https://github.com/whiskerz007/proxmox_tuya-convert_container

-- use this option if your machine supports Vt-d, Vt-x (virtualization technologies) and you desire to run multiple VM.s Its really cool to play with multiple machines on 1 physical machine. Kali-Linux, Ubuntu, Debian, Tuya-Convert - launch on demand)

Option #2 - Do a fresh Kali-Linux install, where the Tuya-Convert team has tested and approved working.

Currently, once a device has been upgraded to a "blocking" firmware, it won't let you convert anymore- Although I suspect the Tuya-Convert team is working on that!

Good luck!

-=SrZ=-

sermayoral, 2019-03-26:
I am using a live version of Kali Linux. Installing dependencies ...

In a while I tell you if everything went well

Thanks!

SirRedZ, 2019-03-26:
Excellent.

sermayoral, 2019-03-26:
Done without issues 😊.

Thanks for everything 

Program Version	6.5.0(release-basic)
Build Date & Time	2019-03-19T12:25:14
Core/SDK Version	2_3_0/1.5.3(aec24ac9)
Uptime	0T00:07:36
Flash write Count	33 at 0xF9000
Boot Count	7
Restart Reason	Software/System restart
Friendly Name 1	Sonoff

ESP Chip Id	16529822
Flash Chip Id	0x144051
Flash Size	1024kB
Program Flash Size	1024kB
Program Size	432kB
Free Program Space	568kB
Free Memory	17kB

SirRedZ, 2019-03-27:
Congrats @sermayoral !! Feels good don't it!?! Enjoy.

kll, 2019-03-29:
I've got brand new out of the box Sonoff basic and Sonoff S31 devices that have never been linked up or connected to anything that I've tried to flash with tuya-convert without success. I am new to this but I also got some Martin Jerry switches (MJ-S01 and MJ-SD01) that did flash fine so I'm reasonably sure my setup is correct and I understand the process. I'm not sure if it is related to the firmware or not.

I've opened up one of each of the Sonoff devices and grabbed a serial dump of the firmware. If anyone wants them I can post them somewhere. I went ahead and flashed those devices since I already had them opened up but I have others I can hold in reserve for testing if that is helpful.

kueblc, 2019-03-29:
tuya-convert does not work on Sonoff devices, only Tuya.

kll, 2019-03-29:
> tuya-convert does not work on Sonoff devices, only Tuya.

:man_facepalming: Sorry for being dumb. I've been drinking home automation info from a fire hose the past few days and got mixed up. I thought I saw somewhere where someone mentioned using tuya-convert on the Sonoff relays.  Move along, nothing to see here.

kueblc, 2019-03-29:
@kll I know the feeling, thanks for trying to help out anyway :+1:

SirRedZ, 2019-03-29:
@kueblc  I have a firmware backup (serial read) of a plug, but this plug does not have a newer version, and I can add to app, and remove, and still perform the tuya-convert process on demand. Woudl this backup be of any value to you?

Also, have you received other/enough backup files to to study and reverse engineer? or are you in need of more?

Thanks for such an awesome process!

-=SrZ=-

kueblc, 2019-03-30:
Thanks for asking @SirRedZ, more data would definitely be a big help. The most valuable would be firmware backups from devices that did not work with the OTA process, or devices that have an upgrade available but not yet downloaded.

garnern2, 2019-04-08:
I'm getting no errors in the log files, and the device does stop blinking on about the third dot when attempting to flash. It's a Teckin RGBW bulb with firmware 1.0.5

Any ideas?

kueblc, 2019-04-08:
@garnern2 According to @SirRedZ at the start of the issue, Teckin RGBW bulbs with 1.0.5 are patched so they are not compatible with the current version of `tuya-convert`. Your options are to flash it the old fashion way, over serial, or wait and see if we can develop a fix. If you manage to flash via serial, remember to take a backup and share this to help us develop a workaround.

uberfunk, 2019-04-16:
Anyone know if the CE Smart [dimmer switch](https://www.costco.ca/CE-Smart-Home-Wi-Fi-Smart-Dimmer-Light-Switch%2C-2-pack.product.100417574.html) and [smart outlet](https://www.costco.ca/CE-Smart-Home-Wi-Fi-Smart-Plug,-2-pack.product.100417575.html) got patched?  
I didn't know about tuya-convert when i updated the firmwares on them last week :/
But i checked and the dimmers are on MCU 1.0.0 and the outlets are on 0.1.2

liamrpowell, 2019-04-17:
I am also getting the same thing as @garnern2. No errors in the logs but my MCU is on 1.0.0 same as @uberfunk.
I just did another bulb and it worked (same MCU and brand) Looks like I might have to do the serial way. If so I will try to get a backup.

meingraham, 2019-05-03:
A bit #offtopic... This is re: OTA for non-Tuya ESP devices
@kll 

Anybody tried [SonOTA](https://github.com/mirko/SonOTA/wiki#known-working-configurations) on your newer Itead devices?

Tim-Dixon, 2019-05-17:
Can confirm that updating to 1.0.5 on the Merkury WW105 Wi-Fi Outlet stops the process after the device has connected to the Tuya-Convert Access Point. Stopping the script and re-linking the device via the Tuya app returns the plug to its factory functionality. Hopefully we can get past this! 

I see people talking about using serial to downgrade the firmware. How does one go about this? Is their a repository of default firmwares?

SirRedZ, 2019-05-17:
Most people with use Tasmota (at least intially) and then upload a custom EspHome yaml config or stay with Tasmota/Easy Esp/Espurna or other exisiting firmwares. With serial, you can go back and forth with no worries. Remember to gnd -> gpio0 (which may not have a breakout pin) and then plug in FTDI device to put the esp chip in flashing mode.

[-=SrZ=-]

sumasage, 2019-06-29:
Any progress on a work around?

phormix, 2019-07-05:
> Anyone know if the CE Smart [dimmer switch](https://www.costco.ca/CE-Smart-Home-Wi-Fi-Smart-Dimmer-Light-Switch%2C-2-pack.product.100417574.html) and [smart [...]

I can't speak to the dinners, but bought some of the smart-plugs last week (on sale 2 @ $19.99 whoop whoop) and managed to flash them successfully

I've also got some of the smart wall outlets, but I've yet to attempt flashing them yet. No idea on the dimmers but hopefully if the plugs still work then... (fingers crossed).

phormix, 2019-07-05:
Speaking of Costco, they've also got a 3pk of "Globe" Smart Bulbs ($29.99). Anyone checked to see if these run Tuya and/or are flashable?

haubke, 2019-09-15:
Anyone any luck with flashing newer tuya devices?

marc-gist, 2019-09-24:
there is a fork that works for some people, but i've yet to get it to work on the Teckin bulbs :(

phormix, 2019-09-26:
Fork, you say?

TThumb, 2019-09-26:
> Fork, you say?

Yes, there is a fork at:
https://github.com/M4dmartig4n/tuya-convert
(ignore the README.md, as that has not been updated).

The link is referenced in issue 273 of the main branch (https://github.com/ct-Open-Source/tuya-convert/issues/273).

I tried it on a Teckin SP10 (Tuya-app updated/locked a few weeks back).  Worked like a charm.
Please Note: It only works on the updated firmware (including those updated by signing into the Tuya app).  It does NOT work on the non-updated firmware.  
Between the main branch and this one, both scenarios are covered. The contributors will probably work to update the main branch to check for firmware version and then run the appropriate process.  For now, though, it's a very small workaround; try one, then try the other.

kueblc, 2019-09-26:
@phormix @TThumb You probably want the `new-api` branch at https://github.com/kueblc/tuya-convert. We're actively working on this to get it merged into master. It has support for both the older and newer firmware.

phormix, 2019-09-28:
Nice stuff! I just flashed three Globe RGBW Bulbs (Costco) that were
previously not working.

kueblc, 2019-09-28:
Awesome, glad it worked for you!

Happy to be closing this as resolved with #279

closed 2019-09-28 by kueblc

pilot1981, 2020-11-03:
so, now using this version of TUYA-CONVERT https://github.com/kueblc/tuya-convertit's possibile to flash via OTA also when installed TUYA firmware is v.1.0.5?

jbd7, 2023-02-19:
> so, now using this version of TUYA-CONVERT https://github.com/kueblc/tuya-convertit's possibile to flash via OTA also when installed TUYA firmware is v.1.0.5?

Today, I flashed OTA successfully with the Tuya-Convert v.2.4.5 of the `ct-Open-Source` repo, which includes the patch mentioned above.
I had a Tuya device with the firmware up to date (v.1.0.5) on a Anccy SP1 plug (same model as Gosund SP1). Used Linux Mint 20.3 with an Alfa Awus wifi adapter.

I spent hours of debugging my own mistake: the wifi adapter name needs to be updated in config.txt. I had tried with multiple names and it left some adapters unavailable but on the 10.42.42.0 network, which allowed other devices to connect to vtrust-flash but got them "honeypotted" away from tuya-convert. 

After resetting my adapters, the flashing was very quick. Tuya-Convert v 2.4.5 offers to flash Tasmota v8.1.0.2, which actually installs Tasmota v9.?, which can be upgraded via the Tasmota UI to the latest version (v.12.4.0 today).

sumasage, 2023-02-21:
> 

do you think it would work if using a different wifi adapter?  I'm planning to use raspberry pi to flash it.

ty-358, 2024-08-31:
Just looking to see if anyone has flashed SB60 Smartbulbs with firmware v.1.0.5 MCU v1.0.5. 

I keep receiving this error:

> "could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac"