ct-Open-Source/tuya-convert #141: Successfully flashed Koogeek KLSP1 with Tasmota

sgrimee, 2019-03-18:
I could not get Wifi Assist in the default image sonoff-classic to work after OTA upgrade, so I compiled a custom sonoff-knx with wifi pre-set and that one worked on a fresh KLSP1.

sgrimee, 2019-03-18:
[image]

SirRedZ, 2019-03-18:
Great job - Have you shared your template with others here?

https://blakadder.github.io/templates/

Thank you!

-=Srz=-

sgrimee, 2019-03-19:
> Great job - Have you shared your template with others here?

Not yet, but I will. I am using 6.4.1 release which does not have template support. Need to compile from the development branch for this. On my list...

kueblc, 2019-03-23:
Thanks for your contribution, feel free to add your findings to the wiki

closed 2019-03-23 by kueblc

bachoo786, 2019-11-24:
@sgrimee with that template the voltage is showing over 245v. thats way too high?

sgrimee, 2019-11-25:
Would the GPIO settings really have an influence on this? I am not sure how it works, but I read this device may need power/voltage calibration: https://blakadder.github.io/templates/koogeek_KLSP1.html
See link at the bottom of the page. I did not do it because I am not using the measuring capabilities.

FaserF, 2019-12-23:
@sgrimee thank you for sharing.
Maybe you have an idea for my problem? 
I have flashed the same device with Tasmota and used the template, but when I press the button on the webpage or the real button it doesn't turn on/off?
Is there anything I am missing?

sgrimee, 2019-12-23:
@FaserF remember the above is with a now old firmware compiled custom... with a newer firmware, have you tried the 'Sonoff Basic' template instead? If not your best guess is to try one gpio after the other for the relay setting. Or open the plug but I do not remember if the pin numbers are visible at all on that one.