ct-Open-Source/tuya-convert #143: Flash thirdparty firmware

File14, 2019-03-19:
Hello!

I have multiple smart plugs of the same type and one of them is running an older firmware version(1.0) and my others are running (1.2). I connected to an outlet with the version 1.2 and grabbed the bin file it created of that firmware(1MB). Am I now able to connect the outdated firmware outlet and flash this new version into it? If so, how? Please let me know the commands.

kueblc, 2019-03-19:
In short, no, that's not what this framework is designed for. The point of tuya-convert is to overwrite Tuya firmware with your own choice of custom firmware.

Technically you might be able to extract the userbin from 0x01000 or 0x81000 and it *might* work on another Tuya device, but why would you want to do that? If you want to upgrade your Tuya firmware to a later Tuya firmware, just use the vendor app.

File14, 2019-03-19:
Okay. Thank you a lot for the info!

Does this "vendor" app exist on iPhone(AppStore) or where can I find it?

kueblc, 2019-03-19:
The vendor app is whatever app you use to control the device. Each brand might have their own version, but they are all based off of the SmartLife app published by Tuya. Any Tuya based device should work with the SmartLife app AFAIK.

File14, 2019-03-19:
> The vendor app is whatever app you use to control the device. Each brand might have their own version, but they are all based off of the SmartLife app published by Tuya. Any Tuya based device should work with the SmartLife app AFAIK.

Yes. I use Smart Life to control the current outlets. These outlets do not have a brand as I purchased them on eBay and there is no company name printed on them. The instructions that are included tells me to use Smart Life and no other app.

It's not possible to update via Smart Life, right? What are my options? Should I try flashing the firmware that is a backup for the 1.2 device as an alternative firmware for the 1.0 device? Do you think it will work? The size of the bin is 1M.

kueblc, 2019-03-19:
You can absolutely use SmartLife to update your device firmware, if an update is available. I certainly do not recommend trying to force another device's stock firmware to a different device. At best, you'll get the same update as you would through SmartLife. At worst, you will brick your device.

File14, 2019-03-19:
> You can absolutely use SmartLife to update your device firmware, if an update is available. I certainly do not recommend trying to force another device's stock firmware to a different device. At best, you'll get the same update as you would through SmartLife. At worst, you will brick your device.

How do I do that? I can only find "control firmware" in the Smart Life app and that only tells me the current version of the device. No button to update.

Screenshot: https://i.gyazo.com/ab433ee095229857f00b900d454b351e.png

kueblc, 2019-03-19:
Sorry, I don't read the language. If you can't find an update option there may not be an update available. Is there a reason you need to run version 1.2?

The intention of this project is to install custom firmware, not to update vendor firmware. Custom firmware allows you to take security and privacy into your own hands, detach from cloud dependence, and implement functionality however you see fit, often above and beyond what stock firmware can do. If you are interested in these benefits, look into Sonoff-Tasmota or Espurna as two popular options. tuya-convert can help you install these onto your device.

For any other purpose I do not advise you to use this software. You will likely end up with a device you can't use with Tuya or at all.

File14, 2019-03-19:
> Sorry, I don't read the language. If you can't find an update option there may not be an update available. Is there a reason you need to run version 1.2?
>
> The intention of this project is to install [...]

The reason I want to upgrade is that I need to press the device one extra time to be able to toggle it in SmartLife. For the 1.2 firmware, you can toggle it without going "into" the device.

I have placed the firmware 1.2 in the files folder with the name firm.bin. Would you mind telling me the command to flash it from a file in files folder? And if it breaks it's fine as I have 10 of these outlets.

kueblc, 2019-03-20:
This is not a valid use case of tuya-convert and outside the scope of what I can support. If you decide to proceed you need to figure out yourself how to create a compatible firmware binary.

closed 2019-03-20 by kueblc