ct-Open-Source/tuya-convert #144: Teckin SP22 - Import tornado.web error

gmcbrien, 2019-03-19:
Hi, installation running all the way to when the Teckin SP22 connects to the new AP, and then nothing (just dots....)
Error seen in screenlog.0:

```bash
Traceback (most recent call last):
  File "./fake-registration-server.py", line 9, in <module>
    import tornado.web
  File "/usr/local/lib/python3.4/dist-packages/tornado/web.py", line 1655
    async def _execute(
            ^
```
Could it be related to my python version?

```bash
pi@raspberrypi:/usr/bin $ python --version
Python 2.7.9
```
BTW - no other log files produced.

Also, according to this post the device is supported :-) https://github.com/ct-Open-Source/tuya-convert/issues/58

Would anyone have any advice?

gmcbrien, 2019-03-19:
Short update - I pointed my /usr/bin/python to python3.4 --> same error 

```bash
root@raspberrypi:/home/pi/tuya-convert/scripts# python --version
Python 3.4.2
```

kueblc, 2019-03-20:
Is there any more to the log?

ragon000, 2019-03-20:
I managed to successfully flash the Teckin SP22 without any problems on python 3.7

you need at least python 3.5 to use `async def`, so you should probably update your python

erichegit, 2019-03-20:
I have a similar situation
Teckin SP22 with Raspi 3B+ and the recomanded Kali Image.

./install_prereq.sh took hours. Now I can not connect to vtrust-flash.
not from the Mobilephone nor from my PC via WLAN :(

smarthack-wifi.log
```
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
^Cwlan0: interface state ENABLED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Failed to remove interface (ifidx=4)
Backing up NetworkManager.cfg...
Restarting NetworkManager...
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:45:55:49 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
```

gmcbrien, 2019-03-20:
I'll upgrade my Python version later and get back to you, thanks!

@erichegit Just an idea - make sure you kill the process wpa_supplicant before kicking off the script. When I did this, it shows the vtrust-flash AP.

erichegit, 2019-03-20:
sadly on Kali Linux I see the AP but can not connect,...
On Raspbian on my 2nd Pi I linked python3.5 but also face the Error message like you

```
^CTraceback (most recent call last):
  File "./fake-registration-server.py", line 132, in <module>
    main()
  File "./fake-registration-server.py", line 128, in main
    tornado.ioloop.IOLoop.current().start()
  File "/usr/local/lib/python3.5/dist-packages/tornado/platform/asyncio.py", line 148, in start
    self.asyncio_loop.run_forever()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 421, in run_forever
    self._run_once()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 1388, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.5/selectors.py", line 445, in select
    fd_event_list = self._epoll.poll(timeout, max_ev)
KeyboardInterrupt
Listening on port 80
```

gmcbrien, 2019-03-20:
Upgraded to Python 3.7.2 as far as I can tell and the problem persists. The only log I have is the screenlog.0. No other log files exist. Here is the complete screenlog:

```
root@raspberrypi:/home/pi/tuya-convert/scripts# tail -f screenlog.0
^C^Cwlan0: AP-STA-DISCONNECTED dc:4f:22:XX:XX:XX
wlan0: AP-STA-DISCONNECTED 6c:c7:ec:XX:XX:XX
^C1553105697: mosquitto version 1.3.4 terminating
Traceback (most recent call last):
  File "./fake-registration-server.py", line 9, in <module>
    import tornado.web
  File "/usr/local/lib/python3.4/dist-packages/tornado/web.py", line 1655
    async def _execute(
            ^
SyntaxError: invalid syntax
Backing up NetworkManager.cfg...
Restarting NetworkManager...
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
RTNETLINK answers: File exists
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:XX:XX:XX and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: STA dc:4f:22:XX:XX:XX IEEE 802.11: associated
wlan0: AP-STA-CONNECTED dc:4f:22:XX:XX:XX
wlan0: STA dc:4f:22:XX:XX:XX RADIUS: starting accounting session 5C928329-00000000
wlan0: STA dc:4f:22:XX:XX:XX WPA: pairwise key handshake completed (RSN)
wlan0: STA 6c:c7:ec:XX:XX:XX IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 6c:c7:ec:XX:XX:XX
wlan0: STA 6c:c7:ec:XX:XX:XX RADIUS: starting accounting session 5C928329-00000001
wlan0: STA 6c:c7:ec:XX:XX:XX WPA: pairwise key handshake completed (RSN)
1553105705: mosquitto version 1.3.4 (build date 2018-09-28 22:21:32+0000) starting
1553105705: Using default config.
1553105705: Opening ipv4 listen socket on port 1883.
1553105705: Opening ipv6 listen socket on port 1883.
wlan0: STA dc:4f:22:XX:XX:XX IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED dc:4f:22:XX:XX:XX
wlan0: STA dc:4f:22:XX:XX:XX IEEE 802.11: disassociated
wlan0: STA dc:4f:22:XX:XX:XX IEEE 802.11: disassociated
wlan0: STA dc:4f:22:XX:XX:XX IEEE 802.11: disassociated
wlan0: STA dc:4f:22:XX:XX:XX IEEE 802.11: disassociated
wlan0: STA dc:4f:22:XX:XX:XX IEEE 802.11: disassociated
wlan0: STA dc:4f:22:XX:XX:XX IEEE 802.11: disassociated
wlan0: STA dc:4f:22:XX:XX:XX IEEE 802.11: disassociated
wlan0: STA dc:4f:22:XX:XX:XX IEEE 802.11: disassociated
wlan0: STA dc:4f:22:XX:XX:XX IEEE 802.11: disassociated
wlan0: STA dc:4f:22:XX:XX:XX IEEE 802.11: disassociated
Put Device in Learn Mode!, sending Smartlink Packets now
Smartlink in progress
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
Sending Packets and wait 60 seconds.
wlan0: STA dc:4f:22:XX:XX:XX IEEE 802.11: associated
wlan0: AP-STA-CONNECTED dc:4f:22:XX:XX:XX
wlan0: STA dc:4f:22:XX:XX:XX RADIUS: starting accounting session 5C928329-00000002
wlan0: STA dc:4f:22:XX:XX:XX WPA: pairwise key handshake completed (RSN)
```
The blue light on the plug stops flashing, but then the script just shows me dots.....

erichegit, 2019-03-20:
exact the same behavior like for me - blue light is blinking and when hitting ENTER - it stops after a few seconds,...
Could it be a problem of a new Teckin update mechanism=?

kueblc, 2019-03-20:
@erichegit I'd advise you to open a separate issue for your own problem, from what you posted this issue is not related at all

> I have a similar situation
> Teckin SP22 with Raspi 3B+ and the recomanded Kali Image.
> 
> ./install_prereq.sh took hours. Now I can not connect to vtrust-flash.
> not from the Mobilephone nor from my PC via WLAN :(

`install_prereq` shouldn't take more than a minute or two, something is seriously wrong there

> smarthack-wifi.log
> 
> ```
> wlan0: interface state UNINITIALIZED->ENABLED
> wlan0: AP-ENABLED 
> ^Cwlan0: interface state ENABLED->DISABLED
> wlan0: AP-DISABLED 
> wlan0: CTRL-EVENT-TERMINATING 
> nl80211: deinit ifname=wlan0 disabled_11b_rates=0
> Failed to remove interface (ifidx=4)
> Backing up NetworkManager.cfg...
> Restarting NetworkManager...
> Backing up /etc/dnsmasq.conf...
> Writing dnsmasq config file...
> Creating new /etc/dnsmasq.conf...
> Writing hostapd config file...
> Configuring AP interface...
> Applying iptables rules...
> Starting DNSMASQ server...
> Starting AP on wlan0 in screen terminal...
> Configuration file: /etc/hostapd/hostapd.conf
> wlan0: Could not connect to kernel driver
> Using interface wlan0 with hwaddr b8:27:xxxxx and ssid "vtrust-flash"
> wlan0: interface state UNINITIALIZED->ENABLED
> wlan0: AP-ENABLED 
> ```

I don't see anything wrong with the logs here

> sadly on Kali Linux I see the AP but can not connect,...
> On Raspbian on my 2nd Pi I linked python3.5 but also face the Error message like you
>
> ```
> ^CTraceback (most recent call last):
>   File [...]

This is not the same error, it says you hit Ctrl-C to stop the script

> exact the same behavior like for me - blue light is blinking and when hitting ENTER - it stops after a few seconds,...

This is exactly what is supposed to happen; the device stops blinking when it is paired

kueblc, 2019-03-20:
@gmcbrien do `python --version`, are you sure your recently installed version is being called?

gmcbrien, 2019-03-21:
Yes, I've done a python --version and can assure you it's 3.7.2. I also went to the python CLI and also tried an import of tornado.web and I get the same error.
So, later I'll uninstall all Python versions from my Pi and start with a fresh installation - something might still be messed up somewhere with the Python installation.
I'll report back.

gmcbrien, 2019-03-22:
Seems my problem was related to Python versions, in a way. I had linked /usr/bin/python to /usr/bin/python/3.5. Yes, a python --version shows version 3.5.7, but I was getting all sorts of crazy python errors. Long story short, I reinstalled my Pi and ran into the same problem.... the only think I changed was the soft link as I mentioned above. When I reversed that (i.e. back to the default setting), the weird Python errors disappeared.
Now, I am still having problems, but will open a new issue as it's not directly related to this Python version mixup.

kueblc, 2019-03-23:
Glad you were able to sort out your Python issues, will keep that in mind as we continue to improve.

closed 2019-03-23 by kueblc