ct-Open-Source/tuya-convert #145: i am trying this in Debian and i get this error Cannot find module 'debug'

redwoodimage, 2019-03-20:
tuya-convert# nodejs -v
v10.15.3

tuya-convert# nodejs -v
v10.15.3

i tried installing debug via pip3 install debug

internal/modules/cjs/loader.js:584
    throw err;
    ^

Error: Cannot find module 'debug'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:582:15)
    at Function.Module._load (internal/modules/cjs/loader.js:508:25)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
    at Object.<anonymous> (/home/dave/tuya-convert/scripts/smartconfig/smartcon$
    at Module._compile (internal/modules/cjs/loader.js:701:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:712:10)
    at Module.load (internal/modules/cjs/loader.js:600:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:539:12)
    at Function.Module._load (internal/modules/cjs/loader.js:531:3)

kueblc, 2019-03-20:
`debug` is an `npm` module not `pip`; try `npm i debug`

redwoodimage, 2019-03-21:
Thanks I got it work my light picked 10.42.42.31 so I had to edit script to change ip.

closed 2019-03-21 by redwoodimage