ct-Open-Source/tuya-convert #149: Successfully flashed several Jinvoo Wall Light, Curtain switches

dominikandreas, 2019-03-23:
Tested on several wall switches (1-3 Gang) as well as curtain switches (~same as 3-Gang wall switch):
- https://www.amazon.de/Jinvoo-Lichtschalter-Fernbedienung-Sprachsteuerung-Timing-Funktion
- https://www.amazon.de/Jinvoo-Controller-Compatibile-Neutralleiter-Unterstützung/
- https://www.alibaba.com/product-detail/Jinvoo-Wireless-Remote-Controlled-Works-with_60717264270.html

Sonoff Tasmota [Templates](https://github.com/arendst/Sonoff-Tasmota/wiki/Templates):

- 1-3 Gang Wall Switches:
  ```json
   {"NAME":"Jinvoo T1","GPIO":[52,0,0,0,0,0,0,0,17,21,0,0,0],"FLAG":1,"BASE":18}
   {"NAME":"Jinvoo T2","GPIO":[52,0,0,18,22,0,0,0,17,21,0,0,0],"FLAG":1,"BASE":18}
   {"NAME":"Jinvoo T3","GPIO":[52,0,0,18,22,19,0,0,17,21,0,23,0],"FLAG":1,"BASE":18}
  ```
- Curtain Switch:
  ```json
   {"NAME":"Jinvoo Curtain","GPIO":[52,0,0,18,22,19,0,0,17,21,0,23,0],"FLAG":1,"BASE":18}
  ```
For the Curtain Switch, the following options need to be used:
```
SetOption1 1   # restrict button-multipress to single, double and hold actions
SetOption14 1  # exclusive buttons (important!)
LedPower 0     # disable use of LED as much as possible (optional)
SetOption31 1  # diable wifi led (optional)
PulseTime1 130 # set relay 1 activation to 30 seconds (+100 for 1sec interval)
PulseTime2 1   # deactivate button 2 immediately after activation (disables relays)
Pulsetime3 130 # set relay 3 activation 30 seconds (+100 for 1sec interval)
PowerRetain 0  # Don't retain states
SaveOption 1   # enable autosave
```

kueblc, 2019-03-23:
Thanks for your report, please add it to the wiki for visibility

closed 2019-03-23 by kueblc

Luciano3272, 2019-11-11:
Hi dominikandreas,
Thanks for your job. 
I've a problem, I followed your template:
{"NAME":"Jinvoo T3","GPIO":[52,0,0,18,22,19,0,0,17,21,0,23,0],"FLAG":1,"BASE":18}
because I'd a Jinvoo smart switch 3 gang. 
All ok , 3 relè works, but third led is always off .....
Thanks in advance
Lu

dominikandreas, 2019-11-11:
Hi @Luciano3272, not sure how to help, I've never had that problem. For me the third led has always worked

Luciano3272, 2019-11-11:
Ok, like always I'm unlucky. Last question, Does your test work if you don't connect any light bulbs to the switch?
Always thanks
Lu

dominikandreas, 2019-11-12:
I'm actually not using a 3-Gang light switch but 3-Gang curtain switch. I read that it's the same hardware, but maybe there's a difference. This means that you need to adapt the relay and/or touch gpio numbers in the template. If you look at the difference between 2-Gang and 3-Gang in the template, you can see which ones need to be adapted. I'm not sure how you find out the correct settings, I originally figured them out by googling for other successful hacks and extracting the necessary information from them. I think this was one of the sources: https://forum.iobroker.net/topic/9886/tuya-jinvoo-unterputz-wandschalter/18
Maybe there's a better and easier was to figure out the right settings, but I'm not an expert on this

zolabus, 2019-11-28:
Hi all I have a one gang jinvoo and I'd like to disable the wifi light, how can I add the option? And where do I add the template?
Sorry but I'm a new user...
Thanks

dominikandreas, 2019-11-28:
Heres the documentation for importing templates: https://github.com/arendst/Tasmota/wiki/Templates#importing-templates

For disabling the led light during WiFi there's some command you can execute but I forgot the details. Should be easy to figure out with Google though

zolabus, 2019-11-28:
I found nothing on google do you mean I can disable it without flashing it? i think I have the new firmware because the trick does not work for me....

dominikandreas, 2019-11-29:
Did you already flash the sonoff tasmota  firmware? Without it you can't do any of this

zolabus, 2019-11-29:
No i didn't I tried yesterday but searching the switch I have only 
...............................................................................
Do i have the new firmwaer or I do some mistake?
Thanks

fastcolors, 2020-01-05:
LedPower 0 to disable use of LED as much as possible (optional)

For the Curtain Switch, i find this to be a better solution (blue led not turning on at press time and immediate response instead of "on release" )

{"NAME":"Jinvoo Curtain","GPIO":[52,0,0,10,22,11,0,0,9,21,0,23,0],"FLAG":1,"BASE":18}

SwitchMode 3  # pushbutton mode

Interlock 1   # restrict button-multipress to single, double and hold actions
SetOption14 1  # exclusive buttons (important!)
LedPower 0     # disable use of LED as much as possible (optional)
SetOption31 1  # diable wifi led (optional)
PulseTime1 130 # set relay 1 activation to 30 seconds (+100 for 1sec interval)
PulseTime2 1   # deactivate button 2 immediately after activation (disables relays)
Pulsetime3 130 # set relay 3 activation 30 seconds (+100 for 1sec interval)
PowerRetain 0  # Don't retain states
SaveOption 1   # enable autosave

MarksLiveMusic, 2020-01-13:
Hello guys. I was trying to figure out why SetOption14 1 didn't work (console error).

I found that it has been replaced by "Interlock" features:
https://tasmota.github.io/docs/#/Commands?id=interlock

Hope this can help!

Nin-D-Tendo, 2020-03-11:
Hey it seems like on mine relay 2 and 3 are swapped (with respective buttons).
I am trying to integrate them into Home Assistant, but can't get ShutterStop to work. Shutter up and down work well, also the state is reported 'unknown' in HA.
Hope someone has some pointers & thanks in advance.

HA config.yaml

`  - platform: mqtt
    name: "Bedroom Blinds"
    availability_topic: "tele/bedroom_blinds/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"
    position_topic: stat/bedroom_blinds/Shutter
    position_open: 100
    position_closed: 0
    set_position_topic: "cmnd/bedroom_blinds/ShutterPosition1"
    command_topic: "cmnd/bedroom_blinds/Backlog"
    payload_open: "ShutterOpen"
    payload_close: "ShutterClose"
    payload_stop: "ShutterStop"
    retain: false
    optimistic: false
    qos: 1`

fastcolors, 2020-03-11:
@Nin-D-Tendo 

try this

{"NAME":"Jinvoo Curtain","GPIO":[52,0,0,11,23,10,0,0,9,21,0,22,0],"FLAG":1,"BASE":18}

Nin-D-Tendo, 2020-03-11:
@fastcolors 
Hey thanks,
I already figured how to swap IO pins.
I can Open and Close the curtains via HA but can't get `ShutterStop1` to work. I also tried to use `ShutterStop1 `in console on tasmota admin interface with no results.
At the moment I have `payload_stop: "Power3 on"` which is not a very elegant solution.

fastcolors, 2020-03-12:
do shutteropen1 and shutterclose1 work?

Nin-D-Tendo, 2020-03-12:
@fastcolors 
yes shutteropen1 and shutterclose1 work. I just can't get the shutter to stop on issuing shutterstop1.

Right now I have PulseTime1 and PulseTime3 set to 130 (as far as I understood, after pressing button, the button stays active for the maximum of 130s) and PulseTime2 ( my stop button) set to 1.

Thanks for clarifying the Power Retain!

fastcolors, 2020-03-12:
> Right now I have PulseTime1 and PulseTime3 set to 130 (as far as I understood, after pressing button, the button stays active for the maximum of 130s) and PulseTime2 ( my stop button) set to 1.

set them to 0

only pulsetime3 (the stop button) is to be set to 1

fastcolors, 2020-03-12:
let me clarify something:

the above setup is correct if you use jinvoo curtain module as 3 switches.

if you activate SETOPTION80 and use the module as a proper shutter control in tasmota THEN

pulsetime needs to be 0 for up and down switches
and 1 for stop switch

up to you

morettty, 2020-06-26:
I use the same template and settings and it doesn't work properly for me. Neither with the GUI nor with the physical buttons the relays are disconnected when reaching the limits. When moving the curtain line in gui it only goes up, if I hit it down it does not go down. Can you help me?

prismplex, 2021-02-20:
I got it working with the following template, thank you @fastcolors 
`{"NAME":"Jinvoo Curtain","GPIO":[52,0,0,11,23,10,0,0,9,21,0,22,0],"FLAG":1,"BASE":18}`
Console: `Backlog SetOption80 1;SetOption31 1;LedPower 0;ledstate 0;timezone 99;setoption13 1;PowerRetain 0;PowerOnState 0;Interlock 1;SetOption1 1;Interlock 1`
After that I set the Shutter Durations:
`ShutterCloseDuration 23` for 23 sec
`ShutterOpenDuration 27` for 27 sec

The only problem I am experiencing is that the "close button" must be pressed the whole time otherwise the closing stops immediately (the button does not stay active). The open button works perfectly, pause too. Using template `SwitchMode 3`, already tried all other modes but did not succeed. Did somebody experience the same issue? How did you solve it? Thank you!

prismplex, 2021-02-20:
Got it working with the following template
`{"NAME":"Jinvoo Curtain","GPIO":[288,0,0,34,226,33,0,0,32,224,0,225,0,4704],"FLAG":0,"BASE":18}`
source: [](https://gitee.com/mirrors/Tasmota/blob/master/TEMPLATES.md)

Buttons work now as expected, blue led only lights up every time for a second