ct-Open-Source/tuya-convert #150: Teckin SP22 - nothing happens

gmcbrien, 2019-03-23:
Nothing happens....

I connect my phone to the newly created vtrust-flash AP, I set the plug to start flashing, I press Enter.... and all I get are dots. Below arte the log files.
NOTE, all logs start with a ^C  --> I am NOT touching the keyboard, so I'm not sure this is coming from.

This is a freshly installed Raspberry Pi 3B.

Anybody any ideas where I might be going wrong?

```bash
root@raspberrypi:/home/pi/tuya-convert# python -V
Python 2.7.13
root@raspberrypi:/home/pi/tuya-convert# python3 -V
Python 3.5.3
```
smarthack-mqtt.log

```bash
root@raspberrypi:/home/pi/tuya-convert/scripts# cat smarthack-mqtt.log
^C1553327860: mosquitto version 1.4.10 terminating
1553327865: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1553327865: Using default config.
1553327865: Opening ipv4 listen socket on port 1883.
1553327865: Opening ipv6 listen socket on port 1883.
```

smarthack-smartconfig.log

```bash
module.js:549
    throw err;
    ^

Error: Cannot find module 'debug'
    at Function.Module._resolveFilename (module.js:547:15)
    at Function.Module._load (module.js:474:25)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/home/pi/tuya-convert/scripts/smartconfig/smartconfig.js:16:15)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
```

smarthack-web.log

```bash
root@raspberrypi:/home/pi/tuya-convert/scripts# cat smarthack-web.log
^CTraceback (most recent call last):
  File "./fake-registration-server.py", line 132, in <module>
    main()
  File "./fake-registration-server.py", line 128, in main
    tornado.ioloop.IOLoop.current().start()
  File "/usr/local/lib/python3.5/dist-packages/tornado/platform/asyncio.py", line 148, in start
    self.asyncio_loop.run_forever()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 421, in run_forever
    self._run_once()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 1388, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.5/selectors.py", line 445, in select
    fd_event_list = self._epoll.poll(timeout, max_ev)
KeyboardInterrupt
Listening on port 80
[I 190323 07:57:55 web:2246]☼ 200 GET / (10.42.42.36) 7.76ms
```
smarthack-wifi.log
```bash
wlan0: STA 6c:c7:ec:XX:XX:XX IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED 6c:c7:ec:XX:XX:XX
^Cwlan0: interface state ENABLED->DISABLED
wlan0: AP-DISABLED
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
RTNETLINK answers: File exists
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:26:1e:fa and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: STA 6c:c7:ec:XX:XX:XX IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 6c:c7:ec:XX:XX:XX
wlan0: STA 6c:c7:ec:XX:XX:XX RADIUS: starting accounting session 881F7A79914AF2D6
wlan0: STA 6c:c7:ec:XX:XX:XX WPA: pairwise key handshake completed (RSN)
```

dave-castle, 2019-03-23:
same for me

erichegit, 2019-03-23:
same for me - blue blinking LED - after hitting "Enter" after 2-3 seconds the LED turns off followed by many dots in the console without doing anything.

gmcbrien, 2019-03-23:
@erichegit As mentioned in another post, the LED is supposed to stop flashing after a couple of seconds - to me it seems your device is making a successful connection to the AP - so it sounds to me like you are one step ahead of me :-)

kueblc, 2019-03-23:
@gmcbrien Try pulling the latest version, I've merged the Python based SmartConfig

@dave-castle Need more details, pull the latest version, try again, and open an issue with your logs

@erichegit Yes, @gmcbrien is correct, the LED should stop blinking when it pairs successfully. Pull, test again, and post logs

gmcbrien, 2019-03-23:
Very good @kueblc - I pulled the latest build and it works perfectly! Many thanks!

kueblc, 2019-03-23:
Very happy to hear it, thanks for the feedback! Others, please open a new issue for any other problems.

closed 2019-03-23 by kueblc

dave-castle, 2019-03-23:
after pulling latest build it worked for me immediately. Great Job, thanks!

kueblc, 2019-03-23:
Awesome, thanks for the feedback

dave-castle, 2019-03-23:
maybe dumb question but how to go on after successful flashing, i compiled ESPHome bin, uploaded to my rasp into the /tuya-converter/files/ dir, but if i typed curl http://10.42.42.42/flashURL?url=http://10.42.42.1/files/temp_flash.bin, I got this

pi@raspberrypi:~/tuya-convert $ curl http://10.42.42.42/flashURL?url=http://10.42.42.1/files/temp_flash.bin
File Not Found

kueblc, 2019-03-23:
My bad, I forgot to update the README. I'll update this soon. Try http://10.42.42.42/flash3?url=http://10.42.42.1/files/temp_flash.bin instead.

erichegit, 2019-03-24:
thanks - I can confirm flashing with a Raspbian Stretch image is now working :+1: 
BUT - I miss a step.
Even when I use your link "temp_flash.bin" I only get "Hello World" when going to 10.42.42.1 or the info page on 10.42.42.42.

What to do next=?
With hardware flashing I was used to provide SSID + pwd to access the device after flashing directly via webbrowser - please can you update this steps in the readme?

kueblc, 2019-03-24:
This is already in the README
> Currently a Sonoff-Tasmota v6.5.0 basic build is already included. It will open an WiFi access point named `sonoff-XXX`.

> @kueblc - how can the vtrust-flash WLAN be turned off after flashing?
> Restarting the Pi is one option, but I would like a command to turn off the hotspot.
That's what `stop_flash.sh` is for

bseibold, 2019-03-24:
I had the same problem: smartconfig failed, the LED kept blinking. In the end I had to use the offical Tuya android app to get the plug to connect to the `vtrush-flash` wifi. I was using 99a317efd72b51566ccf774efb645b843d7ed8cb.

I noticed only one difference between the packets `smartconfig.py` sends and the original Tuya app: The app uses `EUSWIlyA7wC5iD` as the token_group instead of `US000000000101`.

kueblc, 2019-03-24:
The app sends a unique token and secret for each pairing attempt. It doesn't really matter what we use as we are not actually registering the device to the cloud and instead running our own fake registration server, which couldn't care less what the token, secret, and region are.

Did you connect a device to the `vtrust-flash` AP while running `smartconfig.py`?

bseibold, 2019-03-25:
I first ran `start_flash.sh` and followed the instructions, but the plug did not connect. Manually calling `main.py` again didn't help either. Only when using the Tuya app I was able to get the plug to connect.

If required, I can provide a packet dump of what the Tuya app sends, as well as a flash dump of the original firmware.