ct-Open-Source/tuya-convert #152: tuya-convert on ubuntu18.04

FransM, 2019-03-23:
Hi all,

SInce I had no PI handy, I decided to do this under Ubuntu 18.04.
In order to make this work I had do do three things:
- install a recent version of nodejs (somehow I had 5.0.0 or so. I uninstalled and installed 10.15.3 from the PPA. You'll need a nodejs version > 7.something
- kill aolserver4-nsd. Somehow this was running on my system although it didn't seem to serve content, but it was listening to port 80. Maybe you do not have this.
- for dns resolving 18.04 uses systemd-resolved. In the script when stopping the dnsmask server I added
    sudo service systemd-resolved stop

After that I could flash the script. If things do not work as expected be sure to examine the log files in the scripts folder. Not all errors are caught properly (e.g. if the upgrade web server cannot start because port 80 is taken or if dnsmasq cannot be restarted because port 53 is in use.

BTW: probably it is a bit simpler to just add the addresses in /etc/hosts instead of modifying the dnsmasq conf file

kueblc, 2019-03-23:
Thanks for the input @FransM!

I should note that `nodejs` is no longer a dependency of this project.

I'm not sure what `aolserver4-nsd` is, that's interesting. I'm planning to implement a generic check for processes running on port 80 to handle these cases.

I'm also planning on replacing the AP script with a third party script that's well maintained, and should handle most if not all of the edge cases folks are running into. This will also take care of DNS resolution without having to muck around with the hosts file.

kueblc, 2019-04-04:
Hi @FransM,

Would you try the `create-ap` branch to see if this resolves your issues?

`git clone -b create-ap https://github.com/kueblc/tuya-convert.git`

FransM, 2019-04-08:
Hi Colin,
I'd love to but recently I bricked my switch. I need to reflash it using serial but did not get to it yet. May take a few weeks before I get to this again.
Kind regards, Frans

kueblc, 2019-04-08:
No problem, thanks anyway

kueblc, 2019-10-04:
Closing as resolved by #225

closed 2019-10-04 by kueblc