ct-Open-Source/tuya-convert #154: problem flashing CE smart home outlet from costco

sbeland2, 2019-03-24:
CE smart home outlet LA-WF3 from costco.
Using raspberry pi 3 b+ with ethernet and wifi connection based on your instructions from this link:
https://github.com/ct-Open-Source/tuya-convert 
under Procedure title, everything installed without any errors.

exec start_flash.sh, type yes, enter, then hold the button on the outlet so the blue led blink fast.
able to connect to ap with my phone.

monitoring connection tell me: phone at 10.42.42.19 and outlet at 10.42.42.17

once typed enter, the script will never be able to ping something at 10.42.42.42 where i can see ....... forever.

the outlet will have a blue led not blinking anymore

here's my logs:

cat smarthack-mqtt.log
1553444228: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1553444228: Using default config.
1553444228: Opening ipv4 listen socket on port 1883.
1553444228: Opening ipv6 listen socket on port 1883.
^C1553444267: mosquitto version 1.4.10 terminating
1553444270: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1553444270: Using default config.
1553444270: Opening ipv4 listen socket on port 1883.
1553444270: Opening ipv6 listen socket on port 1883.
^C1553448091: mosquitto version 1.4.10 terminating
1553448184: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1553448184: Using default config.
1553448184: Opening ipv4 listen socket on port 1883.
1553448184: Opening ipv6 listen socket on port 1883.
^C1553451974: mosquitto version 1.4.10 terminating
1553451978: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1553451978: Using default config.
1553451978: Opening ipv4 listen socket on port 1883.
1553451978: Opening ipv6 listen socket on port 1883.
^C1553452239: mosquitto version 1.4.10 terminating
1553452244: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1553452244: Using default config.
1553452244: Opening ipv4 listen socket on port 1883.
1553452244: Opening ipv6 listen socket on port 1883.

 cat smarthack-smartconfig.log
Put Device in Learn Mode!, sending Smartlink Packets now
Smartlink in progress
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
Sending Packets and wait 60 seconds.
^CPut Device in Learn Mode!, sending Smartlink Packets now
Smartlink in progress
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
Sending Packets and wait 60 seconds.
^CPut Device in Learn Mode!, sending Smartlink Packets now
Smartlink in progress
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
Sending Packets and wait 60 seconds.
^CPut Device in Learn Mode!, sending Smartlink Packets now
Smartlink in progress
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
Sending Packets and wait 60 seconds.

cat smarthack-web.log
Traceback (most recent call last):
  File "./fake-registration-server.py", line 132, in <module>
    main()
  File "./fake-registration-server.py", line 126, in main
    app.listen(options.port)
[... 281 lines cut ...]
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:47:c6:20 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: STA dc:4f:22:ee:bb:df IEEE 802.11: associated
wlan0: AP-STA-CONNECTED dc:4f:22:ee:bb:df
wlan0: STA dc:4f:22:ee:bb:df RADIUS: starting accounting session B6680A539F6A62ED
wlan0: STA dc:4f:22:ee:bb:df WPA: pairwise key handshake completed (RSN)
wlan0: STA dc:4f:22:ee:bb:df IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED dc:4f:22:ee:bb:df
wlan0: STA dc:4f:22:ee:bb:df IEEE 802.11: disassociated
wlan0: STA dc:4f:22:ee:bb:df IEEE 802.11: disassociated
wlan0: STA dc:4f:22:ee:bb:df IEEE 802.11: disassociated
wlan0: STA dc:4f:22:ee:bb:df IEEE 802.11: disassociated
wlan0: STA dc:4f:22:ee:bb:df IEEE 802.11: disassociated
wlan0: STA dc:4f:22:ee:bb:df IEEE 802.11: disassociated
wlan0: STA dc:4f:22:ee:bb:df IEEE 802.11: disassociated
wlan0: STA dc:4f:22:ee:bb:df IEEE 802.11: disassociated
wlan0: STA dc:4f:22:ee:bb:df IEEE 802.11: disassociated
wlan0: STA dc:4f:22:ee:bb:df IEEE 802.11: disassociated
wlan0: STA dc:4f:22:ee:bb:df IEEE 802.11: associated
wlan0: AP-STA-CONNECTED dc:4f:22:ee:bb:df
wlan0: STA dc:4f:22:ee:bb:df RADIUS: starting accounting session 68B546A08CB262B7
wlan0: STA dc:4f:22:ee:bb:df WPA: pairwise key handshake completed (RSN)

kueblc, 2019-03-24:
It looks like the web server failed to start due to port 80 being unavailable. Check what might be running on that port with `sudo lsof -i :80 -s TCP:LISTEN`.

sbeland2, 2019-03-24:
wow quick answer! i checked:

sudo lsof -i :80 -s TCP:LISTEN
COMMAND PID     USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
nginx   621     root    6u  IPv4  13786      0t0  TCP *:http (LISTEN)
nginx   621     root    7u  IPv6  13787      0t0  TCP *:http (LISTEN)
nginx   623 www-data    6u  IPv4  13786      0t0  TCP *:http (LISTEN)
nginx   623 www-data    7u  IPv6  13787      0t0  TCP *:http (LISTEN)
nginx   624 www-data    6u  IPv4  13786      0t0  TCP *:http (LISTEN)
nginx   624 www-data    7u  IPv6  13787      0t0  TCP *:http (LISTEN)
nginx   626 www-data    6u  IPv4  13786      0t0  TCP *:http (LISTEN)
nginx   626 www-data    7u  IPv6  13787      0t0  TCP *:http (LISTEN)
nginx   627 www-data    6u  IPv4  13786      0t0  TCP *:http (LISTEN)
nginx   627 www-data    7u  IPv6  13787      0t0  TCP *:http (LISTEN)

so i kill it
 sudo service nginx stop
pi@raspberrypi:~/tuya-convert/scripts $ sudo lsof -i :80 -s TCP:LISTEN
pi@raspberrypi:~/tuya-convert/scripts $

and redo the procedure, works!
wow, so much fun now, i have been struggling with that a lot for few days, trying tons of things including other way of flashing the device, thank you very much Kueblc!!!

sbeland2, 2019-03-24:
so after seeing the sonoff ap, i connected to it, enter the ssid+password of my wifi, change the hostname to outletoffice (i wonder if this was my mistake, i doubt), then it never come back alive again. here's the log when i did the flashing:

./start_flash.sh
~/tuya-convert/scripts ~/tuya-convert
======================================================
TUYA-CONVERT

https://github.com/ct-Open-Source/tuya-convert
TUYA-CONVERT was developed by Michael Steigerwald from the IT security company VTRUST (https://www.vtrust.de/) in collaboration with the techjournalists Merlin Schumacher, Pina Merkert, Andrijan Moecker and Jan Mahn at c't Magazine. (https://www.ct.de/)

======================================================
PLEASE READ THIS CAREFULLY!
======================================================
TUYA-CONVERT creates a fake update server environment for ESP8266/85 based tuya devices. It enables you to backup your devices firmware and upload an alternative one (e.g. ESPEasy, Tasmota, Espurna) without the need to open the device and solder a serial connection (OTA, Over-the-air).
Please make sure that you understand the consequences of flashing an alternative firmware, since you might lose functionality!

Flashing an alternative firmware can cause unexpected device behavior and/or render the device unusable. Be aware that you do use this software at YOUR OWN RISK! Please acknowledge that VTRUST and c't Magazine (or Heise Medien GmbH & Co. KG) CAN NOT be held accountable for ANY DAMAGE or LOSS OF FUNCTIONALITY by typing yes + Enter

yes
======================================================
  Starting AP in a screen
  Stopping any apache web server
  Starting web server in a screen
  Starting Mosquitto in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   The wpa-password is flashmeifyoucan
   This step is IMPORTANT otherwise the smartconfig will not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
3. Press ENTER to continue

======================================================
Starting pairing procedure in screen
RTNETLINK answers: File exists
Waiting for the upgraded device to appear
If this does not work have a look at the '*.log'-files in the 'scripts' subfolder!
.................................
IoT-device is online with ip 10.42.42.42
Fetching firmware backup
Create backup of entire FLASH from 10.42.42.42
Connected... Flashsize= 1048576
b'read 0x00000000 to 0x000003FF'
b'read 0x00000400 to 0x000007FF'
...
b'read 0x000FF800 to 0x000FFBFF'
b'read 0x000FFC00 to 0x000FFFFF'
~/tuya-convert
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.0
(c) VTRUST GMBH https://www.vtrust.de/35c3/

READ FLASH: http://10.42.42.42/get?read=HEX-ADDRESS
ChipID: eebbdf
MAC: DC:4F:22:EE:BB:DF
SdkVersion: 1.5.3(aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144051
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 3
system_upgrade_userbin_check: user2 0x81000

======================================================
Please make sure to note the correct SPI flash mode!
Installing an alternative firmware with the wrong flash mode will leave the ESP unable to boot!

Next steps:
1. To go back to the orginal software
   # curl http://10.42.42.42/undo

2. Be sure the conversion software runs in user2
   # curl http://10.42.42.42/flash2

3. Flash a third party firmware to the device
BE SURE THE FIRMWARE FITS THE DEVICE AND USES THE CORRECT FLASH MODE!
MAXIMUM SIZE IS 512KB
put or link it to ./files/thirdparty.bin
A basic build of Sonoff-Tasmota v6.5.0 is already included in this repository.
   # curl http://10.42.42.42/flash3
Alternatively let the device download and flash a file via HTTP:
   # curl http://10.42.42.42/flashURL?url=http://10.42.42.1/files/thirdparty.bin

HAVE FUN!
pi@raspberrypi:~/tuya-convert $ curl http://10.42.42.42/flash3
Device should flash http://10.42.42.1/files/thirdparty.bin and restart{\rtf1}

kueblc, 2019-03-24:
Nice, glad you flashed successfully.

I can't provide support for Tasmota configuration issues, but I assure you the device is probably not dead! Check your router for any new devices and if you don't see it, [start troubleshooting here](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#Wi-Fi-does-not-connect).

closed 2019-03-24 by kueblc

sbeland2, 2019-03-24:
i had a second device, i did the same thing except not changing the hostname, works, have access to it at the new ip, can toggle it, so in the future, will be worth to mention: NEVER CHANGE THE HOSTNAME at the ssid config level.