ct-Open-Source/tuya-convert #155: Orange Pi Support?

ned-kelly, 2019-03-25:
Hi all,

Has anyone had any luck getting this running on an Orange Pi and/or could point me in the right direction?

I'm running Armbian - and when running `start_flash.sh` the process starts the four main processes in individual screen sessions _(smarthack-smartconfig, smarthack-mqtt, smarthack-web, smarthack-wifi)_ however when connecting with my Smart phone (as suggested in the setup steps) I'm not getting any IP from the DHCP server on the Pi.

Here's the logs from my `smarthack-wifi` screen session:

```
Backing up NetworkManager.cfg...
Restarting NetworkManager...
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...

dnsmasq: failed to create listening socket for port 53: Address already in use
RTNETLINK answers: File exists
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Using interface wlan0 with hwaddr 12:42:de:b3:82:25 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: STA f4:31:c3:52:a0:11 IEEE 802.11: authenticated
wlan0: STA f4:31:c3:52:a0:11 IEEE 802.11: associated (aid 1)
wlan0: AP-STA-CONNECTED f4:31:c3:52:a0:11
wlan0: STA f4:31:c3:52:a0:11 RADIUS: starting accounting session 6C729FA9394EE9D6
wlan0: STA f4:31:c3:52:a0:11 WPA: pairwise key handshake completed (RSN)
wlan0: AP-STA-DISCONNECTED f4:31:c3:52:a0:11
wlan0: STA f4:31:c3:52:a0:11 IEEE 802.11: disassociated
wlan0: STA f4:31:c3:52:a0:11 IEEE 802.11: authenticated
wlan0: STA f4:31:c3:52:a0:11 IEEE 802.11: associated (aid 1)
wlan0: AP-STA-CONNECTED f4:31:c3:52:a0:11
wlan0: STA f4:31:c3:52:a0:11 RADIUS: starting accounting session 6C729FA9394EE9D6
wlan0: STA f4:31:c3:52:a0:11 WPA: pairwise key handshake completed (RSN)
```

Any help would be greatly appreciated!

Cheers,

blakadder, 2019-03-27:
Works on my Orange Pi Pc with armbian image without issues

jvyoralek, 2019-03-27:
Problem is only with Ubuntu version of Armbian OS. 

You have two options, how to resolve this problem:
1. stop systemd-resolved service and re-run tuya-convert
`sudo systemctl stop systemd-resolved`
2. use Debian version of Armbian OS - is has no conflict with dnsmasq service

ned-kelly, 2019-03-29:
Ok - I'll try the Debian distro and report back!

kueblc, 2019-03-31:
Hi @ned-kelly,

I'm testing out an alternative for generating an AP. Would you mind trying it out and seeing if this resolves your issue? `git clone -b create-ap https://github.com/kueblc/tuya-convert.git`

rbreaves, 2019-05-19:
I have an Orange Pi running Armbian and the create-ap branch does not appear to be working for me. I do not see an AP name being available at all.

kueblc, 2019-05-19:
Thanks for testing @rbreaves!

rbreaves, 2019-06-07:
I am going to try again, on an Orange Pi, but it might be later this weekend. I may have done something wrong during the setup, as I also had initial difficulties getting it to work on an actual Raspberry Pi, but did get it working.

BuxtonCalvin, 2019-06-07:
I ran the program on an Orange Pi 2 pc and was not able to get it to work using Armbian, as DNSmasq port (52) conflicted with another network manager.  There were some other AP problems as well.  But, I was able to get everything running smoothly using Debian. 

However, after getting up and running, I was not able to flash my Tuya plugs due to the HTTPS issue that's been widely commented on.

BuxtonCalvin, 2019-10-02:
As an update, I was able to flash my gosund wp5 plugs with new convert-api branch on and OrangePI PC with a wifi dongle.  I loaded standard Debian on the orangePI, upgraded to buster from stretch ( you have to manually update the debian AND the armbian apt sources to buster).   Next bring down the wifi (ip link set dev "your wifi name" down), rename the wifi to wlan0, bring up wlan0, and then run the .start_flash.sh program.   Also, you may have to do some manually updating of any failed per-requisites.  I needed to install Python3 pip and some other Python utils that were causing errors.  Once everything was installed, and there were no errors in the per-requisites, I was able to flash without a problem.

Great work here guys.  Much appreciated to get my devices off the cloud.

closed 2019-10-02 by ned-kelly