ct-Open-Source/tuya-convert #156: Luminea WLAN LED Bulb ZX-2831-675

22MarioZ, 2019-03-26:
I'm trying to convert the mention WiFi LED bulb (is listed as compatible devices) but I have no success.

The script come to "Waiting for the upgraded device to appear" and I get a lot of ........

Some how the device cannot be find.

When the bulb in paring mode (flashing on regular basis (switch on/switch off) I can see the bulb AP. IS this OK?

I used the latest version of tuya-convert.

kueblc, 2019-03-26:
The device needs to be in EZ Config mode, not AP Config mode. It should blink quickly and no AP should appear from the device.

22MarioZ, 2019-03-26:
Thank you. Managed to convert. To help others: to put the bulb in EZ Config mode the switch on/off needs to be really quick (5 time at least).

closed 2019-03-26 by 22MarioZ