ct-Open-Source/tuya-convert #157: Cannot obtain ip-address (Ubuntu 18.04)

jurgen272, 2019-03-27:
I cannot get an ip-address when connecting to vtrust-flash (tried with 2 different androis devices)
Looked in other issues for similar problems and tried the stuff that solved those, but still no luck.
(did stop systemd-resolved service with: sudo systemctl stop systemd-resolved.service)

Logs:
 Wifi log:
Backing up NetworkManager.cfg...
Restarting NetworkManager...
sudo: ifconfig: command not found
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
sudo: ifconfig: command not found
Applying iptables rules...
Starting DNSMASQ server...

dnsmasq: failed to create listening socket for port 53: Address already in use
Error: Device for nexthop is not up.
Starting AP on wlp3s0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Using interface wlp3s0 with hwaddr 24:0a:64:ae:46:67 and ssid "vtrust-flash"
wlp3s0: interface state UNINITIALIZED->ENABLED
wlp3s0: AP-ENABLED 
wlp3s0: STA e8:b4:c8:15:88:aa IEEE 802.11: authenticated
wlp3s0: STA e8:b4:c8:15:88:aa IEEE 802.11: associated (aid 1)
wlp3s0: AP-STA-CONNECTED e8:b4:c8:15:88:aa
wlp3s0: STA e8:b4:c8:15:88:aa RADIUS: starting accounting session C7D20C1F0FC7573B
wlp3s0: STA e8:b4:c8:15:88:aa WPA: pairwise key handshake completed (RSN)
wlp3s0: AP-STA-DISCONNECTED e8:b4:c8:15:88:aa
wlp3s0: STA e8:b4:c8:15:88:aa IEEE 802.11: disassociated
wlp3s0: STA e8:b4:c8:15:88:aa IEEE 802.11: authenticated
wlp3s0: STA e8:b4:c8:15:88:aa IEEE 802.11: associated (aid 1)
wlp3s0: AP-STA-CONNECTED e8:b4:c8:15:88:aa
wlp3s0: STA e8:b4:c8:15:88:aa RADIUS: starting accounting session C7D20C1F0FC7573B
wlp3s0: STA e8:b4:c8:15:88:aa WPA: pairwise key handshake completed (RSN)
wlp3s0: AP-STA-DISCONNECTED e8:b4:c8:15:88:aa
wlp3s0: STA e8:b4:c8:15:88:aa IEEE 802.11: disassociated

MQTT log:
1553701616: mosquitto version 1.4.15 (build date Wed, 13 Feb 2019 00:27:01 +0000) starting
1553701616: Using default config.
1553701616: Opening ipv4 listen socket on port 1883.
1553701616: Error: Address already in use

Web log:
Listening on port 80

kueblc, 2019-03-27:
> dnsmasq: failed to create listening socket for port 53: Address already in use

Find out what's occupying that port with `sudo lsof -i :53`

jvyoralek, 2019-03-27:
I think the same issue as described in #155.

jurgen272, 2019-03-28:
sudo lsof -i :53 before running the script:

COMMAND   PID            USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
systemd-r 588 systemd-resolve   12u  IPv4  19098      0t0  UDP localhost:domain 
systemd-r 588 systemd-resolve   13u  IPv4  19099      0t0  TCP localhost:domain 

I stopped the service with: sudo service systemd-resolved stop

sudo lasof -i :53 returned blank now

Ran the script with: sudo ./start_flash.sh

log wifi:
Backing up NetworkManager.cfg...
Restarting NetworkManager...
sudo: ifconfig: command not found
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
sudo: ifconfig: command not found
Applying iptables rules...
Starting DNSMASQ server...

dnsmasq: failed to create listening socket for port 53: Address already in use
Error: Device for nexthop is not up.
Starting AP on wlp3s0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Using interface wlp3s0 with hwaddr 24:0a:64:ae:46:67 and ssid "vtrust-flash"
wlp3s0: interface state UNINITIALIZED->ENABLED
wlp3s0: AP-ENABLED 
wlp3s0: STA 6c:c7:ec:9b:24:01 IEEE 802.11: authenticated
wlp3s0: STA 6c:c7:ec:9b:24:01 IEEE 802.11: associated (aid 1)
wlp3s0: AP-STA-CONNECTED 6c:c7:ec:9b:24:01
wlp3s0: STA 6c:c7:ec:9b:24:01 RADIUS: starting accounting session 64C7C9A27596B4AA
wlp3s0: STA 6c:c7:ec:9b:24:01 WPA: pairwise key handshake completed (RSN)
wlp3s0: AP-STA-DISCONNECTED 6c:c7:ec:9b:24:01
wlp3s0: STA 6c:c7:ec:9b:24:01 IEEE 802.11: disassociated
wlp3s0: STA 6c:c7:ec:9b:24:01 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

and result of sudo lsof -i :53 :
COMMAND    PID            USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
systemd-r 3466 systemd-resolve   12u  IPv4  70377      0t0  UDP localhost:domain 
systemd-r 3466 systemd-resolve   13u  IPv4  70378      0t0  TCP localhost:domain

Backing up NetworkManager.cfg...
Restarting NetworkManager...
sudo: ifconfig: command not found
...

This made me thinking.
Ubuntu 18.04 is using netplan instead off NetworkManager.
I'll try to change my config to NetworkManager ans see if this goes better.

jurgen272, 2019-03-28:
I forced Ubuntu to use NetworkManager:
created etc/netplan/01-network-manager-all.yaml with
   # Let NetworkManager manage all devices on this system
   network:
     version: 2
     renderer: NetworkManager

Installed net-tools to get ifconfig back (Ubuntu is using IP)

But error regarding to port 53 is still there (even if I stop the service with: sudo service systemd-resolved stop before running the flash script)

Wifi log:
Backing up NetworkManager.cfg...
Restarting NetworkManager...
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...

dnsmasq: failed to create listening socket for port 53: Address already in use
Starting AP on wlp3s0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Using interface wlp3s0 with hwaddr 24:0a:64:ae:46:67 and ssid "vtrust-flash"
wlp3s0: interface state UNINITIALIZED->ENABLED
wlp3s0: AP-ENABLED 
wlp3s0: STA 6c:c7:ec:9b:24:01 IEEE 802.11: authenticated
wlp3s0: STA 6c:c7:ec:9b:24:01 IEEE 802.11: associated (aid 1)
wlp3s0: AP-STA-CONNECTED 6c:c7:ec:9b:24:01
wlp3s0: STA 6c:c7:ec:9b:24:01 RADIUS: starting accounting session 4727817A8071E9CA
wlp3s0: STA 6c:c7:ec:9b:24:01 WPA: pairwise key handshake completed (RSN)
wlp3s0: AP-STA-DISCONNECTED 6c:c7:ec:9b:24:01
wlp3s0: STA 6c:c7:ec:9b:24:01 IEEE 802.11: disassociated
wlp3s0: STA 6c:c7:ec:9b:24:01 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

I guess I'll have to find a Debian distribution to try.

jurgen272, 2019-03-28:
Installed Debian on the machine.
I got the AP working after some problems with dependencies which were not installed (net-tools, hostapd, ...)

BUT
the switch is connecting to the AP, but nothing else happens.
(red light was blinking first, now light is always on)

Wifi log:
Backing up NetworkManager.cfg...
Restarting NetworkManager...
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
Starting AP on wlp3s0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Using interface wlp3s0 with hwaddr 24:0a:64:xxxxxxx and ssid "vtrust-flash"
wlp3s0: interface state UNINITIALIZED->ENABLED
wlp3s0: AP-ENABLED 
wlp3s0: STA 6c:xxxxx 1 IEEE 802.11: authenticated
wlp3s0: STA 6c:xxxxx IEEE 802.11: associated (aid 1)
wlp3s0: AP-STA-CONNECTED 6c:c7:ec:9b:24:01
wlp3s0: STA 6c:xxxxx RADIUS: starting accounting session 5C9CA4A1-00000000
wlp3s0: STA 6c:xxxxx WPA: pairwise key handshake completed (RSN)
wlp3s0: STA dc:4f:22:xxxxx IEEE 802.11: authenticated
wlp3s0: STA dc:4f:22:xxxxx IEEE 802.11: associated (aid 2)
wlp3s0: AP-STA-CONNECTED dc:4f:22:xxxxx
wlp3s0: STA dc:4f:22:xxxxx RADIUS: starting accounting session 5C9CA4A1-00000001
wlp3s0: STA dc:4f:22:xxxxx WPA: pairwise key handshake completed (RSN)
wlp3s0: STA dc:4f:22:xxxxx WPA: group key handshake completed (RSN)
wlp3s0: STA 6c:xxxxx WPA: group key handshake completed (RSN)
wlp3s0: AP-STA-DISCONNECTED 6c:xxxxx
wlp3s0: STA 6c:xxxxx IEEE 802.11: disassociated
wlp3s0: STA 6c:xxxxx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
wlp3s0: STA dc:4f:22:xxxxx WPA: group key handshake completed (RSN)
wlp3s0: STA dc:4f:22:xxxxx WPA: group key handshake completed (RSN)

dc:4f:22:xxxxx= the tuya device

Weblog:
Listening on port 80

mqttlog:
1553769633: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1553769633: Using default config.
1553769633: Opening ipv4 listen socket on port 1883.
1553769633: Opening ipv6 listen socket on port 1883.

smartconfiglog:
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.

tcpdump:
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wlp3s0, link-type EN10MB (Ethernet), capture size 262144 bytes
12:12:44.116232 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:12:45.119032 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:12:46.143027 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:12:47.734351 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:12:48.735025 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:12:49.759028 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:12:51.352322 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:12:51.598384 ARP, Request who-has 10.42.42.38 tell 10.42.42.38, length 28
12:12:51.598414 ARP, Request who-has 10.42.42.38 tell 10.42.42.38, length 28
12:12:52.383025 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:12:53.407028 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:12:54.970342 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:12:55.999032 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:12:57.023037 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:12:58.588277 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:12:59.615030 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:13:00.639033 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:13:01.602943 ARP, Request who-has 10.42.42.38 tell 10.42.42.38, length 28
12:13:01.602973 ARP, Request who-has 10.42.42.38 tell 10.42.42.38, length 28
12:13:02.207196 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:13:03.231030 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:13:04.255031 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:13:05.825593 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:13:06.847032 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:13:07.871039 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:13:09.443578 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:13:10.463027 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:13:11.487027 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:13:11.603026 ARP, Request who-has 10.42.42.38 tell 10.42.42.38, length 28
12:13:11.603054 ARP, Request who-has 10.42.42.38 tell 10.42.42.38, length 28
12:13:13.062568 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:13:14.079032 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:13:15.103030 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:13:16.680928 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:13:17.695033 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:13:18.719029 ARP, Request who-has 10.42.42.42 tell Debian, length 28
12:13:20.299860 ARP, Request who-has 10.42.42.42 tell Debian, length 28
...

jurgen272, 2019-03-28:
Firmware off the device: 1.0.4

ragenhe, 2019-03-28:
Try putting the device in pairing mode before running ./start_flash.
I was banging my head against the wall with the timing of my device.
1) put device in pairing mode.
2) started flash script
3) device paired and started updating within 10 seconds.

jurgen272, 2019-03-28:
> Try putting the device in pairing mode before running ./start_flash.
> I was banging my head against the wall with the timing of my device.
> 
>     1. put device in pairing mode.
> 
>     2. started flash script
> 
>     3. device paired and started updating within 10 seconds.

Tried this, but still no updates. Devices pairs, but seems to do nothing after that.

kueblc, 2019-03-28:
Device pairing is a good sign. Can you capture your network activity with `tcpdump` or Wireshark?

kueblc, 2019-03-28:
Also, please post device details including model and firmware version.

Have you already added the device to the official app? If so, try running `nc -u -l -k -p 6666 | strings` on your regular network to verify that these devices are compatible; you should see some JSON.

If not, try `nc -u -l -k -p 6667`, if you see garbage (encrypted content) then unfortunately you may be affected by the recent patch to close our firmware options. If that's the case, you won't be able to flash OTA, but you can still flash the old fashion way, physically connecting to the MCU's serial port. We are still investigating the new patch, and any data you can gather would be tremendously helpful in developing a workaround.

kueblc, 2019-03-31:
I'm working on an alternative for generating a new AP. Please consider trying it out and report back if it resolves your issues `git clone -b create-ap https://github.com/kueblc/tuya-convert.git`

jurgen272, 2019-04-01:
I'll try to do some more testing Wednesday or Thursday.

ema987, 2019-04-02:
Hello everyone,
I was able to successfully use tuya-convert on a Ubuntu 18.04 live usb, but I had to make some changes.

**config.txt** -> **WLAN** value must be changed to **wlp1s0** (or another, **wlan0** is not the name used in Ubuntu 18.04 I assume, I checked it using **iwconfig**)
**script/setup_ap.sh** -> go to line 72 `echo "Starting DNSMASQ server..."` and after that add `sudo service systemd-resolved stop` (otherwise you get the error port 53 is already in use, as stated in #152 )

Then, **setup_ap.sh** uses **ifconfig** which is not installed by default on Ubuntu 18.04 so I had to install it using `sudo apt-get install net-tools`

Try now again and you should be able to start the script, get an IP assigned on the device you connect to the just created WiFi network, and then start the flash procedure successfully.

kueblc, 2019-04-02:
Hi @ema987 

Would you be able to test if the `create-ap` branch works for you on Ubuntu 18.04?

`git clone -b create-ap https://github.com/kueblc/tuya-convert.git`

jurgen272, 2019-04-03:
I tested the create-ap version on Debian.
Had to kill DNSMASQ before running the script.

The result stays the same. Switch gets connected, but rest in this state. no update...

TCPDUMP which is related to Tuya:

7:35:49.078140 ARP, Request who-has 192.168.0.192 tell 192.168.0.156, length 46
07:35:49.078165 ARP, Request who-has 192.168.0.185 tell 192.168.0.156, length 46
07:35:49.506905 IP 192.168.0.110.38924 > 255.255.255.255.10001: UDP, length 144
07:35:49.554053 ARP, Request who-has 192.168.0.10 tell gateway, length 46
07:35:50.552832 ARP, Request who-has 192.168.0.10 tell gateway, length 46
07:35:50.853067 IP6 ptr-fvmht5owv11wqz2yt1q.18120a2.ip6.access.telenet.be.56401 > gent01.dnsresv6.prd.telenet-ops.be.domain: 25950+ A? accscdn.m.taobao.com. (38)
07:35:50.853120 IP6 ptr-fvmht5owv11wqz2yt1q.18120a2.ip6.access.telenet.be.56401 > gent02.dnsresv6.prd.telenet-ops.be.domain: 25950+ A? accscdn.m.taobao.com. (38)
07:35:50.853215 IP Debian.1014 > gent.dnscache01.telenet-ops.be.domain: 25950+ A? accscdn.m.taobao.com. (38)
07:35:50.853265 IP Debian.1014 > gent.dnscache02.telenet-ops.be.domain: 25950+ A? accscdn.m.taobao.com. (38)
07:35:50.853407 IP6 ptr-fvmht5owv11wqz2yt1q.18120a2.ip6.access.telenet.be.56024 > gent01.dnsresv6.prd.telenet-ops.be.domain: 64191+ PTR? 2.0.0.0.2.4.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.1.0.0.0.8.1.2.0.a.2.ip6.arpa. (90)
07:35:50.875953 IP6 gent01.dnsresv6.prd.telenet-ops.be.domain > ptr-fvmht5owv11wqz2yt1q.18120a2.ip6.access.telenet.be.56401: 25950 9/0/0 CNAME zb-center-accscdn.m.taobao.com., A 203.119.201.254, A 203.119.216.122, A 203.119.204.253, A 203.119.204.254, A 203.119.201.255, A 203.119.215.254, A 203.119.216.1, A 203.119.215.255 (198)
07:35:50.879797 IP6 gent01.dnsresv6.prd.telenet-ops.be.domain > ptr-fvmht5owv11wqz2yt1q.18120a2.ip6.access.telenet.be.56024: 64191 1/0/0 PTR gent02.dnsresv6.prd.telenet-ops.be. (138)
07:35:50.880235 IP6 ptr-fvmht5owv11wqz2yt1q.18120a2.ip6.access.telenet.be.57388 > gent01.dnsresv6.prd.telenet-ops.be.domain: 13674+ PTR? 2.130.130.195.in-addr.arpa. (44)
07:35:50.882421 IP gent.dnscache01.telenet-ops.be.domain > Debian.1014: 25950 9/0/0 CNAME zb-center-accscdn.m.taobao.com., A 203.119.201.255, A 203.119.215.254, A 203.119.204.254, A 203.119.204.253, A 203.119.216.122, A 203.119.201.254, A 203.119.216.1, A 203.119.215.255 (198)
07:35:50.900382 IP6 ptr-fvmht5owv11wqz2yt1q.18120a2.ip6.access.telenet.be.34082 > gent01.dnsresv6.prd.telenet-ops.be.domain: 25820+ PTR? 2.131.130.195.in-addr.arpa. (44)
07:35:51.564365 ARP, Request who-has 192.168.0.10 tell gateway, length 46
07:35:54.966315 ARP, Request who-has 192.168.0.185 tell 192.168.0.156, length 46
07:35:54.966342 ARP, Request who-has 192.168.0.192 tell 192.168.0.156, length 46
07:35:55.415745 ARP, Request who-has 192.168.0.137 tell gateway, length 46
07:35:55.632880 ARP, Request who-has 192.168.0.10 tell gateway, length 46

re-added the device to the android app:
nc -u -l -k -p 6667
���e&��pOn�DD�DL��l�>b�p�G����mts��#׫ߧ�
A;�э/���ي �̆>S$���^�1"�����X��[#i~����[�@�x.���!���_w��
�6ɻA�ο�}�v��j�4��f�ۃ��"�i���U^C

I guess this is not good news ...

ema987, 2019-04-03:
> Hi @ema987
> 
> Would you be able to test if the `create-ap` branch works for you on Ubuntu 18.04?
> 
> `git clone -b create-ap https://github.com/kueblc/tuya-convert.git`

Hello @kueblc, unfortunately I have no more devices which I can convert :( 
I can try to see if my phone which I use to connect to the AP gets the IP assigned, if that is enough.

kueblc, 2019-04-03:
> nc -u -l -k -p 6667
> ����e&���pOn�DD��DL��l�>b�p�G������mts��#׫ߧ�
> �A;�э/���ي �̆>S$���^�1"�����X��[#i~�����[�@�x.����!���_�w��
> �6ɻA�ο�}�v��j�4��f�ۃ��"��i���U^C
> 
> I guess this is not good news ...

You're right @jurgen272, unfortunately this means your device has been patched by Tuya and will no longer work with `tuya-convert` in its current form. We're investigating possible workarounds.

At this point if you still want to flash your device, you'll have to do it the "old fashion way" by physically opening the device and flashing over serial. If you do, please make a backup of the firmware as this would help us understand the changes.

@ema987 No problem, just checking to see if using `create_ap` helps people with AP issues.

mrand, 2019-04-08:
@kueblc 

I just tried out create_ap branch and it worked on three Lohas bulbs using Lubuntu 18.04  (basically Ubuntu)!

Here are a few minor things of note:

1. dnsmasq is still installed, but apt-get complains about it failing to start up since port 53 is in use.  I ignored this 
2. start_flash.sh appears to try to launch screen, but in my case, never seemed to work for some reason, causing me to miss that something wasn't working.  I did verify screen is installed, but haven't tried to debug any further.
3. The problem is that my Ubuntu machine was still connected to a 5 GHz internet connection, and when create_ap tried to create the AP, it was trying to do so at 5 GHz.  I had assumed it would just try to do it at 2.4 GHz.  I simply had to tell it to disconnect from the 5 GHz network and everything started up after that. This got me to wondering - would the AP bridge the Tuya device to the internet if I'd had a wired connection?  If so, that could have been bad - perhaps worth a warning somewhere?
4. I assume it's normal to have a challenging time getting the bulbs to connect up to the convert software.  Even after doing three of them, with the third going on the first try, I'm not sure what the magic formula is to have it connect successfully.  I just kept re-trying and eventually it worked.

I have all the logs from the failed and successful attempts, as well as the backed-up firmware if there is a strong interest.

Thanks so much for this utility - much nicer than trying to get the cover off the bulbs and wiring to them!

kueblc, 2019-04-08:
@mrand great, thank you for trying it out!

> I changed WLAN to wlp1s0 in config.txt.  I didn't even try it without that since iwconfig showed wlp1s0

Good. Perhaps `install_prereq` should run `iwconfig` and let the user know if they need to change config?

> start_flash.sh appears to try to launch screen, but in my case, never seemed to work for some reason

The `screen` instances are terminals running in the background. You normally won't see these but you can manually connect to them if you want to see what's going on live, or just check the logs afterwards.

> The problem is that my Ubuntu machine was still connected to a 5 GHz internet connection, and when create_ap tried to create the AP, it was trying to do so at 5 GHz

Great to note, thanks! Hadn't considered this, perhaps a note in the README would suffice.

> This got me to wondering - would the AP bridge the Tuya device to the internet if I'd had a wired connection?  If so, that could have been bad

Yes that would be bad, no, this branch does not permit internet access from `vtrust-flash`. You can test this by connecting with another device and trying any site. You should be able to run this branch without any internet connection whatsoever.

> I have all the logs from the failed and successful attempts, as well as the backed-up firmware if there is a strong interest.

Do you know if an update is available for the device? If so, the firmware backup would be helpful in our efforts to develop a workaround for the patch.

Thanks for the detailed feedback! Very helpful

mrand, 2019-04-17:
@kueblc How would I know if there was an 

> Good. Perhaps install_prereq should run iwconfig and let the user know if they need to change config?

Yes.  Or at a bare minimum, a mention in the documentation.

> The screen instances are terminals running in the background. You normally won't see these but you can manually connect to them if you want to see what's going on live, or just check the logs afterwards.

I somehow missed that.  When it didn't work, I just started going through the logs.

> Do you know if an update is available for the device? If so, the firmware backup would be helpful in our efforts to develop a workaround for the patch.

Hmmm...  not sure how I'd know since I didn't look for updates before flashing them.  All of them say BootVersion: 4.  If you think the firmware files would help, let me know the best way to get them to you.

I did forget to mention that my ability to flash also didn't seem to be impacted by this crash:

```
Traceback (most recent call last):
  File "./fake-registration-server.py", line 129, in <module>
    main()
  File "./fake-registration-server.py", line 125, in main
    tornado.ioloop.IOLoop.current().start()
  File "/usr/local/lib/python3.6/dist-packages/tornado/platform/asyncio.py", line 148, in start
    self.asyncio_loop.run_forever()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 427, in run_forever
    self._run_once()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 1404, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.6/selectors.py", line 445, in select
    fd_event_list = self._epoll.poll(timeout, max_ev)
KeyboardInterrupt
```

ThomasS-arch, 2019-08-12:
Disable systemd-resolve, it normally takes up port 53

kueblc, 2019-08-12:
This has already been discussed and implemented

closed 2019-08-12 by kueblc

barrystaes, 2019-09-08:
I could not flash my KLSP1 (Koogeek Smart Plug) using Ubuntu 18.04.3 LTS and noticed my phone got no IP. So found this issue, tried this suggestion, and now flashing works:

> **script/setup_ap.sh** -> go to line 72 `echo "Starting DNSMASQ server..."` and after that add `sudo service systemd-resolved stop` (otherwise you get the error port 53 is already in use, as stated in #152 )

As pointed out by @ema987 in [a comment](https://github.com/ct-Open-Source/tuya-convert/issues/157#issuecomment-478972681) above. Apparently running `sudo systemctl stop systemd-resolved` before `./start_flash.sh` is not enough.