ct-Open-Source/tuya-convert #159: File Not Found during flash

onlize, 2019-03-30:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**
Hello,

I am trying to flash Teckin SP-20 Smart-WiFi Plug. It uses ESP8266.

I tried to flash device with my firmware file, but the error message says that file does not exist.
After that, I tried to flash sonoff-basic.bin file, but it says that the file does not exit, but it is there.

```
Next steps:
1. To go back to the orginal software
   # curl http://10.42.42.42/undo

2. Be sure the conversion software runs in user2
   # curl http://10.42.42.42/flash2

3. Flash a third party firmware to the device
BE SURE THE FIRMWARE FITS THE DEVICE AND USES THE CORRECT FLASH MODE!
MAXIMUM SIZE IS 512KB
put or link it to ./files/thirdparty.bin
A basic build of Sonoff-Tasmota v6.5.0 is already included in this repository.
   # curl http://10.42.42.42/flash3
Alternatively let the device download and flash a file via HTTP:
   # curl http://10.42.42.42/flashURL?url=http://10.42.42.1/files/thirdparty.bin

HAVE FUN!
pi@raspberrypi:~/tuya-convert/files $ ls -la
total 1508
drwxr-xr-x 2 pi pi   4096 Mar 30 00:07 .
drwxr-xr-x 6 pi pi   4096 Mar 29 23:42 ..
-rw-r--r-- 1 pi pi 325648 Mar 30 00:07 firmware.bin
-rw-r--r-- 1 pi pi 442528 Mar 29 22:30 sonoff-basic.bin
lrwxrwxrwx 1 pi pi     16 Mar 29 22:30 thirdparty.bin -> sonoff-basic.bin
-rw-r--r-- 1 pi pi 506587 Mar 29 22:30 upgrade.bin
-rw-r--r-- 1 pi pi 253268 Mar 29 22:30 user2.bin
pi@raspberrypi:~/tuya-convert $ curl http://10.42.42.42/flashURL?url=http://10.42.42.1/files/firmware.bin
File Not Found
pi@raspberrypi:~/tuya-convert $ curl http://10.42.42.42/flashURL?url=http://10.42.42.1/files/sonoff-basic.bin
File Not Found
pi@raspberrypi:~/tuya-convert $
```

Can you help?
Thank you.

kueblc, 2019-03-30:
Thanks for reporting, I forgot to update that message when I combined `flash3` and `flashURL`. I'll fix it shortly.

If you just want to flash sonoff-basic.bin, do `curl http://10.42.42.42/flash3`. If you want it to download a given URL, use `curl http://10.42.42.42/flash3?url=...`.

closed 2019-03-30 by kueblc

onlize, 2019-03-30:
I want to upload the specific firmware, not sonoff. I used sonoff just to test that neither custom nor sonoff firmware works.

Should I use 
```
curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/custom.bin
```
when I want to use custom firmware?

kueblc, 2019-03-30:
Yes

daxkamala, 2019-08-05:
Hi, I'm having this problem with latest everything:
3. Press ENTER to continue

======================================================
Starting pairing procedure in screen
RTNETLINK answers: File exists
/home/pi/tuya-convert
Waiting for the upgraded device to appear
If this does not work have a look at the '*.log'-files in the 'scripts' subfolder!

IoT-device is online with ip 10.42.42.42
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   302  100   302    0     0  13130      0 --:--:-- --:--:-- --:--:-- 13727
======================================================
Getting Info from IoT-device
Hello, world
======================================================
Please make sure to note the correct SPI flash mode!
Installing an alternative firmware with the wrong flash mode will leave the ESP unable to boot!

Next steps:
1. To go back to the orginal software
   # curl http://10.42.42.42/undo

2. Be sure the conversion software runs in user2
   # curl http://10.42.42.42/flash2

3. Flash a third party firmware to the device
BE SURE THE FIRMWARE FITS THE DEVICE AND USES THE CORRECT FLASH MODE!
MAXIMUM SIZE IS 512KB
put or link it to ./files/thirdparty.bin
A basic build of Sonoff-Tasmota v6.5.0 is already included in this repository.
   # curl http://10.42.42.42/flash3
Alternatively let the device download and flash a file via HTTP:
   # curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/thirdparty.bin

HAVE FUN!
root@raspberrypi:/home/pi/tuya-convert# curl http://10.42.42.42/flash3
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/tornado/web.py", line 1569, in _execute
    result = self.prepare()
  File "/usr/lib/python3/dist-packages/tornado/web.py", line 2322, in prepare
    raise HTTPError(self._status_code)
tornado.web.HTTPError: HTTP 404: Not Found
root@raspberrypi:/home/pi/tuya-convert#

Seems like a simple path error but I've tried all I can think of. Any ideas please?

daxkamala, 2019-08-05:
Further:
the device is a smart plug from Shenzhen Core Image Co but I've tried a Blitzwolf SHP3 with the same results. I'm getting a Hello World from the device, not the flash detail I expected.

kueblc, 2019-08-05:
Please open a new issue for this is unrelated to the issue you are commenting on. Are you running `tuya-convert` on a device with IP address 10.42.42.42? This would break the script as the script checks for your IoT device at that address.