ct-Open-Source/tuya-convert #160: Blitzwolf BW-SHP2 can't be found

voodoobg, 2019-03-31:
I have followed the install procedure and tried it on 2 raspberry pi's 3B and 1B with wifi adapter.
The outcome is the same, tuya-convert cannot pick up the plugs ( I have 3 )
I tried one with the tuya app and everything works ok.
Here are the logs:
https://pastebin.com/mX3mAxWq

kueblc, 2019-03-31:
Can you record your network activity with `tcpdump` or Wireshark and share the capture files? What firmware version are the plugs?

voodoobg, 2019-04-01:
Version is 1.0.5, here is the tcp dump:
https://pastebin.com/F4ide0VD

kueblc, 2019-04-01:
Is your device `10.42.42.23`? Looks like it's trying to talk to Home Assistant

voodoobg, 2019-04-01:
Indeed this is strange in the logs i saw it too. My network is on very different ip range than this ip. I have pihole as dns resolver and my router is on ddwrt with  DNSMasq as you can see in the logs. The device is tablet with HADashboard installed, maybe this is requesting Home Assistant. Is this relevant to the convert script? What device should I use? I've tried also my phone, the result was the same.

voodoobg, 2019-04-02:
I've read the thread for the 1.0.5 firmware... So i'll try with the Kali Linux way, since one guy reported working. My switches came with the 1.0.5 version, maybe that's why tuya-convert didn't work. I'll post the dump of the stock firmware when i get it, if it works from the Kali.

kueblc, 2019-04-02:
I'm not sure I can identify your smart device in the `tcpdump`. Do you see your plug's mac or IP in the log?

It shouldn't make too much of a difference with Kali linux. If the issue is that the firmware is too new, it'll make requests via HTTPS instead of HTTP, and then we're out of luck for the time being, no matter what OS.

It's inconclusive as to whether your SHP2 firmware has the HTTPS update. Firmware versioning is per device, so Blitzwolf firmware 1.0.5 isn't necessarily the same as Teckin 1.0.5.

If you do discover that the device is demanding HTTPS, the only method for flashing we have available right now is to flash it "the old fashion way" by physically opening the device and flashing it over the serial port. If you decide to do that, you can make a backup first, and that would be tremendously useful in reverse engineering efforts to update the `tuya-convert` process for the new firmware updates.

pcmester, 2019-04-30:
My BW-SHP2 arrived yesterday, and I think they applied the patch to the new ones, because it connects, but the script does not continue

JBS5, 2019-05-13:
Tried it with a BlitzWolf SHP2 which I received a few days ago, but no luck finding it after the './start_flash.sh' command. Any way to fix this?

eterpstra, 2019-05-21:
How can I check the firmware of my SPH111 withhout connecting it to the internet?

sebastianklein96, 2019-05-21:
@eterpstra If you're not soldering to it and dumping the firmware via serial, I don't think you can. Tuya-connect not recognizing the device is a pretty good indicator for new firmware though, if you're sure you set up everything correctly.

eterpstra, 2019-05-21:
@sebastianklein96 Yeah i am pretty sure of my setup.. have flashd 8 socks without any problems a few weeks ago... The other problem is that i bought a 4 pack of these SP111 Go-sund plugs.. 1 was flashed succesfully,bot the other 3 are not working.. So i cant send them back to Amazon I guess :P

probonopd, 2019-05-21:
To everyone having issues: Have you connected the light bulb to the app/the internet before flashing?

eterpstra, 2019-05-21:
Not a light bulb but my sp111 power plug was straight from the box without connecting it to the internet

jokerigno, 2019-05-22:
> To everyone having issues: Have you connected the light bulb to the app/the internet before flashing?

Nope.

Ixtalo, 2019-05-28:
New v2.4 20181101 devices (bought 05/2019) have a new firmware with version 1.06; old working versions were 1.05.
New devices with new firmware v1.06 use HTTPS (e.g., https://a3.tuyaus.com/gw.json, found in firmware) and try to connect to port 443 (instead of 80).
Firmware hacking scripts do not work with that, i.e., the `start_flash.sh` script hangs in endless loop because no 10.42.42.42 device is to be found.

See also:
#195 
#197 
#199 
...

Niek, 2019-06-04:
Also: https://github.com/ct-Open-Source/tuya-convert/issues/166

kueblc, 2019-09-28:
Happy to share #279 support for newer firmware

closed 2019-09-28 by kueblc