ct-Open-Source/tuya-convert #162: Teckin SP20 - Follow-up - backup/flashing difficulties/ logs + bin files attached

SirRedZ, 2019-04-01:
arendst/Sonoff-Tasmota#4175
[my post](https://github.com/arendst/Sonoff-Tasmota/issues/4175#issuecomment-478409043)
[login.sh](https://github.com/whiskerz007/proxmox_tuya-convert_container/blob/master/login.sh)
[sp20 logs.zip](https://github.com/ct-Open-Source/tuya-convert/files/3029606/sp20.logs.zip)

### Before opening a new issue, please read the following

**Are you reporting a successful flash?** No.
**Do you have a question about configuring your newly flashed device?** No.
**Did you have issues getting a device to flash?** Yes.

Teckin Sp20 US Smart Plug w/Energy Monitoring. FW: 1.0.0 / MCU: 1.0.0 (no updates available)

I have sacrificed a module specifically to share this information. After multiple attempts (more than 10) I was unable to smoothly convert this plug, as many others have mentioned/referenced above. It was my original goal to update the firmware, confirm blocked access to the process and provide the bin files for research. I was unable to do this.

I took a serial backup prior to this unit ever hitting AC. I have included my logs, the tuya-convert_backup.bin (which fails and stalls out), the Before and after serial reads, and the sonoff bin file I have flashed.

I use Whiskerz007 (proxmox) Tuya-convert script, which I have used many times -- it calls the current tuya-convert script and works within it's structure, with a few nice add-ons like indexing a folder of 3rd party firmware and samba file sharing. I have referenced that script above for you also.

I have asciinema'd the process, so you can see exactly what happens, and all relevant information is contained in the zip file. I truly hope this helps resolve this issue and gives some insight to the potential "blocking" of tuya-convert I posted about a week or two ago.

Whiskerz0007 has really helped me prepare this information, so I'll do my best to get additional information upon request. Thanks to those individuals in the original thread over at Tasmota who provided the required information for serial interfacing.

The process does eventually complete, through a 'brute force' approach and re-running the script a second time to finish the 3rd party flash. This particular module does not communicate/flash LEDs to let you know it's still alive, it will appear dormant/dead/bricked until you complete the process a second time, and see a sonoff-xxxx appear afterwards.

Good luck guys! And thanks so much for an awesome, much  appreciated process. 

[-=SrZ=-] aka Tollbringer

kueblc, 2019-04-01:
Hey @SirRedZ, thrilled you were able to collect so much data. I'll take a look and follow up soon. Thanks!

WilkyF18, 2019-04-02:
@SirRedZ so you were able to get the SP20 to connect to the "fake server"?  I get never ending dots.  Never connects.

SirRedZ, 2019-04-02:
Yes- it took me multiple attempts. I am not sure why some people struggle, some people report it works no issues. I purposely tried myself in order to try to solidify an answer.  Here are some thing syour might try:

a.) Try a different device to connect to Vtrust-flash. 
b.) Try setting the SP20 in pairing mode, remove from power. Start Tuya-Convert process and get dot's flowing, then plug in  SP20. Validate it is in pairing mode, or immediately put in pairing mode

If it connects at all, you may lose access - that doesn't mean it's bricked, but it is a possibility. It usually freezes during the backup stage for me, I abort and restart tuya-convert immediately, upon connection it instantly connects, skips the backup (as file already exists- even though it's corrupt/invalid) and then allows 3rd party flashing. It works at that point. No more lights, no more "proof of life" but wait 2-4 minutes, and check for a new 'sonoff-xxxx' network to appear. 

That's all I can think of at the moment. Keep us posted here if you don't mind.

[-=SrZ=-]

WilkyF18, 2019-04-03:
@SirRedZ a) worked!  I connected to vtrust-flash using a spare laptop instead of my iPhone and both the SP10s and one SP20 worked.  Flashed them to Tasmota.

Unfortunately, the other SP20 I can’t even get it into paring mode, the LED on it never flashes no matter how long I hold the power button. Makes no “clicking” sound.  I must have bricked it.  When I try “./start_flash.sh” never ending dots.  I also tried “curl http://10.42.42.42/flash3” but it says nothing can be found at that IP address.

SirRedZ, 2019-04-03:
when you first plug in the SP20, it gives a slight blue flicker right? And then never again, unless power removed and reinserted, and a flicker again?

SirRedZ, 2019-04-03:
If thats the case, it's likely in Stage1 (waiting for firmware push). That one.. get the dots going, and then plug it in. A couple of tries and it should recover. Play with it. It will go. (I hope)

[-=SrZ=-]

SirRedZ, 2019-04-03:
don't forget tor econnect to vtrust-flash each and everytime. The start script actually stops the last instance, so it's easy to miss cause devices falsely report the Wifi signal still present, even though it's gone. That would definitely have endless dots... once your sure vtrust is connected -- start the dots... THEN plug in the SP20.

[-=SrZ=-]

WilkyF18, 2019-04-03:
Yes, I get a very quick “purple” flash when plugged in (red and blue led?).  I’ll keep trying it.

kueblc, 2019-04-04:
Hi @SirRedZ 

Just a quick update, I was unable to reproduce your issues with the backup you provided. I'll keep trying and let you know if I find anything.

SirRedZ, 2019-04-04:
Thanks for the update. I appreciate you looking in to it.

[-=SrZ=-]

SirRedZ, 2019-04-08:
What a mystery! After spending many hours, I was finally able to complete the Teckin SP20 tuya-convert process. My best understanding is this module struggles with dirty power and likely uses inferior brdige rectification components, leaving it open to some unpredictable behaviors. Flashing is a very critical process, so if this is true, this would explain the erratic behaviour for me, while others report instant success. I have created a document of tricks & tips to help people recover their 'bricked' modules. 

Please note: This document references Whiskerz007 Promox Tuya-Convert script, although much of the content is applicable to any implementation of Tuya-Convert.

Once again, I must thank the creators/maintainers of Tuya-Convert. And Thanks goes to Whiskerz007 for putting in the hours to exhaust all options with me. I am now closing this issue.

[-=SrZ=-] / Tollbringer
[tuya-convert-help.pdf](https://github.com/ct-Open-Source/tuya-convert/files/3055188/tuya-convert-help.pdf)

closed 2019-04-08 by SirRedZ

shanerehm1234, 2019-05-12:
I struggled with this for half a day as well. What finally helped me was using the SSH built into the Windows 10 command line instead of putty. Don't know how or why, but I flashed 4 of them successfully once I made the switch.

SirRedZ, 2019-05-12:
Can you explain more, not sure how SSH fits @shanerehm1234 . (my lack of knowledge)

[-=SrZ=-]

shanerehm1234, 2019-05-12:
I wish I could. I struggled all day trying to flash a couple SP20 using putty and tuya-convert, but kept getting endless dots. When I closed putty and SSH to the pi using Windows 10 CLI instead, it worked. I didn't change anything else. I can't explain why, but it worked like a charm.

Scope666, 2020-06-19:
They must have pushed an even more aggressive update recently because I have two SP20's that are bricked.  I've flashed 3 other models with this rig successfully so I know it's not the PC / Wi-Fi dongle.  I've tried the hot plug after getting the dots going but no dice, it never shows up in the wi-fi log.