ct-Open-Source/tuya-convert #163: Device now has SSID of vtrust-recovery

BostonMikeGetHub, 2019-04-02:
I have successfully used Tuya convert on 5 various devices.  Something when wrong when I tried it on an AWP04L Power Monitoring Plug.  This device now has SSID of vtrust-recovery.  I get the below when I connect to it's web.  I'm using a raspberry pi.  Any advice?  Thanks. 

VTRUST-FLASH 1.0 
(c) VTRUST GMBH https://www.vtrust.de/35c3/ 

READ FLASH: http://10.42.42.42/get?read=HEX-ADDRESS 
ChipID: 6b2e46 
MAC: 5C:CF:7F:6B:2E:46 
SdkVersion: 1.5.3(aec24ac9) 
BootVersion: 4 
BootMode: 1 
FlashMode: 1M QIO @ 40MHz 
FlashChipId: 1440a1 
FlashChipRealSize: 1048576 
FlashChipSize: 1048576 
FlashChipMode: 0 
system_upgrade_userbin_check: user2 0x81000
 have successfully used Tuya convert on 5 various devices.  Something when wrong when I tried it on an AWP04L Power Monitoring Plug.  This device now has SSID of vtrust-recovery.  I get the below when I connect to it's web.  I'm using a raspberry pi.  Any advice?  Thanks. 

VTRUST-FLASH 1.0 
(c) VTRUST GMBH https://www.vtrust.de/35c3/ 

READ FLASH: http://10.42.42.42/get?read=HEX-ADDRESS 
ChipID: 6b2e46 
MAC: 5C:CF:7F:6B:2E:46 
SdkVersion: 1.5.3(aec24ac9) 
BootVersion: 4 
BootMode: 1 
FlashMode: 1M QIO @ 40MHz 
FlashChipId: 1440a1 
FlashChipRealSize: 1048576 
FlashChipSize: 1048576 
FlashChipMode: 0 
system_upgrade_userbin_check: user2

kueblc, 2019-04-02:
Unplug the device, run `start_flash` again, and plug it back in.

PS. You're using an older version of `tuya-convert`, if you have more devices to convert, consider pulling the latest version after you recover this device.

BostonMikeGetHub, 2019-04-02:
Thank you for your quick reply and for creating tuya-convert.  

That fixed it.  :)

I could not find a change log for the latest tuya-convert but I assume the changes is to make it more robust?

Anyway I will be updating to the one before I flash next device.

Thanks again.

closed 2019-04-02 by BostonMikeGetHub

kueblc, 2019-04-02:
Glad to hear it.

I'm working to make the process easier and more reliable. The latest version has a few bug fixes to prevent flashing issues and potential bricks, as well as improvements to the pairing process.