ct-Open-Source/tuya-convert #165: Only able to connect to flash one of two bulbs

eflemingtra, 2019-04-03:
Okay, got a strange one. Today I purchased two 'Brilliant Smart LED Bulbs' a generic Tuya bulb. They connect to the Tuya app when added manually.

Bulb one took many attempts to flash, but eventually did (I don't know why on the seventh attempt it worked, but it did). Unfortunately, no such luck with the second bulb.

It appears that tuya-convert is making a connection because once start_flash is begun and the bulb is in pairing mode, after about 10 seconds it goes out of pairing mode. So something is happening.

I have tried all the tricks I have read - connecting to vtrust-flash with a different device, pairing the device with the app and then removing it/resetting factory settings, starting vtrust-flash first and then putting the bulb into pairing mode. No good and as i said, the first bulb worked on the sixth or seventh attempt without doing anything different.

I have attached both web and wifi logs.
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3037512/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3037513/smarthack-wifi.log)

kueblc, 2019-04-03:
Can you please verify that the MQTT service is running?

ragenhe, 2019-04-03:
I just knocked out 2 Merkury bulbs in under 10 min.
I power on the bulb, while it's flashing, I run ./start_flashing. Within 5 seconds, it is paired and updating.  If it doesn't pair and start within 15 seconds. Run ./stop_flashing.  The ./start_flashing again.

eflemingtra, 2019-04-03:
> Can you please verify that the MQTT service is running?

Yes, it appears to be. I've attached the MQTT log.
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3041014/smarthack-mqtt.log)

> Run ./stop_flashing. The ./start_flashing again.

I will give that a go next.

kueblc, 2019-04-04:
@eflemingtra Try the following command with the device connected to your network.

`nc -u -l -k -p 6666`

If you see JSON output, this is good! If not, try

`nc -u -l -k -p 6667`

If you see random output, this is bad. Let us know!

eflemingtra, 2019-04-09:
Unfortunately I can't experiment on the second bulb, I appear to have bricked it perhaps. Again randomly, Tuya-Convert connected and it appeared to be backing up the default firmware, however, it seemed to hang and despite eventually saying the firmware was updated, turning on the bulb now results in a flash of blue color and then nothing.

kueblc, 2019-09-28:
Closing due to inactivity, feel free to open a new issue if you have any concerns

closed 2019-09-28 by kueblc