ct-Open-Source/tuya-convert #167: Issues running on RPI3 Kali 2019.1

jgaret, 2019-04-04:
Hello, I've successfully flashed a Maxcio YX-DE02-F wall plug (https://www.amazon.fr/Maxcio-Connect%C3%A9e-Intelligente-Compatible-Contr%C3%B4le/dp/B07GWN7BVH?th=1), but I needed some little changes to make it work (mainly some kind of problems due to setting up AP).
I would like to share what i had to do to make it work, if that helps anyone.

First, kali 2019.1 comes with some kind of "new-generation" iptables that just simply don't work with rpi3 kernel, so you needed to switch to legacy mode with update-alternatives
`root@kali:~/tuya-convert# update-alternatives --config iptables `

Secondly, there is also some kind of changes to allow the AP mode on RPI3. I needed to activate a monitor interface before running the start_flash.sh script
`iw phy phy0 interface add mon0 type monitor && ifconfig mon0 up`

kueblc, 2019-04-04:
Thanks for the feedback.

Would you be able to test out the `create-ap` branch and see if that works without modification?

`git clone -b create-ap https://github.com/kueblc/tuya-convert.git`

jgaret, 2019-04-05:
I think you forgot to commit a script. I get this in the logs: 

> ==> scripts/smarthack-wifi.log <==
> sudo: create_ap : command not found

kueblc, 2019-04-05:
You need to run `install_prereq` again to pick up the new dependency

jgaret, 2019-04-05:
After installing prereqs, I still have an issue with iptables-nft (needs to switch to iptables-legacy).
And hostapd does not allow my smartphone connecting with this error:
`wlan0: STA c0:c9:76:65:5a:aa IEEE 802.11: Could not add STA to kernel driver`

The complete startup log is:
`WARN: brmfmac driver doesn't work properly with virtual interfaces and
      it can cause kernel panic. For this reason we disallow virtual
      interfaces for your adapter.
      For more info: https://github.com/oblique/create_ap/issues/203
WARN: Your adapter does not fully support AP virtual interface, enabling --no-virt
Config dir: /tmp/create_ap.wlan0.conf.kLcb4p5e
PID: 1164
No Internet sharing
hostapd command-line interface: hostapd_cli -p /tmp/create_ap.wlan0.conf.kLcb4p5e/hostapd_ctrl
Configuration file: /tmp/create_ap.wlan0.conf.kLcb4p5e/hostapd.conf
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:cc:f2:0c and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED`

Log is the same if monitor interface is enabled

kueblc, 2019-04-07:
Thanks again for the feedback!

kueblc, 2019-10-04:
May be resolved by #225, if anyone can confirm we'll get this merged

kueblc, 2019-10-09:
Closing due to inactivity, feel free to open a new issue if you are still having problems after trying the `master` and `development` branches.

closed 2019-10-09 by kueblc