ct-Open-Source/tuya-convert #169: fake-registration-server.py - SyntaxError: invalid syntax

dennislv, 2019-04-10:
Hi I'm trying to start by flashing a single light bulb and after many hours of fighting wifi devices/drivers, I was finally able to get past the access point created and the device connecting to it.  I'm running into an issue with fake-registration-server.py.  
pi@raspberrypi:~/tuya-convert/scripts $ ./fake-registration-server.py
Traceback (most recent call last):
  File "./fake-registration-server.py", line 9, in <module>
    import tornado.web
  File "/usr/local/lib/python3.4/dist-packages/tornado/web.py", line 1655
    async def _execute(
            ^
SyntaxError: invalid syntax

dennislv, 2019-04-10:
[image]

kueblc, 2019-04-10:
Duplicate of #144

closed 2019-04-10 by dennislv