ct-Open-Source/tuya-convert #171: AISIRER bricked? sonoff-* access point unstable / device in boot loop(?)

m-reiter, 2019-04-11:
I've tried to run tuya-convert on 2 Aisirer smart plugs.

The second one worked without issues. The only surprise to me was that the falsh2 step reported the device was already booting from user2. The firmware backup is 7300 bytes.

The first device, however, stalled for quite a while (I'd guess around 20 minutes but I could be off) in the backup stage with received staying at 2920 bytes IIRC. I did a keyboard interrupt at that stage and moved on to the second device since the first one seemed unresponsive.

After having finished with the other device, I returned to the first one and plugged it in again. It seemed to reconnect to vtrust-flash. The flash2 step gave the same message as on device 2 and flash3 stated that the device should now flash the firmware and restart.

After that, a new sonoff-* access point showed up but I couldn't get a connection to it (time out). If since unplugged and replugged the device a couple of times and also tried holding the button for over 40 seconds.

Current state is that the sonoff-* access point shows up and I can connect to it. However, after about 5 seconds, the connection is lost again and the access point is no longer visible. The device gives off what sounds like a relay click, the access point reappears and the process repeats.

During one attempt I was directed to the Sonoff Wifi setup page but immediately lost connection again. I was not able to reproduce that.

I'm not sure whether this is a hardware problem, a tasmota issue or related to tuya-convert. I thought I'd go through the possibilities chronologically and start here.

Thanks a lot for any assistance!

[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3068404/smarthack-smartconfig.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3068406/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3068410/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3068413/smarthack-mqtt.log)

kueblc, 2019-04-11:
Tasmota is installed so this is a Tasmota issue. Best of luck!

m-reiter, 2019-04-11:
Thank you. I was expecting as much but couldn't rule out the possibility that something went wrong and the installation was not complete. I'll take this to Tasmota.

m-reiter, 2019-04-13:
Followup: I've since flashed three more of the same devices. One went flawlessly, the others showed almost the same behaviour. The only difference was that the cycle was slower which gave me just enough time to enter my wifi information after a couple of tries.

Interestingly (and the reason I post this here), for both problematic devices the tuya-convert download of the original firmware stalled at 2920 bytes (same as above) and eventually timed out. The firmware download for the "good" device went without issues.

Could this be a tuya-convert issue after all? In all three cases, the first irregularity showed before the new firmware was flashed.

kueblc, 2019-04-14:
Would you be able to share the backups you obtained?

m-reiter, 2019-04-15:
Sure, see attached zip file.

the files have time stamps. The order of flash attempts was
1. failed attempt, backup stalled at 2920 bytes, fw file is 0 bytes
2. successful flash, backup stalled as well but later (don't remember the count, sorry), fw file is 7300 bytes
3. successful flash, backup finished without problems, fw file is 1M
4. and 5. failed attempts, backup stalled at 2920 bytes again, fw file are 1460 and 2920 bytes.
[fw.zip](https://github.com/ct-Open-Source/tuya-convert/files/3080087/fw.zip)

kueblc, 2019-11-21:
Unfortunately I was never able to reproduce this issue. The core cause may have been resolved in a more recent release. Perhaps related to dirty flash memory resolved with #376.

Closing due to age, feel free to comment and/or reopen if it can be reproduced with a recent release. Thanks again for your diligence in reporting to improve our project.

closed 2019-11-21 by kueblc