ct-Open-Source/tuya-convert #172: vtrust-flash wifi not showing up

MarkLurie90, 2019-04-13:
Hi All,

I've been tring to flash my Tyta plug (FrankEver):
https://www.aliexpress.com/item/FrankEver-Tuya-Cloud-10A-16A-Israel-WiFi-Smart-Socket-Wireless-Plug-Work-With-Alexa-Google-Assistant/32979207234.html?spm=a2g0s.9042311.0.0.27424c4dEdXr5S

The flash process is Kali-Linux 2018.4 in VMWARE with seperate wifi dongle.
I got to the stage that the AP need to be created, but the vtrust-flash wifi never appears (tried on several devices).

Can anyone assist please?

Thanks!
Mark

maxtheking, 2019-04-13:
I never got it to work using a Virtual Machine with a Wifi dongle passed through to it. The vtrush-flash network would show up, i could also connect to it but the flashing wouldn't actually go through. When using a laptop with Ubuntu it works just fine!

melikyuksel, 2019-04-14:
vtrust-flash seems to only be able to be deployed on wlan0. Make sure your Wi-Fi dongle's interface name is wlan0. If you track all the logs in tuya-convert/scripts, it'll tell you why the the AP isn't coming up. Also make sure to kill wpa_supplicant if it's running.

MarkLurie90, 2019-04-14:
Hi,

Tried to kill "wpa_supplicant" and started the procedure again:

root@kali20184:~# kill `pidof wpa_supplicant`
root@kali20184:~# cd tuya-convert
root@kali20184:~/tuya-convert# ./start_flash.sh
~/tuya-convert/scripts ~/tuya-convert
======================================================
TUYA-CONVERT

https://github.com/ct-Open-Source/tuya-convert
TUYA-CONVERT was developed by Michael Steigerwald from the IT security company VTRUST (https://www.vtrust.de/) in collaboration with the techjournalists Merlin Schumacher, Pina Merkert, Andrijan Moecker and Jan Mahn at c't Magazine. (https://www.ct.de/)

======================================================
PLEASE READ THIS CAREFULLY!
======================================================
TUYA-CONVERT creates a fake update server environment for ESP8266/85 based tuya devices. It enables you to backup your devices firmware and upload an alternative one (e.g. ESPEasy, Tasmota, Espurna) without the need to open the device and solder a serial connection (OTA, Over-the-air).
Please make sure that you understand the consequences of flashing an alternative firmware, since you might lose functionality!

Flashing an alternative firmware can cause unexpected device behavior and/or render the device unusable. Be aware that you do use this software at YOUR OWN RISK! Please acknowledge that VTRUST and c't Magazine (or Heise Medien GmbH & Co. KG) CAN NOT be held accountable for ANY DAMAGE or LOSS OF FUNCTIONALITY by typing yes + Enter

yes
======================================================
  Starting AP in a screen
  Stopping any apache web server
  Starting web server in a screen
  Starting Mosquitto in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   The wpa-password is flashmeifyoucan
   This step is IMPORTANT otherwise the smartconfig will not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
3. Press ENTER to continue

-----------------------------------------------------------------------------------

Above is the log from the terminal, no AP started.
P.S.: my W-Fi adapter connected via "Removeable devices".

Thanks!

melikyuksel, 2019-04-15:
@MarkLurie90 This is the output of the application, not the logs. Please run `tail -f ./scripts/*.log` from the `tuya-convert` folder and paste that.

yoarch, 2019-04-18:
Hi,
I tried several time on archlinux, debian, ubuntu on 3 different laptop. Can not get the AP. None of my 3 laptops have a wlan0 interface.
Is it possible to get it to work on a different wlp interface?
Thanks for your time.

melikyuksel, 2019-04-18:
@yoarch As of now, you have to rename your wireless interface to wlan0.

```
ip link set dev <interface> down
ip link set <interface> name wlan0
```

Then run tuya-convert.

MarkLurie90, 2019-04-20:
> @MarkLurie90 This is the output of the application, not the logs. Please run `tail -f ./scripts/*.log` from the `tuya-convert` folder before you run `start_flash.sh`, wait until it hangs, and then paste the output of the tail command.

Hi, is this what you ment?
root@kali20184:~# cd tuya-convert
root@kali20184:~/tuya-convert# tail -f ./scripts/*.log
==> ./scripts/smarthack-mqtt.log <==
1555171038: Opening ipv4 listen socket on port 1883.
1555171038: Opening ipv6 listen socket on port 1883.
1555174375: mosquitto version 1.5.7 starting
1555174375: Using default config.
1555174375: Opening ipv4 listen socket on port 1883.
1555174375: Opening ipv6 listen socket on port 1883.
1555253209: mosquitto version 1.5.7 starting
1555253209: Using default config.
1555253209: Opening ipv4 listen socket on port 1883.
1555253209: Opening ipv6 listen socket on port 1883.

==> ./scripts/smarthack-smartconfig.log <==
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
Traceback (most recent call last):
  File "./smartconfig/main.py", line 23, in <module>
    smartconfig( passwd, ssid, region, token, secret )
  File "/root/tuya-convert/scripts/smartconfig/smartconfig.py", line 46, in smartconfig
    sock = SmartConfigSocket()
  File "/root/tuya-convert/scripts/smartconfig/smartconfig.py", line 29, in __init__
    self._socket.bind((address, 0))
OSError: [Errno 99] Cannot assign requested address

==> ./scripts/smarthack-web.log <==
Listening on port 80
Listening on port 80
Listening on port 80
Listening on port 80
Listening on port 80

==> ./scripts/smarthack-wifi.log <==
Starting DNSMASQ server...
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlan0 wasn't started

MarkLurie90, 2019-04-20:
@melikyuksel 
the output after running "./start_flash.sh":
root@kali20184:~# cd tuya-convert
root@kali20184:~/tuya-convert# tail -f ./scripts/*.log
==> ./scripts/smarthack-mqtt.log <==
1555171038: Opening ipv4 listen socket on port 1883.
1555171038: Opening ipv6 listen socket on port 1883.
1555174375: mosquitto version 1.5.7 starting
1555174375: Using default config.
1555174375: Opening ipv4 listen socket on port 1883.
1555174375: Opening ipv6 listen socket on port 1883.
1555253209: mosquitto version 1.5.7 starting
1555253209: Using default config.
1555253209: Opening ipv4 listen socket on port 1883.
1555253209: Opening ipv6 listen socket on port 1883.

==> ./scripts/smarthack-smartconfig.log <==
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
Traceback (most recent call last):
  File "./smartconfig/main.py", line 23, in <module>
    smartconfig( passwd, ssid, region, token, secret )
  File "/root/tuya-convert/scripts/smartconfig/smartconfig.py", line 46, in smartconfig
    sock = SmartConfigSocket()
  File "/root/tuya-convert/scripts/smartconfig/smartconfig.py", line 29, in __init__
    self._socket.bind((address, 0))
OSError: [Errno 99] Cannot assign requested address

==> ./scripts/smarthack-web.log <==
Listening on port 80
Listening on port 80
Listening on port 80
Listening on port 80
Listening on port 80

==> ./scripts/smarthack-wifi.log <==
Starting DNSMASQ server...
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlan0 wasn't started
Backing up NetworkManager.cfg...
Restarting NetworkManager...
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlan0 wasn't started

==> ./scripts/smarthack-web.log <==
Listening on port 80

==> ./scripts/smarthack-mqtt.log <==
1555770591: mosquitto version 1.5.7 starting
1555770591: Using default config.
1555770591: Opening ipv4 listen socket on port 1883.
1555770591: Opening ipv6 listen socket on port 1883.

demian85, 2019-04-23:
I'm having the same issut, but the AP seems to be enabled. I'm trying to connect from an Android device. I added the vtrust-flash hidden network with the specified WPA password. SHould I run the script as root?

```==> ./scripts/smarthack-wifi.log <==
Applying iptables rules...
Starting DNSMASQ server...
RTNETLINK answers: File exists
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:b3:e2:d6 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
```

yoarch, 2019-04-25:
Same issue for me. Thanks _melikyuksel_, I changed the interface for wlan0 and solved several issues (arch compatibility). But now I am at the point:
```
Starting DNSMASQ server...
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED
wlan0: CTRL-EVENT-TERMINATING
hostapd_free_hapd_data: Interface wlan0 wasn't started
```

Going to dig on that today. If I can get the whole thing to work I will update the script to have an arch compatibility..

melikyuksel, 2019-04-25:
@yoarch Did you do `killall wpa_supplicant` before starting the script? Also, did you run the install script before running the main script? I had this exact problem in both those cases.

kueblc, 2019-04-25:
Also worth trying an alternative branch based on @oblique's `create_ap` with `git clone -b create-ap https://github.com/kueblc/tuya-convert.git`. Note that you'll have to run `install_prereq.sh` again.

yoarch, 2019-04-26:
Hi @kueblc! How are you?
So, @melikyuksel, let me explain you a little my situation. I have two smartplugs (with **TYWE3S** wifi controller, probably _new_ firmware from Tuya) and tried (with @kueblc help) to flash it through serial.

I did not succeed and basically destroyed it, (not turning on at all now), very hard to perform solder on it.

Before trying serial I tried to flash it OTA with **codetheweb/tuyapi**. But we finally observed that the UDP packets were uncrypted, @kueblc told me that this is the new firmware, and we have currently no solution to flash it OTA.

**But**, I have another one, I did not succeed first time to perform the **tuya-convert** process, and I am a little **stubborn**. So I want to try and fail on your tuya_convert process before to resign, sorry @kueblc.
I made some progress, after a long dig I am able to create the access point on my personal laptop. The other one I tried on could not, apparently, switch to AP mode. 'iw list | grep AP' did not return anything.
So, now I have a AP using hostapd and dnsmasq.

I read the **tuya-convert** code several times, I am little confused about what I should do now when I have a AP on my laptop, my smartphone connected to it and my smartplug flickering quickly. That part is not very clear in the code to me. Could you tell me what to do next? I guess I want my smartplug to connect to my AP, right?

Thanks for your time.

yoarch, 2019-04-26:
I just tried your suggestion @kueblc with @oblique's create_ap.
I made some progress. Been able to connect my smartphone and the **smartplug** to the AP with create_ap and some modifications of **main.py** and **smartconfig.py**:
```
$ create_ap --list-clients ap0
MAC                  IP                 Hostname
98:xx:xx:xx    10.42.42.38        phone
68::xx:xx:xx    10.42.42.36        ESP_B31F49
```
I can observe that its IP is not 10.42.42.**42** but 10.42.42.**36**.
And:
```
curl -JO https://10.42.42.36/backup
curl -JO http://10.42.42.36/backup
curl -JO --ipv4 http://10.42.42.36/backup
```
all return:
`curl: (7) Failed to connect to 10.42.42.36 port 80: Connection refused`

```
$ nmap 10.42.42.36
Nmap scan report for 10.42.42.36
Host is up (0.015s latency).
Not shown: 999 closed ports
PORT     STATE SERVICE
6668/tcp open  irc

Nmap done: 1 IP address (1 host up) scanned in 8.56 seconds
```

What do you think? Not listening on port 80, neither 443 is bad sign and confirm that I won't be able to do anything OTA?
Should I try something like:
`curl -JO http://10.42.42.36:6668/backup`
The only listenning port?

Thanks for your time.

kueblc, 2019-04-26:
The `curl` commands are there to interact with the intermediate firmware, which is loaded OTA by the `tuya-convert` process. Once the intermediate is loaded, it will switch to 10.42.42.42. You are still running the Tuya firmware, as the port 6668 is for the Tuya protocol, so `curl` isn't received by anything.

Unfortunately if you are running the newer firmware with TLS, `tuya-convert` won't be able to load the intermediate and none of the intermediate functions will work.

yoarch, 2019-04-26:
Ok, kind of frustrating.
Also, I don't get any log as I run **main.py** directly with python. So, everything looks to be good, I even get a connexion to the AP, but if I understand you well, when running this line:
`python smartconfig/main.py`
I should get out a connexion with the 10.42.42.**42** IP address, right? So, we can (one more time) conclude, regarding that observation, that I have the newer firmware with TLS?

kueblc, 2019-04-26:
> So, everything looks to be good, I even get a connexion to the AP, but if I understand you well, when running this line:
> `python smartconfig/main.py`

`smartconfig` tells the Tuya firmware to connect to your AP and nothing else. The next step is assigning keys and installing the intermediate, but the new firmware has blocked us here.

> I should get out a connexion with the 10.42.42.**42** IP address, right?

No, Tuya firmware connects to any IP address it can. It is the intermediate firmware that will request 10.42.42.42. Since you haven't loaded the intermediate, it won't necessarily connect to that address, nor will it understand any `curl` commands.

> So, we can (one more time) conclude, regarding that observation, that I have the newer firmware with TLS?

I wish it wasn't so, but yes, it looks like you won't be able to use `tuya-convert` with this device as of now. We're working on this, but no promises when or if we succeed in developing a workaround.

demian85, 2019-04-26:
Can someone please specify which is the new firmware version that does not
work?

I own a "Lumiman" light bulb and in the Smart Life app it says:
Wi-Fi module: 1.3.1
MCU module: 1.3.1

It is supposed to be the latest firmware release. Should I still try to
flash the firmware with this script?

kueblc, 2019-04-26:
@demian85 it depends on the device, there is no one firmware version guaranteed to have the TLS update. No harm in trying `tuya-convert`, as it will simply not proceed if the firmware is too new.

yoarch, 2019-04-27:
Thanks @kueblc for taking the time to answer.
I am confused. The
`$screen_with_log smarthack-smartconfig.log -S smarthack-smartconfig -m -d ./smartconfig/main.py`
or
`python smartconfig/main.py` in my case is just supposed to **get** the connexion to the AP. So, until this point I am good, I am connected.

But I can not find any code **loading the intermediate firmware** after `python smartconfig/main.py` as the next lines are:
```
echo "Waiting for the upgraded device to appear"
echo "If this does not work have a look at the '*.log'-files in the 'scripts' subfolder!"

while ! timeout 0.2 ping -c 1 -n 10.42.42.42 &> /dev/null
```
Can you tell me what I am missing. What code is supposed to load the intermediate firmware?
Thanks.

yoarch, 2019-04-27:
I made some progress. I got another smartplug that is in the [compatible devices](https://github.com/ct-Open-Source/tuya-convert/wiki/Compatible-devices) list, the **TECKIN SP21** (should work ..)

I made lots of tests on the code (still on arch) and was able to run start_flash.sh with two main differences:
- I changed **wlan0** for **wlp59s0** in config.txt
- **create_ap** doesn't have a **-e** options for the **tuya_hosts**, so I removed it as I didn't find any equivalent in the current create_ap options.

The process goes well and get stuck at:
```
Waiting for the upgraded device to appear
If this does not work have a look at the '*.log'-files in the 'scripts' subfolder!
............................................................................
```

I got these logs (no errors):
```
PID: 18877
wlp59s0 is already associated with channel 1 (2412 MHz)
Creating a virtual WiFi interface... ap0 created.
No Internet sharing
hostapd command-line interface: hostapd_cli -p /tmp/create_ap.wlp59s0.conf.GemqfVaT/hostapd_ctrl
Configuration file: /tmp/create_ap.wlp59s0.conf.GemqfVaT/hostapd.conf
Using interface ap0 with hwaddr 9c:b6:d0:c2:c6:a8 and ssid "xxxxx"
ap0: interface state UNINITIALIZED->ENABLED
ap0: AP-ENABLED 
ap0: STA 98:0c:82:f2:2a:2d IEEE 802.11: authenticated
ap0: STA 98:0c:82:f2:2a:2d IEEE 802.11: associated (aid 1)
ap0: AP-STA-CONNECTED 98:0c:82:f2:2a:2d
ap0: STA 98:0c:82:f2:2a:2d RADIUS: starting accounting session B5F474B2B68DD78B
ap0: STA 98:0c:82:f2:2a:2d WPA: pairwise key handshake completed (RSN)
ap0: STA cc:50:e3:21:2f:15 IEEE 802.11: authenticated
ap0: STA cc:50:e3:21:2f:15 IEEE 802.11: associated (aid 2)
ap0: AP-STA-CONNECTED cc:50:e3:21:2f:15
ap0: STA cc:50:e3:21:2f:15 RADIUS: starting accounting session D23E66E9C8031ABC
ap0: STA cc:50:e3:21:2f:15 WPA: pairwise key handshake completed (RSN)
ap0: AP-STA-DISCONNECTED cc:50:e3:21:2f:15
ap0: STA cc:50:e3:21:2f:15 IEEE 802.11: disassociated
ap0: STA cc:50:e3:21:2f:15 IEEE 802.11: disassociated
ap0: STA cc:50:e3:21:2f:15 IEEE 802.11: disassociated
ap0: STA cc:50:e3:21:2f:15 IEEE 802.11: disassociated
ap0: STA cc:50:e3:21:2f:15 IEEE 802.11: disassociated
ap0: STA cc:50:e3:21:2f:15 IEEE 802.11: disassociated
ap0: STA cc:50:e3:21:2f:15 IEEE 802.11: disassociated
ap0: STA cc:50:e3:21:2f:15 IEEE 802.11: disassociated
ap0: STA cc:50:e3:21:2f:15 IEEE 802.11: disassociated
ap0: STA cc:50:e3:21:2f:15 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
ap0: STA cc:50:e3:21:2f:15 IEEE 802.11: authenticated
ap0: STA cc:50:e3:21:2f:15 IEEE 802.11: associated (aid 2)
ap0: AP-STA-CONNECTED cc:50:e3:21:2f:15
ap0: STA cc:50:e3:21:2f:15 RADIUS: starting accounting session AC4637047F76CA27
ap0: STA cc:50:e3:21:2f:15 WPA: pairwise key handshake completed (RSN)
```

and I get my smartplug connected:
`cc:50:e3:21:2f:15    10.42.42.97        ESP_XXXX15`

So, why the process doesn't finish? Is it the interface (must be wlan0)? Or is it normal that it doesn't work as the option `-e /xxxx/tuya-convert/scripts/tuya_hosts` is not specified?

This is the help of create_ap:
```
Usage: create_ap [options] <wifi-interface> [<interface-with-internet>] [<access-point-name> [<passphrase>]]

Options:
  -h, --help              Show this help
  --version               Print version number
  -c <channel>            Channel number (default: 1)
  -w <WPA version>        Use 1 for WPA, use 2 for WPA2, use 1+2 for both (default: 1+2)
  -n                      Disable Internet sharing (if you use this, don't pass
                          the <interface-with-internet> argument)
  -m <method>             Method for Internet sharing.
                          Use: 'nat' for NAT (default)
[... 38 lines cut ...]
  -d                      DNS server will take into account /etc/hosts

Useful informations:
  * If you're not using the --no-virt option, then you can create an AP with the same
    interface you are getting your Internet connection.
  * You can pass your SSID and password through pipe or through arguments (see examples).
  * On bridge method if the <interface-with-internet> is not a bridge interface, then
    a bridge interface is created automatically.

Examples:
  create_ap wlan0 eth0 MyAccessPoint MyPassPhrase
  echo -e 'MyAccessPoint\nMyPassPhrase' | create_ap wlan0 eth0
  create_ap wlan0 eth0 MyAccessPoint
  echo 'MyAccessPoint' | create_ap wlan0 eth0
  create_ap wlan0 wlan0 MyAccessPoint MyPassPhrase
  create_ap -n wlan0 MyAccessPoint MyPassPhrase
  create_ap -m bridge wlan0 eth0 MyAccessPoint MyPassPhrase
  create_ap -m bridge wlan0 br0 MyAccessPoint MyPassPhrase
  create_ap --driver rtl871xdrv wlan0 eth0 MyAccessPoint MyPassPhrase
  create_ap --daemon wlan0 eth0 MyAccessPoint MyPassPhrase
  create_ap --stop wlan0
```
Do you see an equivalent to specify the tuya_hosts?

OMT, thanks for your time.

kueblc, 2019-04-29:
>     * I changed the AP and password (should not be a problem)

The intermediate firmware is currently hardcoded to these values, so it would not work if you change them.

>     * **create_ap** doesn't have a **-e** options for the **tuya_hosts**, so I removed it as I didn't find any equivalent in the current create_ap options.

It should have `-e`. Not sure why your help dialog doesn't contain it, but it is certainly present in the github version of `create_ap`. This is what allows us to intercept requests to the Tuya cloud, so without it, we won't capture any requests.

> What code is supposed to load the intermediate firmware?

`fake_registration_server.py` will respond to the device's upgrade request, and then serve the intermediate firmware when the device requests it.

Note that the firmware is installed by the device only if it agrees to, as such newer firmware will not work since they have included a new signing mechanism and communicates over HTTPS instead of HTTP.

yoarch, 2019-04-29:
Thanks a lot kueblc for the answer. I will try all of that tomorrow.
I am a little concerned, you are saying that even if a device is in the supported devices list, if I purchase it now, they could have changed the firmware without changing the model? Like the TECKIN SP21 for example?

kueblc, 2019-04-29:
> I am a little concerned, you are saying that even if a device is in the supported devices list, if I purchase it now, they could have changed the firmware without changing the model? Like the TECKIN SP21 for example?

Yes, they can release an update at any time that will block `tuya-convert`

yoarch, 2019-05-07:
@kueblc. I am back with news.

First, thanks for all your answers, very helpful.

I have redone all the steps with:
- **create_ap-git** from **git** working with the **-e** option. You were right. However, I don't understand why it works as **create_ap** and **create_ap-git** seem both to point on the same git repo.. Anyway

- let the default ap name and password

- change the interface from wlan0 to wlp59s0

and.. it works ! With some difficulties, I finally got all my new four **TECKIN SP21** devices flashed to tasmota 6.5.0. Thanks a lot.
I tried 10 times for my old **S126 Tuya_smart**, never worked. So, OMT, you were right. I think we can conclude that the TECKIN SP21 still have the old/good firmware to be flashed (bought on amazon on april 2019), if you want to notice it in the wiki.
Do you want the TECKIN SP21 firmware that I backuped? I don't think so.

Thanks a lot for your help, now I can run MQTT commands from my laptop! I am just happy.

I know this is out of the topic but now I have some more question points (if not the right place, please tell me):
- I tried to combine TLS with MQTT but got at the conclusion that the tasmota firmware doesn't support it and is not possible with any of the available tasmota firmware project, is it right?
- when the devices are not connected to a network they are blinking blue. I want them to not blink during the night as I turn everything off. How can I get them to not blink when not connected. I guess the solution is in the direction of the **template**? Right now I use the **BlitzWolf SHP** template. That one is displaying the consumption but the devices don't support it and it blinks when not connected. If I can manage that with the template, I can work on it to make a new one, more relevant. Do you have good documentations to make a template please?
- I searched everywhere how to run a MQTT command to turn on or off or toggle my switch with a time delay. Like turn off in 1 minute. Something like:
`mosquitto_pub -h 192.168.xx.xxx -t cmnd/test/POWER -m 2`
Do you have any suggestion?
- Also, I can not find a command just returning me the current state of the switch device (ON or OFF) and returning the prompt (not keeping listening). Do you have a suggestion?

Thanks a lot.

demian85, 2019-05-07:
Tasmota supports TLS. You just need to enable some flags before
compilation.

kueblc, 2019-05-14:
@yoarch Glad to hear of your successes. You can check the power state of your device by MQTT if you enable the power retain option. The rest of your questions are best addressed by the Sonoff-Tasmota group. Good luck!

yoarch, 2019-05-20:
Ok, thanks @demian85 and @kueblc for your answers. I finally got everything to work (MQTT, android MQTT client). Now I understand that some confs can be set from compiling and MQTT and Webserver, others only by compiling (SSL for MQTT). So, it's important to read several time the doc.
That is very pleasant to take the control back of the devices.

I made a mistake on one of mine that was running well with the last tasmota. I compiled the firmware but changed the `#define WEB_SERVER` from 2 to 0. Then I was not able to connect throught Web server (normal) and so was not able to reflash it (no web ui). So, I pressed the button for a long time (more than 40s) until a click and then the device was not doing anything. Now I have a device that is not blinking at all and I don't know how to reflash it or reset it. I guess the default module type is not working with my device and it doesn't have a web server. However, it is connecting to my network.
What should I do to reset it and be able to flash it one more time in this situation (after pressing a long time the button)?
Thanks.

ministryofsillywalks, 2019-05-25:
I'm also having issue with the AP showing up.
I had to rename my wifi interface as it doesn't show up as wlan0
`ip link set wlx90f652e2cd41 name wlan0`
After that it shows up as wlan0 however its disabled.
So:
`ifconfig wlan0 up`
after that the interface is running on wlan0
however starting tuya convert never gets to a point where the AP shows up.
Heres the error log smarthack-wifi.log:
`Backing up NetworkManager.cfg...
Restarting NetworkManager...
SIOCSIFFLAGS: Operation not possible due to RF-kill
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
SIOCSIFFLAGS: Operation not possible due to RF-kill
SIOCSIFFLAGS: Operation not possible due to RF-kill
Applying iptables rules...
Starting DNSMASQ server...

dnsmasq: failed to create listening socket for port 53: Address already in use
Error: Device for nexthop is not up.
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
rfkill: WLAN soft blocked
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr 90:f6:52:e2:cd:41 and ssid "vtrust-flash"
Failed to set beacon parameters
wlan0: Could not connect to kernel driver
Interface initialization failed
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: Unable to setup interface.
wlan0: interface state DISABLED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
`

Any ideas?

yoarch, 2019-05-25:
I tried that at the beginning too. But finally I solved this by directly changing the interface name in the tuya-convert/config.txt file:
`WLAN=wlpXXsX`

edit: don't try to change something else in this conf file.

ministryofsillywalks, 2019-05-25:
Ok great that worked. The AP is showing.
However trying to connect my android to the AP gives me no IP-Address (stuck at "Getting IP Address)
Is this normal?

My Blitzwolf SHP06 is in the connect mode and flashing but stops flashing shortly after starting the script. I'm guessing I might be unlucky enough to have a version with the patched firmware :(

Logs
wifi.log:
https://pastebin.com/dkKvHR8x
web.log
https://pastebin.com/v0WQMDcS
smartconfig.log
https://pastebin.com/YeXUeEnJ
mqtt.log
https://pastebin.com/TExYVpWC

yoarch, 2019-05-25:
Good that you made progress.

No, you should get an IP from your laptop AP.

The tuya flash won't work if your smartphone (or another device) is not connected to your laptop AP. So, first we have to figure out why you don't get an IP on your smartphone.

Look at the AP logs and confs to see if there is something wrong.

ministryofsillywalks, 2019-05-25:
where do I find the AP logs?
Like I said the AP shows up but my phone just stays at "getting IP address" and never actually progresses from there...

yoarch, 2019-05-25:
Which package do you use to create the ap? What is your distro?
Try with **create_ap** from:
`git clone -b create-ap https://github.com/kueblc/tuya-convert.git`
See if your **ap** works with this package.

kueblc, 2019-05-25:
> My Blitzwolf SHP06 is in the connect mode and flashing but stops flashing shortly after starting the script. I'm guessing I might be unlucky enough to have a version with the patched firmware :(

@ministryofsillywalks Looks like you paired successfully, since it stopped blinking. There's nothing wrong with your AP setup. Unfortunately Blitzwolf SHP6 has been reported to have the patched firmware. See #166.

ministryofsillywalks, 2019-05-25:
@kueblc so you say its normal that my android phone doesn't actually get an IP from the AP?

@yoarch I used the standard package from the main page. Distro is linux mint.
will try the package you posted

yoarch, 2019-05-25:
I believe I did have a similar issue. And it was related to the process running the ap. I am on archlinux so I did have to adjust the scripts, I guess they are more for debian/ubuntu which have a different package manager. I finally run well the ap with the create_ap package. And then I was able to flash it ota with the create-ap branch of the tuya-convert.
I think you might have two issues, the first one is that you can not connect your android phone to your laptop ap, this has nothing to see with tuya-convert, I would solve that first. The second one, and probably the worse is that your device might have the patched firmware (as @kueblc said). In this case you will have to flash it through serial (if you are brave..).

MathijsG, 2019-06-26:
@ministryofsillywalks just check the logs generated in scripts. Must be something wrong, and it will be stated there. The phone should get an IP from the server.

kueblc, 2019-09-28:
Closing due to inactivity, feel free to open a new issue if you have any concerns.

For those that missed it, support for HTTPS firmware is in the works #279

closed 2019-09-28 by kueblc

raulvasquez, 2019-10-04:
> I believe I did have a similar issue. And it was related to the process running the ap. I am on archlinux so I did have to adjust the scripts, I guess they are more for debian/ubuntu which have a [...]

@yoarch any chance of creating a branch or sharing the changes for your arch compatible tuya-convert?

scy0846, 2019-10-04:
I'm assuming something changed that broke this in the last few days because I'm using a Raspberry pi with Raspian and it will turn on the AP for a second and then it turns off. It doesn't stay on long enough to connect just long enough to connect it. This is a freshly installed Raspberry pi. Brand new out of the box did the install_prereqs and still nothing

kueblc, 2019-10-04:
Nothing in `setup_ap` was touched in the last 8 months, at least in `master`. Are you running a fork/branch?

MathijsG, 2019-10-06:
I always need to kill wpa_supplicant, but after that it works perfectly.

scy0846, 2019-10-07:
Kueblc helped me in another issue. There was a PR I grabbed from him that looks to have fixed my issue.

RestOp, 2019-10-26:
Hello,

the network do not start. Here is the log http://prntscr.com/poerrf
Why can it be?

Thanks!

kueblc, 2019-10-26:
@RestOp please open a new issue rather than commenting on this closed one. When you do, be sure to include your full logs as `tail` will only show you the last 10 lines by default.

minax007, 2022-04-17:
Dear all, 

@Raumzeit shared following information, which has resolved my connection issue with vtrust-flash:

https://github.com/ct-Open-Source/tuya-convert/issues/892#issuecomment-1100248239