ct-Open-Source/tuya-convert #173: Does tuya-convert/update.bin erase the flash before flashing the new firmware?

m-reiter, 2019-04-14:
This is related to #171: While investigating the issues I had (with three devices and two different tasmot versions) with the help of the SonoffUsers group, [the question arose](https://groups.google.com/d/msg/sonoffusers/UBcempVJkNQ/7u2kQPaEAwAJ) whether tuya-convert does erase the flash before flashing the new binary.

Since I didn't find any sources for update.bin (BTW, are those available anywhere?) I'm posing this queston here.

kueblc, 2019-04-14:
> This is related to #171: While investigating the issues I had (with three devices and two different tasmota versions) with the help of the SonoffUsers group, [the question arose](https://groups.google.com/d/msg/sonoffusers/UBcempVJkNQ/7u2kQPaEAwAJ) whether tuya-convert does erase the flash before flashing the new binary.

`flash2` only erases sectors 128-256 (0-0x80000), `flash3` only erases 0-128 (0x80000-0x100000). If this is causing issues, I propose we do a full erase on `flash3`. Would this work for you?

I'm still not sure what could have caused the firmware backup to stall.

> Since I didn't find any sources for update.bin (BTW, are those available anywhere?) I'm posing this queston here.

https://github.com/vtrust-de/esp8266-ota-flash-convert

m-reiter, 2019-04-15:
> `flash2` only erases sectors 128-256 (0-0x80000), `flash3` only erases 0-128 (0x80000-0x100000). If this is causing issues, I propose we do a full erase on `flash3`. Would this work for you?

I don't know if it will work but I'm willing to give it a try. I have one device with original firmware remaining.

> https://github.com/vtrust-de/esp8266-ota-flash-convert

Thank you!

kueblc, 2019-04-22:
Let me know how it went for you, if I should implement the change.

m-reiter, 2019-04-22:
Sorry, there might be a slight misunderstanding here: I thought I was waiting for you to implement a full erase so I could try it out. Seems like I was mistaken. Could you please clarify?

kueblc, 2019-04-25:
Yes, misunderstanding, sorry about that. I had a minute so I've compiled a binary with the change. Please note I have not done any testing whatsoever.

[build.zip](https://github.com/ct-Open-Source/tuya-convert/files/3118730/build.zip)

m-reiter, 2019-04-27:
Thanks. Unfortunately I have misplaced the SD card I had tuya-convert installed on. I could try the new build on different hardware (x86 vs. Raspi) but that would be changing more than on parameter at a time. I think I'll give myself a little time to try and find the missing card first.

One question though: your zip file contains a file user1.bin that is not present in the repository. What's the purpose of that file?

kueblc, 2019-04-29:
> One question though: your zip file contains a file user1.bin that is not present in the repository. What's the purpose of that file?

`user1.bin` and `user2.bin` are both used to create `upgrade.bin`. You can safely ignore it. They are two versions of the same program, compiled for each userspace. The device will select the one it needs automatically.

m-reiter, 2019-05-01:
Tried it again on a new SD card.

Backup stalled again. I did about seven retries using `curl http://10.42.42.42/backup` but the best I could get was 92%.

I then did `curl http://10.42.42.42/flash2` and again got the message that the device was already booting from userspace 2.

The flash3 stage seemed to work then:
```
pi@raspberrypi:~/tuya-convert$ curl http://10.42.42.42/flash3
Device should flash http://10.42.42.1/files/thirdparty.bin and restart
```
but the device's AP never showed up. Upon further investigation, it turned out that the device was still on the vtrust-flash WiFi. It still answers to queries as well:
```
pi@raspberrypi:~/tuya-convert$ curl http://10.42.42.42
VTRUST-FLASH 1.1
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: 609794
MAC: B4:E6:2D:60:97:94
BootVersion: 4
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144068
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000
```
I retried the flash3 step once but the device is still in the same state.

Any ideas?

m-reiter, 2019-05-08:
Is anybody still monitoring this?

kueblc, 2019-05-08:
> Is anybody still monitoring this?

Yes. Unfortunately nothing new from me. Keeping it in mind though.

m-reiter, 2019-05-08:
Any idea on the state the device is in at the moment? It behaves as if it's ready to flash but doesn't.

Do you have any suggestion what I could try to get it to flash yet? I assume it's too late for an undo now?

kueblc, 2019-05-08:
I'm at a bit of a loss. If you manage to get a backup out of the device I might be able to reproduce it here, giving me a bit more info on what's going on under the hood.

Another thought, can you confirm that it's requesting the firmware file when you do `/flash3`? You should see it in the web log.

> I assume it's too late for an undo now?

If it's still running the intermediate (connects to `vtrust-flash` at 10.42.42.42) you should still be able to `/undo`.

> P.S.: Wow, looks like I just read a message from the future. ;)

I see that! No idea how that happened

m-reiter, 2019-05-08:
> I'm at a bit of a loss. If you manage to get a backup out of the device I might be able to reproduce it here, giving me a bit more info on what's going on under the hood.

Okay, I'll give it another try. Alternatively, if you're interested in investigating this further, I could also send the device to you.

> Another thought, can you confirm that it's requesting the firmware file when you do `/flash3`? You should see it in the web log.

Yes, it seems like the device is indeen requesting the firmware. I just tried again and it seems to be requesting the firmware 3 times:

```
[I 190508 21:23:28 web:2246] 200 GET /files/thirdparty.bin (10.42.42.42) 27.23ms
[I 190508 21:23:28 web:2246] 200 GET /files/thirdparty.bin (10.42.42.42) 24.60ms
[I 190508 21:23:35 web:2246] 200 GET /files/thirdparty.bin (10.42.42.42) 6778.83ms
```

> > I assume it's too late for an undo now?
> 
> If it's still running the intermediate (connects to `vtrust-flash` at 10.42.42.42) you should still be able to `/undo`.

Thanks. I'll try to get a backup and wait for your answer first.

m-reiter, 2019-05-11:
Just tried it again and managed to get a full backup on the second try. See attachment.

[backup.img.gz](https://github.com/ct-Open-Source/tuya-convert/files/3169169/backup.img.gz)

kueblc, 2019-05-14:
Thanks @m-reiter, this is great news. I will see what I can find and report back.

pritpaul, 2019-06-04:
If you're using a sonoff.bin, try using the included sonoff-basic.bin instead.  I had this problem (~6-7 second fetch and doesn't flash) when I tried using sonoff.bin downloaded from the tasmota site.  Switching back to the basic worked.  The actual fetch on a successful flash for me tends to take more like 20 seconds.

m-reiter, 2019-06-04:
Thank you for the suggestion. I'm not 100% sure now but I think I first had this issue with the included sonoff-basic.bin. I'll use that for my next try anyway.

kueblc, 2019-09-30:
Unfortunately, I was unable to reproduce this issue and did not find any new clues as to why the backup process stalled for you. If you do discover any new information please feel free to share but for now I'm closing this as the original question has been answered and we haven't been able to make any progress on the other issues.

closed 2019-09-30 by kueblc