ct-Open-Source/tuya-convert #174: Can't flash Tuya based bulb

Sawmoke, 2019-04-14:
I'm trying to flash a lamp, LTLED-0178 from [lofter](https://www.amazon.co.uk/dp/B078W6D3Y1) which is recommended to use with Smart Life app, but I did not have problem controlling it with Tuya app too. Maybe is just the model not suitable for this flash, I would like to better understand that.
Wifi-AP seems to be OK, I'm able to connect through phone, tablet and so on. Even the lamp seems to connect succesfully, according to the logs. I did many tries so I'll post last lines of the log files. The start_flash.sh just loops on dots until I kill it with CTRL + C.
I tried running this on both Ubuntu 16.04 LTS and Ubuntu Server 18.04 LTS with similar results.

**smarthack-smartconfig.log**
```bash
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.
```

**smarthack-wifi.log**
```bash
nl80211: deinit ifname=wlx0c9d92b943c4 disabled_11b_rates=0
Backing up NetworkManager.cfg...
Restarting NetworkManager...
Failed to restart network-manager.service: Unit network-manager.service not found.
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
RTNETLINK answers: File exists
Starting AP on wlx0c9d92b943c4 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Using interface wlx0c9d92b943c4 with hwaddr 0c:9d:92:b9:43:c4 and ssid "vtrust-flash"
wlx0c9d92b943c4: interface state UNINITIALIZED->ENABLED
wlx0c9d92b943c4: AP-ENABLED
wlx0c9d92b943c4: STA ac:57:75:34:bc:ca IEEE 802.11: authenticated
wlx0c9d92b943c4: STA ac:57:75:34:bc:ca IEEE 802.11: associated (aid 1)
wlx0c9d92b943c4: AP-STA-CONNECTED ac:57:75:34:bc:ca
wlx0c9d92b943c4: STA ac:57:75:34:bc:ca RADIUS: starting accounting session B341CD45200C8D63
wlx0c9d92b943c4: STA ac:57:75:34:bc:ca WPA: pairwise key handshake completed (RSN)
wlx0c9d92b943c4: AP-STA-DISCONNECTED ac:57:75:34:bc:ca
wlx0c9d92b943c4: STA ac:57:75:34:bc:ca IEEE 802.11: authenticated
wlx0c9d92b943c4: STA ac:57:75:34:bc:ca IEEE 802.11: associated (aid 1)
wlx0c9d92b943c4: AP-STA-CONNECTED ac:57:75:34:bc:ca
wlx0c9d92b943c4: STA ac:57:75:34:bc:ca RADIUS: starting accounting session 9C8A9D907F1FF156
wlx0c9d92b943c4: STA ac:57:75:34:bc:ca WPA: pairwise key handshake completed (RSN)
wlx0c9d92b943c4: AP-STA-DISCONNECTED ac:57:75:34:bc:ca
wlx0c9d92b943c4: STA ac:57:75:34:bc:ca IEEE 802.11: authenticated
wlx0c9d92b943c4: STA ac:57:75:34:bc:ca IEEE 802.11: associated (aid 1)
wlx0c9d92b943c4: AP-STA-CONNECTED ac:57:75:34:bc:ca
wlx0c9d92b943c4: STA ac:57:75:34:bc:ca RADIUS: starting accounting session 7680D81B9C500A32
wlx0c9d92b943c4: STA ac:57:75:34:bc:ca WPA: pairwise key handshake completed (RSN)
wlx0c9d92b943c4: STA b4:e6:2d:07:bc:73 IEEE 802.11: authenticated
wlx0c9d92b943c4: STA b4:e6:2d:07:bc:73 IEEE 802.11: associated (aid 2)
wlx0c9d92b943c4: AP-STA-CONNECTED b4:e6:2d:07:bc:73
wlx0c9d92b943c4: STA b4:e6:2d:07:bc:73 RADIUS: starting accounting session A17C76B551B67D3F
wlx0c9d92b943c4: STA b4:e6:2d:07:bc:73 WPA: pairwise key handshake completed (RSN)
wlx0c9d92b943c4: STA b4:e6:2d:07:bc:73 WPA: group key handshake completed (RSN)
wlx0c9d92b943c4: AP-STA-DISCONNECTED ac:57:75:34:bc:ca
```
b4:e6:2d:07:bc:73 should be the lamp
ac:57:75:34:bc:ca is the phone

**smarthack-web.log**
```bash
Listening on port 80

URI:/gw.json?a=s.gw.update&gwId=02200466b4e62d07bc73&t=190&v=2.0&sign=5ff896439f79b0011cfead00fc084594
Answer s.gw.update
ESC[32m[I 190414 21:18:54 web:2246]ESC[m^O 200 POST /gw.json?a=s.gw.update&gwId=02200466b4e62d07bc73&t=190&v=2.0&sign=5ff896439f79b0011cfead00fc084594 (10.42.42.23) 4.39ms

URI:/gw.json?a=s.gw.dev.update&gwId=02200466b4e62d07bc73&t=7&v=2.0&sign=b7925bd9458faff2db7eb8e32a406033
[... 61 lines cut ...]
1555277063: Received PINGREQ from 02200466b4e62d07bc73
1555277063: Sending PINGRESP to 02200466b4e62d07bc73
1555277093: Received PINGREQ from 02200466b4e62d07bc73
1555277093: Sending PINGRESP to 02200466b4e62d07bc73
1555277123: Received PINGREQ from 02200466b4e62d07bc73
1555277123: Sending PINGRESP to 02200466b4e62d07bc73
1555277153: Received PINGREQ from 02200466b4e62d07bc73
1555277153: Sending PINGRESP to 02200466b4e62d07bc73
1555277183: Received PINGREQ from 02200466b4e62d07bc73
1555277183: Sending PINGRESP to 02200466b4e62d07bc73
1555277213: Received PINGREQ from 02200466b4e62d07bc73
1555277213: Sending PINGRESP to 02200466b4e62d07bc73
1555277243: Received PINGREQ from 02200466b4e62d07bc73
1555277243: Sending PINGRESP to 02200466b4e62d07bc73
1555277273: Received PINGREQ from 02200466b4e62d07bc73
1555277273: Sending PINGRESP to 02200466b4e62d07bc73
1555277303: Received PINGREQ from 02200466b4e62d07bc73
1555277303: Sending PINGRESP to 02200466b4e62d07bc73
1555277333: Received PINGREQ from 02200466b4e62d07bc73
1555277333: Sending PINGRESP to 02200466b4e62d07bc73
1555277363: Received PINGREQ from 02200466b4e62d07bc73
1555277363: Sending PINGRESP to 02200466b4e62d07bc73
1555277393: Received PINGREQ from 02200466b4e62d07bc73
1555277393: Sending PINGRESP to 02200466b4e62d07bc73
```

ragenhe, 2019-04-15:
These steps worked for me when I encountered the same situation.
First reset the bulb.  And stop any running processes.  Note: This process doesn't connect a separate device to the vtrust-flash.

1) Turn bulb on, it should be flashing.
2) Run ./start-flashing as directed.
* Pairing should happen within 10 seconds of power on.  
* Updating should begin within 10 seconds of pairing.

kueblc, 2019-04-15:
It looks like the pairing was successful and it is making calls. 

I don't see the registration call in your web log, nor do I see the upgrade command in your mqtt log. Since you've posted abridged logs, can you check to see if those exist?

Sawmoke, 2019-04-21:
Thanks for the support.
I've been trying several times, but without any success. Unfortunately, I was not able to reproduce the previous situation. With Ubuntu the lamp does not connect to wifi anymore and keeps blinking. Instead, I tried with a Raspberry and what I got was a near instantaneous (few dots..) connection to wifi but this was not logged in the wifi log file. My android phone connected in both case without any problem, and has been correctly logged.
The lamp does not have any switch, and to get into pairing mode you have to play with "on-off" cycle. I don't know whether the lamp actually needs to be reset someway, or just got silently updated via cloud (I don't see the lamp among connected devices to the router, though).

seanomat, 2019-04-23:
I can confirm: Exactly the same reproducible problem (logfiles) with Raspberry Pi 3 B+, Raspbian Strech lite and "Fcmila Bulb", another taste of Tuya-Bulp.

The process seems to hang on the endless "Unknown requests" in the web-log.

kueblc, 2019-04-23:
I'll need to see the rest of the logs @Sawmoke @seanomat 

Without looking it sounds like the device is running a newer firmware wherein this option for flashing custom firmware was patched

Sawmoke, 2019-04-23:
Those are the full new log files of my raspberry pi. Just some notes: the situation doesn't change either if I run this for 10 seconds or 10 minutes. The output is dots in loop, after 2-3 seconds the light stops blinking, though. As said in first message, ac:57:75:34:bc:ca is an android phone. I tried both "connecting first" and "prevent connection" of my android phone, leading to the same result.

**wifi log**

`^Cwlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED ac:57:75:34:bc:ca
wlan0: AP-DISABLED 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
RTNETLINK answers: File exists
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:e8:ed:85 and ssid "vtrust"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: STA ac:57:75:34:bc:ca IEEE 802.11: associated
wlan0: AP-STA-CONNECTED ac:57:75:34:bc:ca
wlan0: STA ac:57:75:34:bc:ca RADIUS: starting accounting session 9F97339ED0D91A06
wlan0: STA ac:57:75:34:bc:ca WPA: pairwise key handshake completed (RSN)
wlan0: STA ac:57:75:34:bc:ca IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED ac:57:75:34:bc:ca
wlan0: STA ac:57:75:34:bc:ca IEEE 802.11: disassociated
wlan0: STA ac:57:75:34:bc:ca IEEE 802.11: associated
wlan0: AP-STA-CONNECTED ac:57:75:34:bc:ca
wlan0: STA ac:57:75:34:bc:ca RADIUS: starting accounting session 2740F32D54FB7618
wlan0: STA ac:57:75:34:bc:ca WPA: pairwise key handshake completed (RSN)
wlan0: STA ac:57:75:34:bc:ca IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED ac:57:75:34:bc:ca
wlan0: STA ac:57:75:34:bc:ca IEEE 802.11: disassociated
wlan0: STA ac:57:75:34:bc:ca IEEE 802.11: disassociated
wlan0: STA ac:57:75:34:bc:ca IEEE 802.11: associated
wlan0: AP-STA-CONNECTED ac:57:75:34:bc:ca
wlan0: STA ac:57:75:34:bc:ca RADIUS: starting accounting session 1203E0136CCAAA32
wlan0: STA ac:57:75:34:bc:ca WPA: pairwise key handshake completed (RSN)
wlan0: STA ac:57:75:34:bc:ca IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED ac:57:75:34:bc:ca
wlan0: STA ac:57:75:34:bc:ca IEEE 802.11: disassociated
wlan0: STA ac:57:75:34:bc:ca IEEE 802.11: disassociated
wlan0: interface state ENABLED->DISABLED
wlan0: AP-DISABLED 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
`

**mqtt log**

```
^C1556054094: mosquitto version 1.4.10 terminating
1556054100: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1556054100: Using default config.
1556054100: Opening ipv4 listen socket on port 1883.
1556054100: Opening ipv6 listen socket on port 1883.
```
** smart config log**

```
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.
```

**Web log**

```
^CTraceback (most recent call last):
  File "./fake-registration-server.py", line 129, in <module>
    main()
  File "./fake-registration-server.py", line 125, in main
    tornado.ioloop.IOLoop.current().start()
  File "/usr/local/lib/python3.5/dist-packages/tornado/platform/asyncio.py", line 148, in start
    self.asyncio_loop.run_forever()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 421, in run_forever
    self._run_once()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 1424, in _run_once
    handle._run()
  File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
    self._callback(*self._args)
  File "/usr/local/lib/python3.5/dist-packages/tornado/ioloop.py", line 743, in _run_callback
    ret = callback()
  File "/usr/local/lib/python3.5/dist-packages/tornado/ioloop.py", line 907, in _run
    return self.callback()
  File "/usr/local/lib/python3.5/dist-packages/tornado/autoreload.py", line 186, in _reload_on_update
    if path.endswith(".pyc") or path.endswith(".pyo"):
KeyboardInterrupt
Listening on port 80
```

seanomat, 2019-04-27:
I was able to flash my bulbs now.
After seeing the process work with some TECKIN Smart-plugs, it seems I had the wrong timing with flashing the bulbs and was to impatient.

demian85, 2019-04-27:
I'm having the same issue. I get stuck in the first step. Logs look the same as above. Bulb keeps flashing forever :(

Sawmoke, 2019-05-01:
Some update with Ubuntu 18.04 and the fix provided in [#152](https://github.com/ct-Open-Source/tuya-convert/issues/152).
After some time of blinking (around 1 minute) the lamp connected and stopped blinking. Anyway dots keep going.
Logs
**Wifi Log**
```
Backing up NetworkManager.cfg...
Restarting NetworkManager...
Failed to restart network-manager.service: Unit network-manager.service not found.
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
Starting AP on wlx0c9d92b943c4 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Using interface wlx0c9d92b943c4 with hwaddr 0c:9d:92:b9:43:c4 and ssid "vtrust-flash"
wlx0c9d92b943c4: interface state UNINITIALIZED->ENABLED
wlx0c9d92b943c4: AP-ENABLED
wlx0c9d92b943c4: STA ac:57:75:34:bc:ca IEEE 802.11: authenticated
wlx0c9d92b943c4: STA ac:57:75:34:bc:ca IEEE 802.11: associated (aid 1)
wlx0c9d92b943c4: AP-STA-CONNECTED ac:57:75:34:bc:ca
wlx0c9d92b943c4: STA ac:57:75:34:bc:ca RADIUS: starting accounting session A9659CD2DF1396B9
wlx0c9d92b943c4: STA ac:57:75:34:bc:ca WPA: pairwise key handshake completed (RSN)
wlx0c9d92b943c4: STA b4:e6:2d:07:bc:73 IEEE 802.11: authenticated
wlx0c9d92b943c4: STA b4:e6:2d:07:bc:73 IEEE 802.11: associated (aid 2)
wlx0c9d92b943c4: AP-STA-CONNECTED b4:e6:2d:07:bc:73
wlx0c9d92b943c4: STA b4:e6:2d:07:bc:73 RADIUS: starting accounting session AB2DB15D353EF92D
wlx0c9d92b943c4: STA b4:e6:2d:07:bc:73 WPA: pairwise key handshake completed (RSN)
```

**MQTT Log** (I think here is the most relevant error)
```
1556698510: mosquitto version 1.4.15 (build date Wed, 13 Feb 2019 00:27:01 +0000) starting
1556698510: Using default config.
1556698510: Opening ipv4 listen socket on port 1883.
1556698510: Error: Address already in use
```

**Web Log**
```
Listening on port 80

URI:/gw.json?a=s.gw.token.get&gwId=02200466b4e62d07bc73&other={"token":"00000000","region":"US","tlinkStat":{"configure":"smartconfig","time":4,"source":"ap","path":"broadcast"}}&t=189&v=3.0&sign=7e78d4b69abbd3431f225adae9b58c2c
Answer s.gw.token.get
ESC[32m[I 190501 08:19:02 web:2246]ESC[m^O 200 POST /gw.json?a=s.gw.token.get&gwId=02200466b4e62d07bc73&other={"token":"00000000","region":"US","tlinkStat":{"configure":"smartconfig","time":4,"source":"ap","path":"broadcast"}}&t=189&v=3.0&sign=7e78d4b69abbd3431f225adae9b58c2c (10.42.42.23) 0.96ms

URI:/gw.json?a=s.gw.dev.pk.active&gwId=02200466b4e62d07bc73&other={"token":"00000000"}&t=7&v=3.0&sign=7ed439329ac236f0c221fc553f81ae86
Answer s.gw.dev.pk.active
READ GW ID 02200466b4e62d07bc73
TRIGGER UPGRADE IN 10 SECONDS
Trigger upgrade in 10 seconds
ESC[32m[I 190501 08:19:02 web:2246]ESC[m^O 200 POST /gw.json?a=s.gw.dev.pk.active&gwId=02200466b4e62d07bc73&other={"token":"00000000"}&t=7&v=3.0&sign=7ed439329ac236f0c221fc553f81ae86 (10.42.42.23) 9.65ms
c6d1e35eb355d9b574d6ef050a4c6fd4
b'2.1b355d9b574d6ef05t5inCMDHpcJKQpxfOe2RbVH1YwLVfr0s4E3bUdPRrItGX7XX6A4wTaEQb+PJfVb9Jsm5uMafEvnzLLz/1MQ4v/MxpXYJi73XFS7hXI04lSjdg4SXZL9kLEC9aqKXAOwI'
Traceback (most recent call last):
  File "mq_pub_15.py", line 110, in <module>
    sys.exit(main())
  File "mq_pub_15.py", line 105, in main
    client.connect(broker)
  File "/usr/local/lib/python3.6/dist-packages/paho/mqtt/client.py", line 839, in connect
    return self.reconnect()
  File "/usr/local/lib/python3.6/dist-packages/paho/mqtt/client.py", line 962, in reconnect
    sock = socket.create_connection((self._host, self._port), source_address=(self._bind_address, 0))
  File "/usr/lib/python3.6/socket.py", line 724, in create_connection
    raise err
  File "/usr/lib/python3.6/socket.py", line 713, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
```

dicko72, 2019-05-19:
When I try and flash my fcmila bulbs sometimes they stop flashing but the dots just keep going. Anybody have a workaround?

giacomotontini, 2019-06-11:
Same here
web.log
```Traceback (most recent call last):

  File "./fake-registration-server.py", line 129, in <module>

    main()

  File "./fake-registration-server.py", line 125, in main

    tornado.ioloop.IOLoop.current().start()

  File "/usr/local/lib/python3.6/dist-packages/tornado/platform/asyncio.py", line 148, in start

    self.asyncio_loop.run_forever()

  File "/usr/lib/python3.6/asyncio/base_events.py", line 438, in run_forever

    self._run_once()

  File "/usr/lib/python3.6/asyncio/base_events.py", line 1415, in _run_once

    event_list = self._selector.select(timeout)

  File "/usr/lib/python3.6/selectors.py", line 445, in select

    fd_event_list = self._epoll.poll(timeout, max_ev)

marrobHD, 2019-09-13:
Same on shp6

kueblc, 2019-09-28:
Closing due to inactivity, feel free to open a new issue if you have any concerns.

For those that missed it, support for HTTPS firmware is in the works #279

closed 2019-09-28 by kueblc